Genesis-Embodied-AI/genesis-world #4: Any timepline of open source

YunxiTang, 2024-11-15:
Hi,
is there any timeline for open sourcing this project?

Milotrince, 2024-11-26:
I heard early December but we never know...

zhouxian, 2024-11-26:
Hi, thank you for your interest in our project! We planned to release it last week (before my phd defense) but in order to polish up some demo associated with it, we decided to postpone it a bit and I had to spend some time on my graduation-related matters. Now everything is ready for release and we are finalizing some humanoid demo, but we accidentally broke a humanoid hardware and we are migrating to a more complex platform, which drags the progress a little bit further. We promise to release it in early december :)

sxyu, 2024-12-16:
Hi, is there any update on the timeline? Thanks

closed 2024-12-18 by zhouxian