Genesis-Embodied-AI/genesis-world #10: Fixing OpenGL.error.Error when running Genesis quick start

youliangtan, 2024-12-19:
## Summary

Thanks for this awesome work! Im getting `OpenGL.error.Error: Attempt to retrieve context when no valid context` when running the [quickstart code ](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html) for genesis. My env is python `3.12` and ubuntu 22.04

The quick fix is to add set `osmesa` instead of `egl`. example:

```
PYOPENGL_PLATFORM=osmesa python test_gs.py 
```

## Error logs

```bash
(genesis) youliang@youliang-Precision-5690:~$ python test_gs.py 
[Genesis] [18:19:55] [INFO] ╭───────────────────────────────────────────────────────────────────────╮
[Genesis] [18:19:55] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [18:19:55] [INFO] ╰───────────────────────────────────────────────────────────────────────╯
[Genesis] [18:19:55] [INFO] Running on [Intel(R) Core(TM) Ultra 7 165H] with backend gs.cpu. Device memory: 62.26 GB.
[Genesis] [18:19:55] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [18:19:55] [INFO] Scene <a9eeac3> created.
[Genesis] [18:19:55] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <ec0dd58>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [18:19:55] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <e3074a9>, morph: <gs.morphs.MJCF(file='/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [18:19:56] [INFO] Building scene <a9eeac3>...
[Genesis] [18:19:59] [INFO] Compiling simulation kernels...
[Genesis] [18:20:03] [INFO] Building visualizer...
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 469, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
              ^^^^^^^^^^^^^^^^^^^^^
  File "/home/youliang/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context
```

sangminkim-99, 2024-12-19:
In Python `3.11` on Ubuntu 22.04, I encountered the following error with OSMesa:
`ImportError: Cannot use EGL rendering platform. The PYOPENGL_PLATFORM environment variable is set to {!r} (should be either unset or 'egl').`

To resolve this, I disabled the viewer and visualized the rendered RGB and depth using an offscreen renderer.
I hope this helps others facing the same issue :)

```
scene = gs.Scene(show_viewer=False)
cam = scene.add_camera(
    res=(640, 480), pos=(2.5, 0.0, 2.0), lookat=(0, 0, 0.8), fov=30, GUI=False
)

for i in range(1000):
    rgb, depth, _, _ = cam.render(depth=True)
```

begin-fang, 2024-12-19:
Actually, I adopt the solution to add `set osmesa instead of egl` but have no use, then I change `PyOpenGL` to `3.1.5` version and then fixed this error ...

ZM-Zhou, 2024-12-21:
> In Python `3.11` on Ubuntu 22.04, I encountered the following error with OSMesa: `ImportError: Cannot use EGL rendering platform. The PYOPENGL_PLATFORM environment variable is set to {!r} (should be [...]

I found that  use 'glx' and setting the MuJoCo GL by `export MUJOCO_GL=glx` could solve the error like this for me (ubuntu 20.04)

maksgepner, 2024-12-26:
Can confirm setting PYOPENGL_PLATFORM fixes it for me as well (Ubuntu 24.04, Python 3.12). By default, the variable did not have any value; setting it to either "glx" or "x11" fixes the problem (but "egl" doesn't work). Perhaps a Wayland-associated issue?

closed 2025-02-22 by YilingQiao