Genesis-Embodied-AI/genesis-world #101: Setting base velocity

dmarew, 2024-12-19:
Is there a way to set the base link linear and angular velocities in the same way we can set position and orientation?

ziyanx02, 2024-12-20:
I guess it's same as setting the velocity of the first six dofs

zswang666, 2024-12-21:
Quick follow up on @ziyanx02 answer (also, remember to set `fixed=False` when instantiating the robot).

If you are looking for something like `rigid_entity.set_pos`, which works for both fixed- or floating-based body, it can be somewhat tricky to do for `set_vel`. If `fixed=True`, attributes related to dynamics (e.g., velocity, acceleration, etc) of the base link are not included in the coordinates `qpos` or dofs `dofs_state` inside the solver. Thus, the velocity of the base link won't take effect on the position, as in [`rigid_solver._func_integrate`](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L2715) where nothing done in the integration can affect the later on [kinematics](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L1966) (`links_state.vel` basically cannot affect `links_state.pos`). 

For sure we can hack this but this may make the code super ugly (at least so far I didn't think of a good way). One workaround is to do the integration outside and use `set_pos` if you really want to set velocity on a fixed-based robot. But again this won't properly handle the dynamics -- you should set `fixed=False` to incorporate dynamics in the base link and do `set_dofs_velocity`.

closed 2024-12-21 by zswang666