Genesis-Embodied-AI/genesis-world #102: black window while running in wsl2

Controlist20, 2024-12-19:
Hi I was trying to run the simplest example that you shared in Hello Genesis. It is running but it is a black window. I tested my opengl with another simple script, it was running without any issue.

[image]

[image]

urish, 2024-12-19:
Try:

```bash
export PYOPENGL_PLATFORM=egl
```

orioljim1, 2024-12-19:
This does not work for me can you share which version of  PYOPENGL and PyOpenGL_accelerate are you running? @urish 
Thanks!

urish, 2024-12-19:
```
$ pip freeze | grep -i pyopengl
PyOpenGL==3.1.7
```

about PyOpenGL_accelerate  - I don't think I have it installed

meijie-jesse, 2024-12-20:
i meet this problem too. but in the black window, it shows "[i] show keyboard instructions", which i can triger by keyboad and shows menus.

RRGGZZ, 2024-12-20:
I also encountered the same problem in WSL2, and the window remained black.

chenqiguo, 2024-12-20:
I met with the same problem with black window when running the hello_genesis, on ubuntu22.

chenqiguo, 2024-12-20:
> I met with the same problem with black window when running the hello_genesis, on ubuntu22.

Decreasing the  max_FPS to 30 or 20 solve the problem but very inefficient

meijie-jesse, 2024-12-20:
> > I met with the same problem with black window when running the hello_genesis, on ubuntu22.
> 
> Decreasing the max_FPS to 30 or 20 solve the problem but very inefficient

Thanks, I tried 40、30、20 but it not work. I successfully run it by switching to the physical machine, not vmware or wsl2.

RRGGZZ, 2024-12-20:
This project does not appear to run properly in the virtual machine

AgentOfLyra, 2024-12-20:
[image]
Very, very slow, very stuck

aidinism, 2024-12-23:
Same here. It runs, and the debug says it's at ~60 FPS, but the GUI is super slow at ~1 FPS.
It's the same on CPU or CUDA/GPU.
Interestingly, it was fine before the latest commit.

[image]
[image]

myamazum, 2025-01-05:
hi, guys. I met same problems in my New year holidays.
Finally I found a solution in my case. (llvmpipe error I guess)

situation : tutorial / hello_genesis.py running in docker on wsl2. 
HW : laptop ryzen 7 (with AMD Grafic card) and Geforce RTX 3050 : **two graphic card cause a problem in docker**.

1. Library linker is complex in new wsl2, if driver was instaled in Windows 11.   
    Nvidia driver lay in /usr/lib/wsl/lib, not installed in default settings.
    Add drivers link in LD_LIBRARY_PATH.
2. Dockerfile in Genesis doesn't refer previous directory, so that you should add volume link in the Dockerfile.
3. Genesis uses mesa library but it's difficult to find nvidia driver if there are two graphic library.  
    Add MESA_D3D12_DEFAULT_ADAPTER_NAME="NVIDIA"

```bash
docker run --gpus all --rm -it -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix -v $PWD:/workspace -v /usr/lib/wsl:/usr/lib/wsl genesis
export LD_LIBRARY_PATH=/usr/lib/wsl/lib:LD_LIBRARY_PATH
export MESA_D3D12_DEFAULT_ADAPTER_NAME="NVIDIA"
python examples/tutorials/hello_genesis.py
```

4. check settings using mesa-utils.

Following status is correct.
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA GeForce RTX 3050 Laptop GPU) (0xffffffff)

```bash
apt update && apt install mesa-utils
glxinfo -B
```    

Sincerely

hongbo-miao, 2025-01-17:
I met same issue. I do need `export PYOPENGL_PLATFORM=glx` to get it work.

However, `export LD_LIBRARY_PATH=/usr/lib/wsl/lib` is not necessary after I remove incorrect driver. I posted the detail at https://stackoverflow.com/a/79357542/2000548

closed 2025-02-23 by YilingQiao