Genesis-Embodied-AI/genesis-world #104: AttributeError: module 'genesis' has no attribute 'generate'

vis7, 2024-12-19:
I created venv in python - 3.12.3 in Ubuntu 24.04 0S

I installed genesis with below command
`pip install genesis-world `

I also installed pytorch with below command from [official site](https://pytorch.org/get-started/locally/)
pip3 install torch torchvision torchaudio --index-url https://download.pytorch.org/whl/cu118

but getting below error Error 
```
$ python
Python 3.12.3 (main, Nov  6 2024, 18:32:19) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import genesis as gs
>>> gs.generate("mario")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'genesis' has no attribute 'generate'

```

RobRoyce, 2024-12-19:
Duplicate of #30

ziyanx02, 2024-12-22:
We greatly appreciate your interest in Genesis! Access to the generative framework will be rolled out gradually in the near future. We will continue to provide updates in #6, so stay tuned!

closed 2024-12-22 by ziyanx02