Genesis-Embodied-AI/genesis-world #108: Failed to run inverse_kinematics_motion_planning example: ModuleNotFoundError: No module named 'ompl'

pengzhenghao, 2024-12-19:
https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/inverse_kinematics_motion_planning.html

Traceback (most recent call last):
  File "/home/zhenghao/anaconda3/envs/go/envs/gene/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 1230, in plan_path
    from ompl import base as ob
ModuleNotFoundError: No module named 'ompl'

bdelhaisse, 2024-12-20:
@pengzhenghao did you install it? https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html#optional-motion-planning

closed 2024-12-22 by ziyanx02

vengov, 2025-06-16:
I do not see the installation instructions in the provided link. Could you suggest how to successfully install OMPL with python bindings in a virtual environment (venv)?

duburcqa, 2025-06-16:
You should install the latest main branch, it will install OMPL automatically. Unless you are on Windows, in which case there is no solution I'm afraid.