Genesis-Embodied-AI/genesis-world #11: OpenGL.error.GLError: GLError(err = 1282, description = b'invalid operation', baseOperation = glBindFramebuffer, cArguments = (GL_DRAW_FRAMEBUFFER, 0) )

sarmadgulzar, 2024-12-19:
Device: Apple MacBook Pro
Processor: M1 Pro
RAM: 16GB
Python: 3.10.16

Running this code:

```py
import argparse

import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(backend=gs.cpu)

    ########################## create a scene ##########################

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(3.5, 0.0, 2.5),
            camera_lookat=(0.0, 0.0, 0.5),
            camera_fov=40,
        ),
        show_viewer=args.vis,
        rigid_options=gs.options.RigidOptions(
            dt=0.01,
            gravity=(0.0, 0.0, -10.0),
        ),
    )

    ########################## entities ##########################
    plane = scene.add_entity(gs.morphs.Plane())
    r0 = scene.add_entity(
        gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
    )

    ########################## build ##########################
    scene.build()

    gs.tools.run_in_another_thread(fn=run_sim, args=(scene, args.vis))
    if args.vis:
        scene.viewer.start()

def run_sim(scene, enable_vis):
    from time import time

    t_prev = time()
    i = 0
    while True:
        i += 1

        scene.step()

        t_now = time()
        print(1 / (t_now - t_prev), "FPS")
        t_prev = t_now
        if i > 200:
            break

    if enable_vis:
        scene.viewer.stop()

if __name__ == "__main__":
    main()
[... 471 lines cut ...]
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._renderer.render_texts(
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 258, in render_texts
    self._configure_forward_pass_viewport(0)
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 1160, in _configure_forward_pass_viewport
    glBindFramebuffer(GL_DRAW_FRAMEBUFFER, 0)
  File "/Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glBindFramebuffer,
        cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)
```

YutoYasunaga, 2024-12-19:
I have same problem with MacBook Pro M1 too.

ariszxxu, 2024-12-19:
Same here with M3 Pro.

YutoYasunaga, 2024-12-19:
Resolved by adding `PYOPENGL_PLATFORM=egl`

But now I have issue #32

RobRoyce, 2024-12-19:
Works for me after:

* Install `ffmpeg` using `brew install ffmpeg`
* Upgrade PyOpenGL with `pip install -U PyOpenGL`
* Remove `pygel3d` with `pip uninstall pygel3d` (that will get rid of the warnings you get when first launching)

CommanderCrowCode, 2024-12-19:
> Works for me after:
> 
> * Install `ffmpeg` using `brew install ffmpeg`
> * Upgrade PyOpenGL with `pip install -U PyOpenGL`
> * Remove `pygel3d` with `pip uninstall pygel3d` (that will get rid of the warnings you get when first launching)

Still doesn't work for me. I have done the above.
```
❯ ffmpeg -version | head -n 1
ffmpeg version 7.1 Copyright (c) 2000-2024 the FFmpeg developers
❯ uv pip list | grep -E "pyopengl | gel3d"
pyopengl               3.1.7
❯ .venv/bin/python render_on_macos.py --vis
```

Still got GLError
```
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)
```

zeng-yifei, 2024-12-19:
me too, still get this error on mac:
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)

zhushunqing, 2024-12-19:
I got same issue.
iMac 27 / 3.8 GHz 8-Core Intel Core i7 / AMD Radeon Pro 5500 XT 8 GB

pkoszalkaa, 2024-12-19:
Same here
M1 Pro

zy-meng, 2024-12-19:
I got sam issue.
ubuntu 20.04 / Intel(R) Xeon(R) Platinum 8358P CPU @ 2.60GHz / NVIDIA GTX 4090

YutoYasunaga, 2024-12-19:
I tried both MacBook and Ubuntu, even Windows, but nothing works. So many errors, it keeps raising errors and other errors again and again.

xiaoxubeii, 2024-12-19:
Same here, m1 pro

eyalmentee, 2024-12-19:
same here, m4 pro

Brief-rf, 2024-12-19:
same here, m2 air

asapsav, 2024-12-19:
same, m2 pro

sarmadgulzar, 2024-12-19:
> Works for me after:
> 
> * Install `ffmpeg` using `brew install ffmpeg`
> * Upgrade PyOpenGL with `pip install -U PyOpenGL`
> * Remove `pygel3d` with `pip uninstall pygel3d` (that will get rid of the warnings you get when first launching)

This didn't work for me:
```
(venv) genesis-tests $ brew install ffmpeg
==> Downloading https://formulae.brew.sh/api/formula.jws.json
######################################################################################## 100.0%
==> Downloading https://formulae.brew.sh/api/cask.jws.json
######################################################################################## 100.0%
Warning: ffmpeg 7.1_3 is already installed and up-to-date.
To reinstall 7.1_3, run:
  brew reinstall ffmpeg
(venv) genesis-tests $ pip install PyOpenGL   
Requirement already satisfied: PyOpenGL in ./venv/lib/python3.10/site-packages (3.1.7)
(venv) genesis-tests $ pip uninstall pygel3d
Found existing installation: PyGEL3D 0.5.2
Uninstalling PyGEL3D-0.5.2:
  Would remove:
    /Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/PyGEL3D-0.5.2.dist-info/*
    /Users/sarmad/Documents/Code/Python/genesis-tests/venv/lib/python3.10/site-packages/pygel3d/*
Proceed (Y/n)? 
  Successfully uninstalled PyGEL3D-0.5.2
```

buck-0x, 2024-12-19:
Same issue for me.

vegito22, 2024-12-20:
I was facing a similar error, but managed to resolve it using conda (https://docs.anaconda.com/miniconda/install/#quick-command-line-install). The viewer loads for me 

Device: Apple MacBook Pro
Processor: M1 Pro
RAM: 16GB
Python: 3.12.8

soumith, 2024-12-20:
this was quite a rabbit-hole, but basically what I found is that on a second "on_draw" of the pyrenderer viewer, the GL context is already in a bad state (from the first draw). What sets it into a bad state is `self.jit.update(scene)` in the renderer.

My speculation is that the JIT renderer isn't written to OpenGL compliance, or MacOS + Python (via uv) by default ships a crappy and incompliant OpenGL. Unclear which.

YutoYasunaga, 2024-12-20:
Confirmed it works by using Conda for environment management. The viewer has rendered for me.
Thank @vegito22 

[image]

CommanderCrowCode, 2024-12-20:
> Confirmed it works by using Conda for environment management. The viewer has rendered for me. Thank @vegito22
> 
> <img alt="Screenshot 2024-12-20 at 11 51 57" width="1442" src="https://private-user-images.githubusercontent.com/6220668/397600312-2dfc7553-e5cd-4e5e-a1bf-198cd8289ac0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NjQzNDUsIm5iZiI6MTczNDY2NDA0NSwicGF0aCI6Ii82MjIwNjY4LzM5NzYwMDMxMi0yZGZjNzU1My1lNWNkLTRlNWUtYTFiZi0xOThjZDgyODlhYzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjBUMDMwNzI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzgwMWM1MjFmOGRmMjZlNGY4Nzk4Y2EyMmY2ZGU4ZDFkMmU4NjIxZDg2Mjg3N2FlZWRjZTEzMjU0ODFlNDExMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.aTRZlkqn51BihmfUwBOx7H_ulJ7M1-kcXup8GGDYea4">

Second confirmation that Conda-managed environment works
[image]

xiaoxubeii, 2024-12-20:
Me to, can run under conda.
[image]

xiaoxubeii, 2024-12-20:
> What python version are you using in conda?
> 
> Python 3.11 is not working for me and still get this error
> 
> ```
> OpenGL.error.GLError: GLError(
> 	err = 1282,
> 	description = b'invalid operation',
> 	baseOperation = glBindFramebuffer,
> 	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
> )
> ```

Python 3.12.8

mgottscho, 2024-12-20:
Same errors. Confirming the miniconda suggestion also worked for me.

M3 Pro
36 GB
MacOS 15.2 (24C101)
python 3.12.8

apandy02, 2024-12-20:
> or MacOS + Python (via uv) by default ships a crappy and incompliant OpenGL (or badly compiled taichi)

@soumith I think the above makes sense. I went down the rabbit hole trying to get it to work in several differently configured uv environments with no success. one shotted it with conda + the above suggestions

emalihin, 2024-12-20:
> I was facing a similar error, but managed to resolve it using conda (https://docs.anaconda.com/miniconda/install/#quick-command-line-install). The viewer loads for me
> 
> Device: Apple MacBook Pro Processor: M1 Pro RAM: 16GB Python: 3.12.8

how did you install `genesis-world` in conda?

soumith, 2024-12-20:
can confirm, it works with `conda`.

@emalihin after installing conda, I did:

```
conda create -n genesis python=3.12
conda activate genesis
pip install genesis-world
pip install torch
```

Tuxliri, 2024-12-20:
I reproduced @soumith instructions and the `render_on_macos.py` works correctly.

PieterBecking, 2024-12-21:
Thank you @soumith .

sonsongithub, 2025-01-08:
Thank you @soumith 

I can execute `render_on_macos.py` correctly, using `pyenv` and `conda`.

zacamaso, 2025-01-27:
> What python version are you using in conda?
> 
> Python 3.11 is not working for me and still get this error
> 
> ```
> OpenGL.error.GLError: GLError(
> 	err = 1282,
> 	description = b'invalid operation',
> 	baseOperation = glBindFramebuffer,
> 	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
> )
> ```

Python 3.10