Genesis-Embodied-AI/genesis-world #110: Invalid device ID [WINDOWS]

DavidSeyserGit, 2024-12-19:
This is the error raised when running the visualization and rendering example from the docs on Windows. 
```
/miniconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/platforms/egl.py", line 84, in get_device_by_index
    raise ValueError("Invalid device ID ({})".format(device_id, len(devices)))
```

dsxksss, 2024-12-20:
> This is the error raised when running the visualization and rendering example from the docs on Windows.这是在Windows上从文档运行可视化和渲染示例时引发的错误。
> 
> ```
> /miniconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/platforms/egl.py", line 84, in get_device_by_index
>     raise ValueError("Invalid device ID ({})".format(device_id, len(devices)))
> ```

用的什么卡跑?

DavidSeyserGit, 2024-12-20:
Hi, thanks for your comment. I'm using a RTX3060 and a Ryzen5 5600. I get this error when running in GPU and CPU mode.

xsp-cz, 2025-02-26:
> Is this issue still occurring ? Do you have a reproduction script ? If so, I should be able to fix it!

Hi I also encountered this issue on A100/A10, Ubuntu system. Would you mind offering your solution  thx!

duburcqa, 2025-02-27:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/issues/110

closed 2025-02-27 by duburcqa