Genesis-Embodied-AI/genesis-world #111: How to export properties in simulation, e.g. point positions

kyleleey, 2024-12-19:
Hi, thanks for open-sourcing such a great framework.

I wonder is there an interface to export the properties of entities during each scene.step()? For example, when using mesh to initialize the entities, is there a function to directly get the positions of mesh vertices in each step?

Thanks!!

kyleleey, 2024-12-20:
Seems it's possible to use `entity.solver.particles.to_numpy()['pos']` to get property like position. I wonder if there exists a more formal interface? Also when entities are with materials like liquid or fluid, the particles are more than original mesh vertices, what's the parameters that store the mapping from those particles to original mesh vertices' indices?

Thanks!!

zswang666, 2024-12-21:
Thanks for sharing the answer. In general, you can check the member function in [entity](https://github.com/Genesis-Embodied-AI/Genesis/tree/main/genesis/engine/entities). For example, things related to what you want are
- [mpm_entity.get_state](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/entities/mpm_entity.py#L348): f, pos, vel, C, F, Jp, active
- [sph_entity.get_state](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/entities/sph_entity.py#L114): pos, vel
- [fem_entity.get_state](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/entities/fem_entity.py#L122): f, pos, vel, active

Also note that if you access directly from the solver, remember to take indexing offset into consideration, i.e., the range of particle indices that belong to the same entity like [particle_start](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/entities/particle_entity.py#L453) -- also for other types of entity as well.

We will try to improve the api as well as the doc

kyleleey, 2024-12-21:
Hi,

Thanks for the reply! Good to know there are a bunch of interfaces on getting properties for particles.

I'm still curious if there is anyway to get the properties of input morphs, especially meshes? At least its vertices position properties. Because if set the visual mode as "visual", the renderer is rendering the mesh instead of particles, so I assume there exists some ways to get the positions of mesh vertices?