Genesis-Embodied-AI/genesis-world #12: Visualization Bug: AttributeError: function 'glGetUniformLocation' not found

Bardia323, 2024-12-19:
Visualizing the elastic dragon example on Windows:

```
(venv) PS S:\Tools\genesis> py .\elastic_dragon.py --vis
[Genesis] [21:41:21] [INFO] Running on [NVIDIA GeForce RTX 3090 Ti] with backend gs.cuda. Device memory: 23.99 GB.
[Genesis] [21:41:22] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.1, commit 0131dce9, win, python 3.10.5
[Genesis] [21:41:22] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [21:41:22] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [21:41:22] [INFO] Scene <2463c84> created.
[Genesis] [21:41:22] [INFO] Adding <gs.PBD3DEntity>. idx: 0, uid: <e6ec9b0>, morph: <gs.morphs.Mesh(file='S:\Tools\genesis\venv\lib\site-packages\genesis\assets\meshes/dragon/dragon.obj')>, material: <gs.PBD.Elastic>.
[Genesis] [21:41:22] [DEBUG] Remeshed file (`.rm`) found in cache.
[Genesis] [21:41:23] [INFO] Building scene <2463c84>...
[Genesis] [21:41:25] [INFO] Compiling simulation kernels...
[Genesis] [21:41:30] [INFO] Building visualizer...
Traceback (most recent call last):
  File "S:\Tools\genesis\elastic_dragon.py", line 58, in <module>
    main()
  File "S:\Tools\genesis\elastic_dragon.py", line 49, in main
    scene.build()
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
  File "S:\Tools\genesis\venv\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found
[Genesis] [21:41:30] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

causes an AttributeError. 
Works fine without the --vis flag.

qiyashandian, 2024-12-19:
I met the same problem

r530044129, 2024-12-19:
same problem here

YilingQiao, 2024-12-19:
Is everyone here using Windows?

r530044129, 2024-12-19:
> Is everyone here using Windows?

Here is win11

Awilekong, 2024-12-19:
same problem
using win11

zhenjia-xu, 2024-12-19:
https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html
Based on our test, rendering is not supported on Windows due to OpenGL issues. 
It would be super appreciated if someone could find a solution. Otherwise, Linux or macOS are the only platforms supporting rasterization.

agenel, 2024-12-19:
I added a small try/except feature in the 'load_func()' function in numba_gl_wrapper.py.
Here is the list of libraries that can be loaded successfully or not. I hope it helps. 
```

Failed to load glGetUniformLocation: function 'glGetUniformLocation' not found
Failed to load glUniformMatrix4fv: function 'glUniformMatrix4fv' not found
Failed to load glUniform1i: function 'glUniform1i' not found
Failed to load glUniform1f: function 'glUniform1f' not found
Failed to load glUniform2f: function 'glUniform2f' not found
Failed to load glUniform3fv: function 'glUniform3fv' not found
Failed to load glUniform4fv: function 'glUniform4fv' not found
Failed to load glBindVertexArray: function 'glBindVertexArray' not found
Failed to load glActiveTexture: function 'glActiveTexture' not found
Loaded glBindTexture successfully.
Loaded glEnable successfully.
Loaded glDisable successfully.
Loaded glBlendFunc successfully.
Loaded glPolygonMode successfully.
Loaded glCullFace successfully.
Failed to load glDrawElementsInstanced: function 'glDrawElementsInstanced' not found
Failed to load glDrawArraysInstanced: function 'glDrawArraysInstanced' not found
Failed to load glUseProgram: function 'glUseProgram' not found
Loaded glFlush successfully.
Loaded glReadPixels successfully.
Failed to load glBindBuffer: function 'glBindBuffer' not found
Failed to load glBufferData: function 'glBufferData' not found
Failed to load glBufferSubData: function 'glBufferSubData' not found 
```

QixuanAI, 2024-12-19:
I met the same problem in Win10.

I found that in pyrender/numba_gl_wrapper.py/load_func, the variable `dll` pointed at 'C:\\WINDOWS\\system32\\opengl32.dll', which we know that windows only support OpenGL1.1, but what we need is OpenGL2.0 or later. so is there anyone know, how to change the default OpenGL version that `dll = GL.platform.PLATFORM.GL` pointed at?

Another interest thing is, code above can work:
```shell
python
>>> from OpenGL.GL import glGetUniformLocation as a
>>> from OpenGL.GL import glActiveTexture as b
>>> a
OpenGL.lazywrapper.lazy( 'glGetUniformLocation' )
>>> b
<OpenGL.platform.baseplatform.glActiveTexture object at 0x000002845FD4C550>
```
So I wonder if there's a way that we just import those function directly instead of load them from `load_func`?

Physics-Lee, 2024-12-19:
Same Problem

TheSeanLavery, 2024-12-19:
Lame. I spent a while trying to get this to run. This is the blocking error.

Should put a warning on the repo that it doesn't work on windows 11 currently

Katehuuh, 2024-12-19:
<details>
<summary>We can make a <strong>workaround</strong>, but simpler and slower, for Windows visualization by modifying <code><a href="https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/elastic_dragon.py">elastic_dragon.py</a></code> <code>--vis</code> to render a GIF using <code>matplotlib</code> instead of OpenGL.</summary>

1. Add imports at top:
```python
import matplotlib.pyplot as plt, imageio, io
```

2. Disable OpenGL viewer:
```python
show_viewer=False
```

3. Replace simulation loop visualization:
```python
for i in range(horizon):
    scene.step()
```
with
```python
if args.vis:
    # Pre-create figure once
    fig = plt.figure(figsize=(8, 8), dpi=100)
    ax = fig.add_subplot(111, projection='3d')
    frames = []

    for i in range(horizon):
        scene.step()
        if i % 20 == 0:
            # Get positions and update scatter plot
            pos = scene.get_state()._solvers_state[5]._pos.cpu().numpy()
            ax.clear()  # Clear previous frame
            ax.scatter(pos[:, 0], pos[:, 1], pos[:, 2], c='red', s=1)

            # Capture frame directly to memory
            buf = io.BytesIO()
            fig.savefig(buf, format='png')
            buf.seek(0)
            frames.append(imageio.imread(buf))
            buf.close()

    plt.close()
    imageio.mimsave('simulation.gif', frames, fps=10)
```
</details>

barius, 2024-12-19:
> Lame. I spent a while trying to get this to run. This is the blocking error.
> 
> Should put a warning on the repo that it doesn't work on windows 11 currently

the compatibility matrix here on the [docs](https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html#prerequisites) does say that windows viewer is not supported. however adding camera is not supported also
[image]

barius, 2024-12-19:
> I met the same problem in Win10.
> 
> I found that in pyrender/numba_gl_wrapper.py/load_func, the variable `dll` pointed at 'C:\WINDOWS\system32\opengl32.dll', which we know that windows only support OpenGL until version 1.1, but what we need is OpenGL2.0 or later. so is there anyone know, how to change the default OpenGL version that `dll = GL.platform.PLATFORM.GL` pointed at?
> 
> Another interesting thing is, the following code can work:
> 
> ```shell
> python
> >>> from OpenGL.GL import glGetUniformLocation as a
> >>> from OpenGL.GL import glActiveTexture as b
> >>> a
> OpenGL.lazywrapper.lazy( 'glGetUniformLocation' )
> >>> b
> <OpenGL.platform.baseplatform.glActiveTexture object at 0x000002845FD4C550>
> ```
> 
> So I wonder if there's a way that we just import those function directly instead of load them from `load_func`?

the `OpenGL.lazywrapper.lazy( 'glGetUniformLocation' )` upon calling raises `NullFunctionError: Attempt to call an undefined function glGetUniformLocation, check for bool(glGetUniformLocation) before calling`, seems like the lazy wrapper can't find the symbol during runtime

Jamesyang-jpg, 2024-12-19:
same problem, GPU:nvidiaRTX3070ti 
C:\Users\32653\PycharmProjects\PythonProject\.venv\Scripts\python.exe C:\Users\32653\PycharmProjects\PythonProject\main.py 
[Genesis] [19:00:39] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [19:00:39] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [19:00:39] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [19:00:39] [INFO] Running on [Intel64 Family 6 Model 154 Stepping 3, GenuineIntel] with backend gs.cpu. Device memory: 31.68 GB.
[Genesis] [19:00:39] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [19:00:39] [INFO] Scene <1c2da54> created.
[Genesis] [19:00:39] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <c72e15b>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [19:00:39] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <ff6abd4>, morph: <gs.morphs.MJCF(file='C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [19:00:40] [INFO] Building scene <1c2da54>...
[Genesis] [19:00:52] [INFO] Compiling simulation kernels...
[Genesis] [19:01:07] [INFO] Building visualizer...
Traceback (most recent call last):
  File "C:\Users\32653\PycharmProjects\PythonProject\main.py", line 10, in <module>
    scene.build()
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
              ^^^^^^^^^^^
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\32653\PycharmProjects\PythonProject\.venv\Lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: function 'glGetUniformLocation' not found
[Genesis] [19:01:08] [INFO] 💤 Exiting Genesis and caching compiled kernels.

Diaode1, 2024-12-19:
I have made some attempts, personally think it should be the problem of PyOpenGL library, PyOpenGL installed directly on 64-bit windows by pip will lack dll files, this problem can be solved, so that PyOpenGL can run normally in other programs. However, the source code seems to use ctypes to load 32-bit dll files, 64-bit python will not find the file, but install 32-bit python will not be able to use "pip install genesis-world", perhaps on linux to solve these problems.

Linus-Civil, 2024-12-19:
same problem
using win11

wwdok, 2024-12-19:
same problem using win11
so which example in [here](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html) can we run on windows 11 ?

tyzero, 2024-12-19:
maybe should try `wglGetProcAddress` to get the opengl func ptr

DearVa, 2024-12-19:
[image]

Using glfw making it works now...

iadolgov, 2024-12-19:
> ![image](https://private-user-images.githubusercontent.com/55902546/397456402-7bfe46d6-1d5e-4ea9-8091-de1a98fe63cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Hello. It stops there.
[image]

Diaode1, 2024-12-20:
> ![image](https://private-user-images.githubusercontent.com/55902546/397456402-7bfe46d6-1d5e-4ea9-8091-de1a98fe63cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

In this way, although the image cannot be obtained directly at present, the particle position of some routines has actually been generated and can be visualized by plt
[image]
[image]
Test program code:
import torch
import matplotlib.pyplot as plt
import genesis as gs

all_particles = []

def main():
    ########################## init ##########################
    gs.init()

    ########################## create a scene ##########################

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            dt=4e-3,
            substeps=10,
        ),
        sph_options=gs.options.SPHOptions(
            lower_bound=(-0.5, -0.5, 0.0),
            upper_bound=(0.5, 0.5, 1),
            particle_size=0.01,
        ),
        vis_options=gs.options.VisOptions(
            visualize_sph_boundary=True,
        ),
        show_viewer=True,
    )

    ########################## entities ##########################
    plane = scene.add_entity(
        morph=gs.morphs.Plane(),
    )

    liquid = scene.add_entity(
        # viscous liquid
        # material=gs.materials.SPH.Liquid(mu=0.02, gamma=0.02),
        material=gs.materials.SPH.Liquid(),
        morph=gs.morphs.Box(
            pos=(0.0, 0.0, 0.65),
            size=(0.4, 0.4, 0.4),
        ),
        surface=gs.surfaces.Default(
            color=(0.4, 0.8, 1.0),
            vis_mode='particle',
        ),
    )

    ########################## build ##########################
    scene.build()

    gs.tools.run_in_another_thread(fn=run_sim, args=(scene, False, liquid))

    #scene.viewer.start()

def run_sim(scene, enable_vis, liquid):
    global all_particles
    i = 0
    while True:
        i += 1

        scene.step()
        particles = liquid.get_particles()
        # print(particles)
        '''
        [-4.8954931e-01  3.9647943e-01  2.0921452e-02]
        [-2.9889485e-01 -3.5985804e-01  2.3578152e-03]
        [-3.0421656e-01  5.0000000e-01  9.5947198e-06]
        '''

        all_particles.append(particles)
        if i > 10:
            break

    if enable_vis:
        scene.viewer.stop()

main()
while True:
    if len(all_particles) == 10:
        for particles in all_particles:
            fig = plt.figure()
            ax = fig.add_subplot(111, projection='3d')
            ax.scatter(particles[:, 0], particles[:, 1], particles[:, 2])
            ax.set_xlabel('x')
            ax.set_ylabel('y')
            ax.set_zlabel('z')
            plt.show()

        plt.show()

hougd7, 2024-12-20:
same problem here

hougd7, 2024-12-20:
> ![image](https://private-user-images.githubusercontent.com/55902546/397456402-7bfe46d6-1d5e-4ea9-8091-de1a98fe63cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Still have the problem

chenfei8888, 2024-12-20:
same problem here

lakeheck, 2024-12-20:
same issue with win 11, cuda 12.4, rtx4090 - if the camera/viewer is the part unsupported on windows, how can I run a simulation using genesis and access the data (via spout lookup to the same gpu memory address, ideally) so that I can render it in a separate process? all the example code for simulations (SPH, etc) includes the scene.build() line, which is what triggers the aforementioned issue. how should we run a simulation without using scene.build()?

supraxylon, 2024-12-20:
Could we rely on PyOpenGL’s Automatic Loading? Based on an earlier comment 
A straightforward “fix” in Python is to rely on PyOpenGL to load extensions for you. Once you’ve created a modern OpenGL context (for example, using GLFW and specifying a minimum core profile version like 3.3), you can:

import glfw
from OpenGL.GL import glGetUniformLocation, glUseProgram, ...

# Create a window and context
glfw.init()
glfw.window_hint(glfw.CONTEXT_VERSION_MAJOR, 3)
glfw.window_hint(glfw.CONTEXT_VERSION_MINOR, 3)
glfw.window_hint(glfw.OPENGL_PROFILE, glfw.OPENGL_CORE_PROFILE)
window = glfw.create_window(800, 600, "OpenGL Window", None, None)
glfw.make_context_current(window)

# Now PyOpenGL will be able to load modern functions
location = glGetUniformLocation(some_program, b"uniformName")

Under the hood, PyOpenGL uses the current context and calls wglGetProcAddress to load entry points for functions not found in opengl32.dll. This typically means you don’t have to do any manual symbol loading yourself, as PyOpenGL mimics what GLEW does in C/C++.

Another option is to get the pointers manually Using wglGetProcAddress with ctypes
If you really want to replicate GLEW-like behavior manually:

First, ensure you have a valid, current OpenGL context created by something like GLFW or any other windowing toolkit that can provide a modern context.

Import the necessary functions and DLLs:

import ctypes

# Load the system OpenGL library
opengl32 = ctypes.windll.opengl32

# Get the wglGetProcAddress function
wglGetProcAddress = opengl32.wglGetProcAddress
wglGetProcAddress.restype = ctypes.c_void_p
wglGetProcAddress.argtypes = [ctypes.c_char_p]

# After making the context current:
func_ptr = wglGetProcAddress(b"glGetUniformLocation")
if not func_ptr:
    raise RuntimeError("Failed to load glGetUniformLocation!")

Once you have the function pointer, you convert it into a callable Python function using ctypes.CFUNCTYPE. For example:

# Define a Python signature for glGetUniformLocation
GLGetUniformLocationType = ctypes.CFUNCTYPE(
    ctypes.c_int,     # return type GLint
    ctypes.c_uint,    # GLuint program
    ctypes.c_char_p   # const GLchar *name
)

glGetUniformLocation = GLGetUniformLocationType(func_ptr)

# Now you can call glGetUniformLocation just like a normal Python function
location = glGetUniformLocation(program, b"myUniform")

This mimics what GLEW does in C/C++: querying the driver for a function pointer and then wrapping it for convenient use.

A valid context is needed because  wglGetProcAddress only works when there’s a current OpenGL context that supports the requested function. If you call it before making the context current or if you have an incompatible context (e.g., too old), it will return None.

Maybe I'm misunderstanding the implementation here but why can't we just use a more modern method to load PyOpenGL. PyOpenGL’s extension loading process is very similar to what GLEW does, but it’s done automatically after you create a modern context and import from OpenGL.GL. That’s why the real fix is usually just to create a proper modern OpenGL context, then rely on PyOpenGL’s from OpenGL.GL import ... approach instead of manually loading symbols.

Can anyone clarify why we can't just do something like: 
import OpenGL.GL as GL
from OpenGL.GL import glGetUniformLocation

Instead of manually loading?

voidiii, 2024-12-22:
has anyone tried running it on WSL?

voidiii, 2024-12-22:
> Can be workarounded using [MESA](https://www.mesa3d.org/).
> 
> ![image](https://private-user-images.githubusercontent.com/12647063/397752140-b7fe2f52-642a-4db1-90dd-d35e5d07d05d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4MzU2NzMsIm5iZiI6MTczNDgzNTM3MywicGF0aCI6Ii8xMjY0NzA2My8zOTc3NTIxNDAtYjdmZTJmNTItNjQyYS00ZGIxLTkwZGQtZDM1ZTVkMDdkMDVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIyVDAyNDI1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkNGRhOWViNTFmZWUzNTIzMWIxZThiMjVmNDQ5ZTIxMDRmZDZjNjNlNWM1ZTI2ZDZhZDJjZTE1NmI2MzU3OTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ljl2sO4ieJhXKvUoC3GorcnTpHaHhm-htl7Ts7wcn-4)
> 
> **Short answer**
> 
> * Download Mesa OpenGL32.dll from here: https://github.com/lightningterror/mesa3D-Windows/releases (not the official source, but avoid compiling it by yourself).
> * Put it so that it is found before Windows/System32/openGL32.dll. If you use python venv, you can put it in venv/script.
> * Comment the `gs.raise_exception` in `venv\Lib\site-packages\genesis\vis\viewer.py line 67`
> 
> **Long answer**
> 
> OpenGL is not strickly speaking a library. It is a spec which is implemented by some other parties, especially by GPU vendors in the driver. Historically, Windows has implemented this library at the begining, but stopped at OpenGL 1.2. That's why glGetUniformLocation symbole is not present in windows's OpenGL.dll as glGetUniformLocation has been introducedd in OpenGL 2.x
> 
> [MESA](https://www.mesa3d.org/) is a CPU implementation of OpenGL spec. Meaning, it defines and implement all the symboles but the code will be executed on CPU, instead of GPU. Meaning performences will lower.
> 
> Real fix would be to load OpenGL symboles from the GPU driver instead of Windows OpenGL32.dll. In C/C++ this is done typically using [GLEW](https://github.com/nigels-com/glew). However, I don't know how to proceed in python.

I tried this method and I also had to change the name of the `opengl32.dll` under `C:\Windows\System32` to make the code load the MESA dll, and it is still now showing a viewer or anything just the fps

Jup33Q, 2024-12-22:
> has anyone tried running it on WSL?

you mean using  a docker image supports opengl?

ldepn, 2024-12-23:
> I tried this method and I also had to change the name of the `opengl32.dll` under `C:\Windows\System32` to make the code load the MESA dll, and it is still now showing a viewer or anything just the [...]

 I tried this method and I also had to change the name of the `opengl32.dll` under  venv/script to make the code load the MESA dll, and it show the fps but still not showing a viewer.

ssm1984119, 2024-12-23:
> I tried this method and I also had to change the name of the `opengl32.dll` under venv/script to make the code load the MESA dll, and it show the fps but still not showing a viewer.

I have the same problem.Follow all the steps,run successfully,but have no viewer shown.

voidiii, 2024-12-23:
> > has anyone tried running it on WSL?
> 
> you mean using a docker image supports opengl?

No just WSL the comes with Windows. I have been using it for my own project and it was working nicely as a linux system. However, it doesn't work on Genesis.

borokov, 2024-12-23:
> I tried this method and I also had to change the name of the `opengl32.dll` under venv/script to make the code load the MESA dll, and it show the fps but still not showing a viewer.

Yes, same here. 

I tried the [2nd example](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html) but ended up in a deadlock in render() method. After digging, it seems on non unix, because of threading issues, you should use [this method](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py) and manually create a render thread.

I scratch the code a bit to try to make it works on Windows but always end up with that missing glGetUniform error. It seems Genesis need a small rework on how to handle OpenGL, like it was suggested by @supraxylon .

borokov, 2024-12-23:
> same issue with win 11, cuda 12.4, rtx4090 - if the camera/viewer is the part unsupported on windows, how can I run a simulation using genesis and access the data (via spout lookup to the same gpu [...]

You can try the MESA workaround I described. Rendering doesn't work but it seems simulation is running. You may manage to get the result of simulation and inject it in another renderengine.

win10ogod, 2024-12-24:
maby can try Vulkan

Jup33Q, 2024-12-25:
thanks you all

craigdsouza, 2024-12-29:
I’m on Windows 11 using genesis-world (v0.2.0). When I try `[hello genesis](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html)` set show_viewer=True in scene = gs.Scene(...), I get an “AttributeError: function ‘glGetUniformLocation’ not found.” My code runs the CPU-based physics fine, but OpenGL visualization fails.

I tried several fixes after researching and getting help from o1:

1. Installing PyOpenGL-accelerate and verifying pyopengl version (3.1.7).
2. Setting PYOPENGL_PLATFORM=glfw or win32 before running quickstart.py. This removed the AttributeError but left the viewer window blank or “Not Responding.”
3. Creating my own GLFW window and calling glfw.make_context_current(window) didn’t stop the blank window. Possibly, Genesis uses a different thread or never pumps window events??
4. Adding time.sleep or manually swapping buffers only led to a blank white/black “Not Responding” window.
5. Trying headless mode (show_viewer=False) works without errors, though it warns that headless rendering is unsupported on Windows. It shows “Running at XXX FPS” while just doing CPU steps.

It looks like genesis-world doesn’t fully implement a Windows viewer loop (e.g., no event polling), so OpenGL calls don’t update the window.

Only after trying out all the steps above did I see that this is in the project roadmap and help is needed. **:-)**  I’m happy to assist if someone can point me in the right direction. I’m not an OpenGL or rendering expert, but I would love to help, particularly with the Windows Interactive Viewer.

DTSwink, 2024-12-29:
i created a discord so ppl can figure out together how to do it : https://discord.gg/cUGScAKVnt

Jup33Q, 2024-12-30:
> I’m on Windows 11 using genesis-world (v0.2.0). When I try `[hello genesis](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html)` set show_viewer=True in [...]

thanks, I finally managed to run them in ubuntu desktop since that is quite straightforward way to do so

Jup33Q, 2024-12-30:
well, I don't think using cpu 'always'  as a render unit is a good behavior in such parallel-ish consuming project

charlesr1971, 2024-12-30:
> show_viewer=False

@Jup33Q Thanks for all the effort you have made. I am a technical dev with 20+ years experience in _Coldfusion_, _Java_, _NodeJs_, _Objective-C_ & _JavaScript_. Unfortunately, my _Python_ is extremely basic. So, I cannot really help here. 

I really hope that you can contribute to this project and find a solution to this rendering issue for Windows users. 

In the meantime, where do I add:

```
show_viewer=False
```

charlesr1971, 2024-12-30:
> 5\. Trying headless mode (show_viewer=False) works without errors, though it warns that headless rendering is unsupported on Windows. It shows “Running at XXX FPS” while just doing CPU steps.

OK. So this does remove the errors but no rendering is done:

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Install the latest PowerShell for new features and improvements! https://aka.ms/PSWindows

PS C:\Users\cdesi> cd C:\Users\cdesi\Desktop\work\genesis_world\python\projects\genesis_world
PS C:\Users\cdesi\Desktop\work\genesis_world\python\projects\genesis_world> python .\genesis_world.py
[Genesis] [15:48:31] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [15:48:31] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [15:48:31] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [15:48:31] [INFO] Running on [Intel64 Family 6 Model 154 Stepping 3, GenuineIntel] with backend gs.cpu. Device memory: 31.68 GB.
[Genesis] [15:48:31] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [15:48:32] [WARNING] Headless rendering not yet supported on Windows.
[Genesis] [15:48:32] [INFO] Scene <7319f94> created.
[Genesis] [15:48:32] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <478eff7>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [15:48:32] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <40213cc>, morph: <gs.morphs.MJCF(file='C:\Users\cdesi\AppData\Local\Programs\Python\Python311\Lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [15:48:32] [INFO] Building scene <7319f94>...
[Genesis] [15:48:45] [INFO] Compiling simulation kernels...
[Genesis] [15:49:03] [INFO] Building visualizer...
[Genesis] [15:49:03] [INFO] Running at 20833.33 FPS.
...
[Genesis] [15:49:04] [INFO] Running at 1152.14 FPS.
[Genesis] [15:49:04] [INFO] 💤 Exiting Genesis and caching compiled kernels...
PS C:\Users\cdesi\Desktop\work\genesis_world\python\projects\genesis_world>
```

Jup33Q, 2024-12-31:
> @Jup33Q Thanks for all the effort you have made. I am a technical dev with 20+ years experience in _Coldfusion_, _Java_, _NodeJs_, _Objective-C_ & _JavaScript_. Unfortunately, my _Python_ is [...]

So do I, it really help if there is any integration version with in touchdesigner or blender, that really helps

Jup33Q, 2024-12-31:
> @Jup33Q Thanks for all the effort you have made. I am a technical dev with 20+ years experience in _Coldfusion_, _Java_, _NodeJs_, _Objective-C_ & _JavaScript_. Unfortunately, my _Python_ is [...]

I wish there is any team working on a touchdesigner or blender version of those

charlesr1971, 2024-12-31:
> I wish there is any team working on a touchdesigner or blender version of those

In fact, integrating _Genesis_ into _Blender_ would be a match made in heaven.🙂
Both, are created in _Python_, and _Genesis_ could be added to the _Blender_ core. 
I might see if I can post this thread on the _Blender Community Forum_?
I am sure the _Blender Core Team_ would be able sort this out.

charlesr1971, 2024-12-31:
@Jup33Q I have just added a post, referencing this matter, on the _Blender_ **Stack Exchange**. 
Maybe someone from the _Blender Core Team_, will see it and offer to help?

https://blender.stackexchange.com/questions/329107/can-the-blender-core-team-help-to-get-genesis-world-physics-engine-working-on-wi

Can everyone jump onto this post, add comments, so that we can generate some exposure on **Stack Exchange**.

charlesr1971, 2025-01-16:
> I've implemented a temporary solution for running the viewer on Windows in this repository: https://github.com/kohya-ss/Genesis/tree/windows-renderer. While I'm not an expert in OpenGL and cannot guarantee the correctness of this implementation, I hope it might be useful as a reference.

Fantastic! Thank you for taking the time to resolve this. 
What is the best way to install your changes?

Teloivts, 2025-01-17:
> I've implemented a temporary solution for running the viewer on Windows in this repository: https://github.com/kohya-ss/Genesis/tree/windows-renderer. While I'm not an expert in OpenGL and cannot guarantee the correctness of this implementation, I hope it might be useful as a reference.

Thanks! It works!

charlesr1971, 2025-01-17:
> I think you can install the forked version with commands like this:
> 
> ```
> git clone https://github.com/kohya-ss/Genesis.git
> cd Genesis
> git checkout windows-renderer
> git branch
> pip install -e .
> ```
> 
> Please note that you must check out `windows-renderer` branch.
> 
> If you already have a Genesis directory, the commands would look something like this:
> 
> ```
> git clone https://github.com/kohya-ss/Genesis.git Genesis-win
> cd Genesis-win
> ```
> 
> > Do you recommend that I should first uninstall the official version @ https://github.com/Genesis-Embodied-AI/Genesis and then install your fork?
> 
> I'm not sure, but I think it's better that way. That's what I did.
> 
> EDIT: I've added a simple installation guide to the README in that branch.

It worked. 
Fantastic job! 

Thank you so much :)

[image]

Genesis-Embodied-AI, 2025-01-18:
This is fantastic! @ziyanx02 could you take a look and merge into main if all look good?

ziyanx02, 2025-01-18:
Thank you @kohya-ss !!! I've submitted a PR (the same as your modifications) in advance for others to review. We would be happy if you could start a new PR to contribute directly to this repository!

closed 2025-02-22 by YilingQiao

weebao, 2025-04-12:
```bash
[Genesis] [20:22:22] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [20:22:22] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [20:22:22] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [20:22:22] [INFO] Running on [AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD] with backend gs.cpu. Device memory: 13.86 GB.
[Genesis] [20:22:23] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [20:22:23] [INFO] Scene <27a2e5d> created.
[Genesis] [20:22:23] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <99dd642>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [20:22:23] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <0c972dd>, morph: <gs.morphs.MJCF(file='C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [20:22:24] [INFO] Preprocessing geom idx 1.
[Genesis] [20:22:25] [INFO] Preprocessing geom idx 2.
[Genesis] [20:22:26] [INFO] Preprocessing geom idx 3.
[Genesis] [20:22:26] [INFO] Preprocessing geom idx 4.
[Genesis] [20:22:27] [INFO] Preprocessing geom idx 5.
[Genesis] [20:22:27] [INFO] Preprocessing geom idx 6.
[Genesis] [20:22:27] [INFO] Preprocessing geom idx 7.
[Genesis] [20:22:27] [INFO] Preprocessing geom idx 8.
[Genesis] [20:22:28] [INFO] Preprocessing geom idx 9.
[Genesis] [20:22:28] [INFO] Preprocessing geom idx 10.
[Genesis] [20:22:28] [INFO] Preprocessing geom idx 11.
[Genesis] [20:22:28] [INFO] Preprocessing geom idx 12.
[Genesis] [20:22:28] [INFO] Preprocessing geom idx 13.
[Genesis] [20:22:28] [INFO] Preprocessing geom idx 14.
[Genesis] [20:22:28] [INFO] Preprocessing geom idx 16.
[Genesis] [20:22:28] [INFO] Building scene <27a2e5d>...
[W 04/11/25 20:22:29.709 20120] [frontend_ir.cpp:taichi::lang::ASTBuilder::begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 04/11/25 20:22:31.776 20120] [frontend_ir.cpp:taichi::lang::ASTBuilder::begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 04/11/25 20:22:53.501 20120] [frontend_ir.cpp:taichi::lang::ASTBuilder::begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [20:22:53] [INFO] Compiling simulation kernels...
[Genesis] [20:24:26] [INFO] Building visualizer...
Traceback (most recent call last):
File "C:\Users\user\Desktop\Codes\random stuff\test-g1\what.py", line 10, in <module>
    scene.build()
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\engine\scene.py", line 607, in build
    self._visualizer.build()
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
              ^^^^^^^^^^^
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\miniconda3\envs\ella\Lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: function 'glGetUniformLocation' not found
```

Is it supported on Windows yet? It seems to be on the installation page but I keep getting the same errors here

duburcqa, 2025-04-20:
You should install the latest version (which is not available on PyPI yet) to get full Windows OS support. The instructions are provided in README.

weebao, 2025-04-21:
I still get the following errors when rendering the scene
`PYOPENGL_PLATFORM = 'osmesa'`
```
rms/osmesa.py", line 19, in init_context
    from OpenGL.osmesa import (
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/OpenGL/osmesa/__init__.py", line 2, in <module>
    from OpenGL.raw.osmesa.mesa import *
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/OpenGL/raw/osmesa/mesa.py", line 32, in <module>
    @_f
     ^^
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/OpenGL/raw/osmesa/mesa.py", line 9, in _f
    function,_p.PLATFORM.OSMesa,
             ^^^^^^^^^^^^^^^^^^
AttributeError: 'EGLPlatform' object has no attribute 'OSMesa'
```

`PYOPENGL_PLATFORM = 'egl'`
```
File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/pyrender/offscreen.py", line 137, in _create
egl_device = egl.get_device_by_index(device_id)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/pyrender/platforms/egl.py", line 83, in get_device_by_index
raise ValueError('Invalid device ID ({})'.format(device_id, len(devices)))
ValueError: Invalid device ID (0)
```

I'm guessing this is more of a Pyrender issue than a Genesis issue but I think Genesis doesn't work quite well on my Windows machine yet

duburcqa, 2025-04-21:
First of all, specify `PYOPENGL_PLATFORM` should not help because all the available OpenGL backend on a given platform are tried automatically. Next, 'osmesa' is not available by default on Windows OS, so the error you are getting is expected. Regarding the second error, it seems that the environment variable `EGL_DEVICE_ID` has been specified.

duburcqa, 2025-04-21:
After checking, it seems that you are still using the old version of Genesis, that does not support Windows OS properly. All I said previously was only applicable for the latest version (ie the main branch of this repo. As I said, you should install latest version by following the readme instructions:
```
pip install git+https://github.com/Genesis-Embodied-AI/Genesis.git
```

duburcqa, 2025-04-21:
After checking twice, it may be something else. I don't know why but you are apparently using the official version of `pyrender` (`/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/pyrender/offscreen.py`). Genesis is shipping its own version of `pyrender`, which has been heavily customised and can be considered as a completely different fork at this point.

weebao, 2025-04-21:
That's weird. I will try restarting and do everything back from scratch to see if there's an issue. Thanks a lot for your time!

weebao, 2025-04-22:
I tried again by installing the latest version of the repo and this is what I got.
```
Traceback (most recent call last):
  File "/mnt/c/Users/baoro/Desktop/Codes/research/Ella/Genesis/sandbox/render_test/pyrender_test.py", line 4, in <module>
    import pyrender
ModuleNotFoundError: No module named 'pyrender'
```
How do I install the customized pyrender from `Genesis/genesis/ext/pyrender`?

weebao, 2025-04-23:
After trying again, it produces a different error:
```bash
libEGL warning: MESA-LOADER: failed to open vgem: /usr/lib/dri/vgem_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

libEGL warning: NEEDS EXTENSION: falling back to kms_swrast
libEGL warning: MESA-LOADER: failed to open kms_swrast: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

libEGL warning: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

libEGL warning: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

Traceback (most recent call last):
  File "/mnt/c/Users/baoro/Desktop/Codes/research/ella/ViCo/modules/robot/robot_test_program.py", line 170, in <module>
    scene.build()
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/genesis/utils/misc.py", line 62, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/genesis/engine/scene.py", line 605, in build
    self._visualizer.build()
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/genesis/vis/visualizer.py", line 149, in build
    self._rasterizer.build()
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/genesis/vis/rasterizer.py", line 29, in build
    self._renderer = pyrender.OffscreenRenderer(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/genesis/ext/pyrender/offscreen.py", line 29, in __init__
    self._create(pyopengl_platform)
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/genesis/ext/pyrender/offscreen.py", line 204, in _create
    self._platform.init_context()
  File "/home/baoro/miniconda3/envs/ella/lib/python3.11/site-packages/genesis/ext/pyrender/platforms/egl.py", line 214, in init_context
    assert eglInitialize(egl_display, major, minor)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

[Genesis] [13:39:08] [ERROR] EGLError: EGLError(
        err = EGL_NOT_INITIALIZED,
        baseOperation = eglInitialize,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x7f6159fdef00>,
                c_long(0),
                c_long(0),
        ),
        result = 0
)
```

Is this Windows-only?

[7 more replies]