Genesis-Embodied-AI/genesis-world #14: Smoke Example

Bardia323, 2024-12-19:
```
(venv) PS S:\Tools\genesis> py .\smoke.py
[Taichi] version 1.7.2, llvm 15.0.1, commit 0131dce9, win, python 3.10.5
[Genesis] [21:47:00] [INFO] Running on [NVIDIA GeForce RTX 3090 Ti] with backend gs.cuda. Device memory: 23.99 GB.
[Genesis] [21:47:01] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [21:47:01] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "S:\Tools\genesis\smoke.py", line 140, in <module>
    main()
  File "S:\Tools\genesis\smoke.py", line 87, in main
    sf_options=gs.options.SFOptions(
  File "S:\Tools\genesis\venv\lib\site-packages\genesis\options\options.py", line 27, in __init__
    gs.raise_exception(f"Unrecognized attribute: {key}")

[Genesis] [21:47:01] [ERROR] Unrecognized attribute: res
[Genesis] [21:47:01] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

The smoke example throws the following error:
**[Genesis] [21:47:01] [ERROR] Unrecognized attribute: res**

zhouxian, 2024-12-19:
@PingchuanMa can you take a look?

PingchuanMa, 2024-12-19:
Yo thanks for trying out the latest feature. The smoke example only works in the main branch though and hasn't been pushed to pypi yet. If you want to try it now please build genesis from scratch by `pip install -e .`, or just simply wait for the next push to pypi ;)

Bardia323, 2024-12-19:
sick! thanks for the prompt response. I'm gonna pull latest and rebuild.

closed 2024-12-19 by Bardia323

huiwenzhang, 2024-12-20:
> Yo thanks for trying out the latest feature. The smoke example only works in the main branch though and hasn't been pushed to pypi yet. If you want to try it now please build genesis from scratch by `pip install -e .`, or just simply wait for the next push to pypi ;)

Dos this repo supports the editable pip install now? Since I didn't see the `setup.py` script is included in the repo.

PingchuanMa, 2024-12-20:
> Dos this repo supports the editable pip install now? Since I didn't see the `setup.py` script is included in the repo.

We followed the modern practice using `pyproject.toml` [[1]](https://peps.python.org/pep-0621/).