Genesis-Embodied-AI/genesis-world #16: Weird Humanoid Performance

zengweishuai, 2024-12-19:
I'm more than excited to see this GREAT project and I give it a shot the moment I see it.

During my trial, I find something weird when I load the URDF of a Unitree H1 robot. To be specific, the movement of legs seems kind of unnatural. I wonder if this phenomenon is normal or if I make some mistakes.

The code I've tried is the same as you give for the Franka arm:
```Python
import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='h1.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```

The performance the robot shows is:
[image]

zhouxian, 2024-12-19:
It looks normal to me?

zengweishuai, 2024-12-19:
> It looks normal to me?

Thanks for your quick reply! But why the leg get suspended in the air when it falls down to the ground. This bewilders me qwq.

ziyanx02, 2024-12-22:
I guess the collision mesh of hip joints causes it.
I don't consider this performance to be a simulation-level issue. If you get any other problems, please feel free to open a new issue.

closed 2024-12-22 by ziyanx02