Genesis-Embodied-AI/genesis-world #24: Validation failed: Samplers of different types use the same texture image unit

YutoYasunaga, 2024-12-19:
I got this error on MacBook Pro M1.

> RuntimeError: Validation failure (0): b"Validation Failed: Sampler error:\n  Samplers of different types use the same texture image unit.\n   - or -\n  A sampler's texture unit is out of range (greater than max allowed or negative).\n"

Command: `python examples/render_on_macos.py -v`

Error detail:

```
[Genesis] [14:52:30] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [14:52:30] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [14:52:30] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [14:52:30] [INFO] Running on [Apple M1 Max] with backend gs.cpu. Device memory: 32.00 GB.
[Genesis] [14:52:30] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [14:52:31] [INFO] Scene <814c494> created.
[Genesis] [14:52:31] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <812fd9d>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [14:52:31] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <aa3e078>, morph: <gs.morphs.MJCF(file='/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [14:52:31] [INFO] Building scene <814c494>...
[Genesis] [14:52:35] [INFO] Compiling simulation kernels...
[Genesis] [14:52:39] [INFO] Building visualizer...
[Genesis] [14:52:40] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`.
[Genesis] [14:52:40] [INFO] Viewer created. Resolution: 744×558, max_FPS: 60.
[Genesis] [14:52:40] [INFO] Running at 4481.21 FPS.
219.87334871042148 FPS
[Genesis] [14:52:40] [INFO] Running at 812.41 FPS.
48.83798700542605 FPS
[Genesis] [14:52:40] [INFO] Running at 483.75 FPS.
55.46920584540104 FPS
[Genesis] [14:52:40] [INFO] Running at 334.56 FPS.
48.598058072439926 FPS
[Genesis] [14:52:40] [INFO] Running at 119.78 FPS.
9.061163823646325 FPS
[Genesis] [14:52:40] [INFO] Running at 111.76 FPS.
49.01949417979524 FPS
[Genesis] [14:52:40] [INFO] Running at 104.99 FPS.
48.62904777915618 FPS
[Genesis] [14:52:40] [INFO] Running at 100.67 FPS.
56.17647295179674 FPS
[Genesis] [14:52:40] [INFO] Running at 95.62 FPS.
48.74943629559032 FPS
uniform not found: V
uniform not found: P
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: V
uniform not found: P
uniform not found: M
uniform not found: V
uniform not found: P
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
uniform not found: M
Traceback (most recent call last):
  File "/Users/yuto/Downloads/Genesis/examples/render_on_macos.py", line 68, in <module>
    main()
  File "/Users/yuto/Downloads/Genesis/examples/render_on_macos.py", line 44, in main
    scene.viewer.start()
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/vis/viewer.py", line 91, in start
    self._pyrender_viewer.start()
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 386, in start
    self._init_and_start_app()
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/Users/yuto/.local/lib/python3.11/site-packages/pyglet/clock.py", line 633, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/Users/yuto/.local/lib/python3.11/site-packages/pyglet/clock.py", line 348, in tick
    self.call_scheduled_functions(delta_t)
  File "/Users/yuto/.local/lib/python3.11/site-packages/pyglet/clock.py", line 292, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 168, in render
    retval = self._forward_pass(scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 534, in _forward_pass
    self.jit.forward_pass(
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/jit_render.py", line 708, in forward_pass
    self.load_programs(renderer, flags, program_flags)
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/jit_render.py", line 349, in load_programs
    program = renderer._get_primitive_program(primitive, flags, program_flags)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 1142, in _get_primitive_program
    program._add_to_context()
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/genesis/ext/pyrender/shader_program.py", line 125, in _add_to_context
    self._program_id = gl_shader_utils.compileProgram(*shader_ids, validate=False)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/OpenGL/GL/shaders.py", line 196, in compileProgram
    program.check_validate()
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/OpenGL/GL/shaders.py", line 105, in check_validate
    raise RuntimeError(
RuntimeError: Validation failure (0): b"Validation Failed: Sampler error:\n  Samplers of different types use the same texture image unit.\n   - or -\n  A sampler's texture unit is out of range (greater than max allowed or negative).\n"
^CException ignored in: <module 'threading' from '/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/threading.py'>
Traceback (most recent call last):
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/threading.py", line 1583, in _shutdown
    lock.acquire()
```

btx0424, 2024-12-19:
Same problem here on Ubuntu 22.04 with PyOpenGL==3.1.0.

RobRoyce, 2024-12-19:
Seems to be the same underlying issue as #23

btx0424, 2024-12-19:
Oh, I solved it by upgrading PyOpenGL to 3.1.7.

RobRoyce, 2024-12-19:
> Oh, I solved it by upgrading PyOpenGL to 3.1.7.

Nice find! Can confirm this also solved it for me on M1 MBP

```pip install -U PyOpenGL```

YutoYasunaga, 2024-12-19:
> Oh, I solved it by upgrading PyOpenGL to 3.1.7.

Thank you! 
Confirm this solved.

But now I have next issue #11 😅

closed 2024-12-20 by YutoYasunaga