Genesis-Embodied-AI/genesis-world #26: Genesis fails silently when ffmpeg is not installed

RobRoyce, 2024-12-19:
Great project, really excited to give it a shot!

I am running Genesis on a MacBook Pro M1 Max. The installation process went fine, but I immediately started getting errors. In particular, this try/except in [rasterizer_context.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/6d719541c29da17d35043d5b2bf2fc5ffaad7429/genesis/vis/rasterizer_context.py#L8-L12) is causing issues. 

### Initial Error 
Initially, the error I get is that `trimesh` is missing:

```
~/Developer/Genesis git:[main]
python3.11 scripts/main.py
objc[21098]: Class GLFWHelper is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x15baf4570) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x318dd17c0). One of the two will be used. Which one is undefined.
objc[21098]: Class GLFWApplicationDelegate is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x15baf45c0) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x318dd1810). One of the two will be used. Which one is undefined.
objc[21098]: Class GLFWWindowDelegate is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x15baf45e8) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x318dd1838). One of the two will be used. Which one is undefined.
objc[21098]: Class GLFWContentView is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x15baf4638) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x318dd1888). One of the two will be used. Which one is undefined.
objc[21098]: Class GLFWWindow is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x15baf46b0) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x318dd1900). One of the two will be used. Which one is undefined.
Failed to import pyrender. Rendering will not work.
[Genesis] [21:59:57] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [21:59:57] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [21:59:57] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [21:59:57] [INFO] Running on [Apple M1 Max] with backend gs.cpu. Device memory: 32.00 GB.
[Genesis] [21:59:57] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/Users/rob/Developer/Genesis/scripts/main.py", line 21, in <module>
    main()
  File "/Users/rob/Developer/Genesis/scripts/main.py", line 8, in main
    scene = gs.Scene(show_viewer=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/genesis/vis/visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/genesis/vis/rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
                           ^^^^^^^
NameError: name 'trimesh' is not defined
[Genesis] [21:59:58] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

So I simply modified the `except` statement to import the system version of trimesh (e.g. `import trimesh` instead of `from genesis.ext import trimesh`.

### Next Error 
Next, I get an error that `pyrender` is missing:

```
~/Developer/Genesis git:[main]
python3.11 scripts/main.py
objc[22903]: Class GLFWHelper is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x143af4570) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30e2b57c0). One of the two will be used. Which one is undefined.
objc[22903]: Class GLFWApplicationDelegate is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x143af45c0) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30e2b5810). One of the two will be used. Which one is undefined.
objc[22903]: Class GLFWWindowDelegate is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x143af45e8) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30e2b5838). One of the two will be used. Which one is undefined.
objc[22903]: Class GLFWContentView is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x143af4638) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30e2b5888). One of the two will be used. Which one is undefined.
objc[22903]: Class GLFWWindow is implemented in both /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x143af46b0) and /Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30e2b5900). One of the two will be used. Which one is undefined.
Failed to import pyrender. Rendering will not work.
[Genesis] [22:02:38] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [22:02:38] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [22:02:38] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [22:02:38] [INFO] Running on [Apple M1 Max] with backend gs.cpu. Device memory: 32.00 GB.
[Genesis] [22:02:38] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [22:02:38] [INFO] Scene <3e94a8b> created.
[Genesis] [22:02:38] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <8034f3b>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [22:02:38] [INFO] Preprocessing geom idx 0.
[Genesis] [22:02:38] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <0bbac21>, morph: <gs.morphs.MJCF(file='/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [22:02:39] [INFO] Preprocessing geom idx 1.
[Genesis] [22:02:39] [INFO] Preprocessing geom idx 2.
[Genesis] [22:02:40] [INFO] Preprocessing geom idx 3.
[... 61 lines cut ...]
    from moviepy.audio.fx.AudioDelay import AudioDelay
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/moviepy/audio/fx/AudioDelay.py", line 5, in <module>
    from moviepy.audio.AudioClip import CompositeAudioClip
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/moviepy/audio/AudioClip.py", line 13, in <module>
[... cut ...]
RuntimeError: No ffmpeg exe could be found. Install ffmpeg on your system, or set the IMAGEIO_FFMPEG_EXE environment variable.
>>> exit()
```

### The Solution
I install `ffmpeg` using Homebrew and finally it works!

So, to summarize, the `try/except` in `rasterize_context.py` is silently failing when `ffmpeg` is not installed, and the resulting errors that get produced are very misleading. It requires quite a bit of digging to figure out that you are missing `ffmpeg`. I will submit a PR to at least notify the user when this happens.

Cheers! 🚀

closed 2024-12-19 by zswang666

analogjedi, 2024-12-20:
@RobRoyce , thank you for the issue and write up. I discovered the same thing independently this morning while trying to get the "getting started" code running on my M3 Max. I ended up installing ffmpeg with Brew as well.

I am about to file an issue for this code snippet. I am getting Open GL errors that I haven't been able to resolve yet. Does this code run for you?

import argparse

import torch

import genesis as gs

def main():

    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(backend=gs.cpu)

    ########################## create a scene ##########################

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(3.5, 0.0, 2.5),
            camera_lookat=(0.0, 0.0, 0.5),
            camera_fov=40,
        ),
        show_viewer=args.vis,
        rigid_options=gs.options.RigidOptions(
            dt=0.01,
            gravity=(0.0, 0.0, -10.0),
        ),
    )

    ########################## entities ##########################
    plane = scene.add_entity(gs.morphs.Plane())
    r0 = scene.add_entity(
        gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
    )

    ########################## build ##########################
    scene.build()

    gs.tools.run_in_another_thread(fn=run_sim, args=(scene, args.vis))
    if args.vis:
        scene.viewer.start()

def run_sim(scene, enable_vis):
    from time import time

    t_prev = time()
    i = 0
    while True:
        i += 1

        scene.step()

        t_now = time()
        print(1 / (t_now - t_prev), "FPS")
        t_prev = t_now
        if i > 200:
            break

    if enable_vis:
        scene.viewer.stop()

if __name__ == "__main__":
    main()

VincentCCandela, 2024-12-29:
SOLUTION:

1) install ffmpeg on your system
2) put this in your main runtime to make sure that your code points correctly to your ffmpeg binary

import os
os.environ["IMAGEIO_FFMPEG_EXE"] = "/usr/bin/ffmpeg"

commit ac760c54 (2024-12-19): fix: silent failure when ffmpeg is not installed. Resolves #26