Genesis-Embodied-AI/genesis-world #30: module 'genesis' has no attribute 'generate'

metanqy, 2024-12-19:
>>> import genesis as gs
>>> gs.generate("A water droplet drops onto a beer bottle and then slowly slides down alone the bottle surface.")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'genesis' has no attribute 'generate'
>>>

[image]

metanqy, 2024-12-19:
[image]

lakbychance, 2024-12-19:
> > > > import genesis as gs
> > > > gs.generate("A water droplet drops onto a beer bottle and then slowly slides down alone the bottle surface.")
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > AttributeError: module 'genesis' has no attribute 'generate'
> 
> ![c6fe4e798e103163d978df4550e953b](https://private-user-images.githubusercontent.com/1705152/397264421-184accc7-8748-476d-aefc-522209e10c39.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ1OTM3NzMsIm5iZiI6MTczNDU5MzQ3MywicGF0aCI6Ii8xNzA1MTUyLzM5NzI2NDQyMS0xODRhY2NjNy04NzQ4LTQ3NmQtYWVmYy01MjIyMDllMTBjMzkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTlUMDczMTEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDkzNGNkZjhkNTZhMTVkMmNhNWI5Mzg4NDNhNjhkNGIxZTZmOTc0OTE0MzExZjg5NDg1M2NkMDU1NzQ0YmQ1MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.PxpPX029RyxIBSUkmNyYVNy3crR6pu9akIzK7t4qnWY)

As per [readme](https://github.com/Genesis-Embodied-AI/Genesis?tab=readme-ov-file#what-is-genesis), they have quoted that "Access to our generative feature will be gradually rolled out in the near future. If you are interested, feel free to explore more the [paper list](https://github.com/Genesis-Embodied-AI/Genesis?tab=readme-ov-file#papers-behind-genesis) below.".

ziyanx02, 2024-12-22:
We greatly appreciate your interest in Genesis. Access to the generative framework will be rolled out gradually in the near future. We will continue to provide updates in #6, so stay tuned!

wyahnke, 2024-12-23:
How do we request access to the generative features, or get on the waitlist?

varshneydevansh, 2025-01-29:
@ziyanx02 just asking what wyahnke asked, is there a way to get some early access? :)