Genesis-Embodied-AI/genesis-world #32: MacOS Error for name 'trimesh' is not defined; Failed to import pyrender. Rendering will not work.

hbai98, 2024-12-19:
(genesis) baihaotian@Bais-MacBook-Pro-2 space_time_inpaint % python test.py      
objc[73181]: Class GLFWHelper is implemented in both /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x1733f4570) and /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x317cd97c0). One of the two will be used. Which one is undefined.
objc[73181]: Class GLFWApplicationDelegate is implemented in both /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x1733f45c0) and /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x317cd9810). One of the two will be used. Which one is undefined.
objc[73181]: Class GLFWWindowDelegate is implemented in both /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x1733f45e8) and /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x317cd9838). One of the two will be used. Which one is undefined.
objc[73181]: Class GLFWContentView is implemented in both /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x1733f4638) and /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x317cd9888). One of the two will be used. Which one is undefined.
objc[73181]: Class GLFWWindow is implemented in both /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x1733f46b0) and /opt/anaconda3/envs/genesis/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x317cd9900). One of the two will be used. Which one is undefined.
Failed to import pyrender. Rendering will not work.
[Genesis] [15:57:22] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [15:57:22] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [15:57:22] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [15:57:22] [INFO] Running on [Apple M1 Pro] with backend gs.cpu. Device memory: 16.00 GB.
[Genesis] [15:57:22] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/Users/baihaotian/code/space_time_inpaint/test.py", line 4, in <module>
    scene = gs.Scene(show_viewer=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/opt/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/vis/visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/vis/rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "/opt/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
                           ^^^^^^^
NameError: name 'trimesh' is not defined

RobRoyce, 2024-12-19:
See #26 and the fix in #27

do-me, 2024-12-19:
Did `brew install ffmpeg` then got another error `import LuisaRenderPy`. Installed `pip install luisa-python` and tried again, but can't import the package:

```
(py3.11) ➜  rendering git:(main) pip install luisa-python
Collecting luisa-python
  Downloading luisa_python-0.3.7-cp311-cp311-macosx_14_0_arm64.whl.metadata (33 kB)
Collecting sourceinspect (from luisa-python)
  Downloading sourceinspect-0.0.4-py3-none-any.whl.metadata (1.1 kB)
Requirement already satisfied: numpy in /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages (from luisa-python) (1.26.4)
Requirement already satisfied: dill in /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages (from sourceinspect->luisa-python) (0.3.8)
Downloading luisa_python-0.3.7-cp311-cp311-macosx_14_0_arm64.whl (23.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 23.9/23.9 MB 7.0 MB/s eta 0:00:00
Downloading sourceinspect-0.0.4-py3-none-any.whl (3.5 kB)
Installing collected packages: sourceinspect, luisa-python
Successfully installed luisa-python-0.3.7 sourceinspect-0.0.4

(py3.11) ➜  rendering git:(main) python demo.py          
objc[53611]: Class GLFWHelper is implemented in both /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x329e00570) and /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x377a517c0). One of the two will be used. Which one is undefined.
objc[53611]: Class GLFWApplicationDelegate is implemented in both /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x329e005c0) and /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x377a51810). One of the two will be used. Which one is undefined.
objc[53611]: Class GLFWWindowDelegate is implemented in both /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x329e005e8) and /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x377a51838). One of the two will be used. Which one is undefined.
objc[53611]: Class GLFWContentView is implemented in both /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x329e00638) and /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x377a51888). One of the two will be used. Which one is undefined.
objc[53611]: Class GLFWWindow is implemented in both /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x329e006b0) and /opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x377a51900). One of the two will be used. Which one is undefined.
[Genesis] [09:35:43] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [09:35:43] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [09:35:43] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [09:35:43] [INFO] Running on [Apple M3 Max] with backend gs.metal. Device memory: 128.00 GB.
[Genesis] [09:35:43] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.7, commit 0131dce9, osx, python 3.11.8
[Genesis] [09:35:43] [DEBUG] [Taichi] Starting on arch=metal
[Genesis] [09:35:43] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/genesis/vis/raytracer.py", line 17, in <module>
    import LuisaRenderPy
ModuleNotFoundError: No module named 'LuisaRenderPy'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/dome/work/general/Genesis/examples/rendering/demo.py", line 166, in <module>
    main()
  File "/Users/dome/work/general/Genesis/examples/rendering/demo.py", line 11, in main
    scene = gs.Scene(
            ^^^^^^^^^
  File "/opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/genesis/vis/visualizer.py", line 60, in __init__
    from .raytracer import Raytracer
  File "/opt/homebrew/Caskroom/miniconda/base/envs/py3.11/lib/python3.11/site-packages/genesis/vis/raytracer.py", line 19, in <module>
    gs.raise_exception(f"Failed to import LuisaRenderer. {e.__class__.__name__}: {e}")

[Genesis] [09:35:45] [ERROR] Failed to import LuisaRenderer. ModuleNotFoundError: No module named 'LuisaRenderPy'
[Genesis] [09:35:45] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

zhenjia-xu, 2024-12-19:
I am sorry that Luisa requires cuda, and it is only supported on ubuntu now. you can use rasterization on macOS. Here is an example: https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py

do-me, 2024-12-19:
The readme states that it supports metal too: https://github.com/LuisaGroup/LuisaCompute?tab=readme-ov-file#backends-context-devices-and-resources

> LuisaCompute currently supports these backends:
CUDA
DirectX
Metal
CPU (Clang + LLVM)

chenglingq, 2024-12-19:
> See #26 and the fix in #27
> 
> tl;dr: install `ffmpeg` using Brew

Sadly I have ffmpeg installed and this rendering error's still happending

tkornuta, 2024-12-19:
> I am sorry that Luisa requires cuda, and it is only supported on ubuntu now. you can use rasterization on macOS. Here is an example: https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py

hey, thank you for providing that example, managed to run it on my Mac without visualization:
```
Genesis] [12:04:26] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [12:04:26] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [12:04:26] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [12:04:26] [INFO] Running on [Apple M2 Pro] with backend gs.cpu. Device memory: 16.00 GB.
[Genesis] [12:04:26] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [12:04:27] [INFO] Scene <206b3e3> created.
[Genesis] [12:04:27] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <0d516d2>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [12:04:27] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <35567b2>, morph: <gs.morphs.MJCF(file='/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [12:04:27] [INFO] Building scene <206b3e3>...
[Genesis] [12:04:30] [INFO] Compiling simulation kernels...
[Genesis] [12:04:34] [INFO] Building visualizer...
2024-12-19 12:04:35.799 Python[45772:491212] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/06/0wpmxt1s16x42b5b953l0tf40000gn/T/org.python.python.savedState
[Genesis] [12:04:35] [INFO] Running at 44897.51 FPS.
2082.573982125124 FPS
...
[Genesis] [12:04:57] [INFO] Running at 4927.31 FPS.
5047.297232250301 FPS
[Genesis] [12:04:57] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

However, when I tried to visualize (-v) then it raises more errors like
```
uniform not found: M
uniform not found: material.base_color_factor
uniform not found: material.metallic_factor
uniform not found: material.roughness_factor
uniform not found: material.emissive_factor
```

and then hangs on:
```
Traceback (most recent call last):
  File "/Users/tkornuta/workspace/genesis-playground/hello_franka2.py", line 66, in <module>
    main()
  File "/Users/tkornuta/workspace/genesis-playground/hello_franka2.py", line 42, in main
    scene.viewer.start()
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/vis/viewer.py", line 91, in start
    self._pyrender_viewer.start()
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 386, in start
    self._init_and_start_app()
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._renderer.render_texts(
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 258, in render_texts
    self._configure_forward_pass_viewport(0)
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 1160, in _configure_forward_pass_viewport
    glBindFramebuffer(GL_DRAW_FRAMEBUFFER, 0)
  File "/Users/tkornuta/Library/Caches/pypoetry/virtualenvs/genesis-playground-qtabWyTy-py3.11/lib/python3.11/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)
```

I've installed ffmpg (brew) and pyopenGL (poetry), here's my pyproject.toml
```
tkornuta@TK-14  ~/workspace/genesis-playground  cat pyproject.toml                                                                                                                 SIGABRT(6) ↵  6534  12:13:08
[tool.poetry]
name = "genesis-playground"
version = "0.1.0"
description = ""
authors = ["Tomasz Kornuta <tkornuta@gmail.com>"]
readme = "README.md"

[tool.poetry.dependencies]
python = "^3.11"
genesis-world = "^0.2.0"
torch = "^2.5.1"
torchvision = "^0.20.1"
pyopengl = "^3.1.7"
trimesh = "^4.5.3"
blender = "^1.4"
manifold3d = "^3.0.0"

[build-system]
requires = ["poetry-core"]
build-backend = "poetry.core.masonry.api"
```

any clue how to solve this?

VioletBenin, 2024-12-20:
Same Error occured when I run the sample code.

Traceback (most recent call last):
  File "/Users/violetbenin/import genesis as gs.py", line 7, in <module>
    scene = gs.Scene(show_viewer=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/vis/visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "/opt/anaconda3/envs/genesis/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
                           ^^^^^^^
NameError: name 'trimesh' is not defined

oktaydbk54, 2024-12-23:
@VioletBenin I'm having same error if someone find solution please let me know. Thank you

VioletBenin, 2024-12-23:
> @VioletBenin I'm having same error if someone find solution please let me know. Thank you

The solution is set an older venison of PyGEL3D like 0.1.0. The most updated one made conflict with Taichi 1.7.2.

peterheneback, 2024-12-23:
> and then hangs on:
> 
> ```
> OpenGL.error.GLError: GLError(
> 	err = 1282,
> 	description = b'invalid operation',
> 	baseOperation = glBindFramebuffer,
> 	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
> )
> ```

Same thing happens for me. Runs without visualisation, but hangs when I try running with `-v` flag. Have installed `ffmpeg` and downgraded `PyGEL3D` to version `0.1.0` to remove conflicts with Taichi.

VincentCCandela, 2024-12-29:
SOLUTION:

1) install ffmpeg on your system
2) put this in your main runtime to make sure that your code points correctly to your ffmpeg binary

import os
os.environ["IMAGEIO_FFMPEG_EXE"] = "/usr/bin/ffmpeg"

Miffyli, 2025-01-16:
Chiming in here in case anybody encounters this and above does not help:

I was running Python 3.12 and I was getting the same issue. Downgrading to Python 3.11 works fine. Seems to be related to Taichi: #513

closed 2025-02-22 by YilingQiao