Genesis-Embodied-AI/genesis-world #33: Unable to perform visual simulation when running on a server via SSH

Niemand03, 2024-12-19:
### 标题：SSH 连接服务器运行时无法进行可视化仿真

**问题描述:**

我通过 SSH 连接到一台 Linux 服务器运行 `genesis` 库的代码。这台服务器是 headless 的，没有图形显示设备。当我尝试使用 `gs.Scene(show_viewer=True)` 初始化场景时，程序抛出错误，提示 “No display detected. Use `show_viewer=False` for headless mode.”。这导致我无法在 SSH 连接的情况下直接进行可视化仿真。

**期望:**

我希望了解是否有推荐的方法，可以在 SSH 连接服务器且服务器无显示设备的情况下，仍然进行可视化仿真，或者至少获取可视化输出结果（例如保存为图像或视频文件）。

---

### Title: Unable to perform visual simulation when running on a server via SSH

**Description:**

I'm running `genesis` library code on a Linux server via SSH. This server is headless and does not have a graphics display device. When I try to initialize the scene using `gs.Scene(show_viewer=True)`, the program throws an error, saying "No display detected. Use `show_viewer=False` for headless mode.". This makes it impossible to directly perform visual simulation when connected via SSH.

**Expectation:**

I would like to know if there are recommended methods to still perform visual simulations or at least obtain visual output (e.g., save as images or video files) when connected to a server via SSH, which is headless.

HaithamIsmail, 2024-12-19:
same issue here

galeselee, 2024-12-19:
Using X11 forwarding (GPT maybe helpful)

WYYAHYT, 2024-12-19:
I tried X11 forwarding, but the window is black.
[image]

CCCalcifer, 2024-12-19:
[image]
I tried X11 Forwarding too, but I don't know how to deal with this error

satheeshKOLA532, 2024-12-19:
> ### 标题：SSH 连接服务器运行时无法进行可视化仿真
>
> **问题描述:**
>
> 我通过 SSH 连接到一台 Linux 服务器运行 `genesis` 库的代码。这台服务器是 headless 的，没有图形显示设备。当我尝试使用 `gs.Scene(show_viewer=True)` 初始化场景时，程序抛出错误，提示 “No display detected. Use [...]

Exactly same error that i am facing too.

galeselee, 2024-12-19:
@WYYAHYT 
I don't get this error, I haven't actually tried ssh, but in my experience Taichi can be forwarded via X11

linxiagong, 2024-12-19:
Try Headless Rendering, which may be a compromise:
https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#camera-headless-rendering

WYYAHYT, 2024-12-20:
@galeselee 
I tried to run `example/tutorials/visualization.py,` nothing wrong and  I can see rendered pictures. However, `hello_genesis` is still black.....

CostaliyA, 2024-12-20:
same issue here

caiobarrosv, 2024-12-23:
Same issue here. Black screen using docker as well.

koukoucoconut, 2024-12-23:
Does anyone face it？
[image]

milong26, 2025-02-17:
Is this solved?

YilingQiao, 2025-02-22:
Could use headless rendering
https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#camera-headless-rendering

closed 2025-02-22 by YilingQiao