Genesis-Embodied-AI/genesis-world #34: float64 issue on macOS MPS running examples/pbd_liquid.py

barius, 2024-12-19:
Dear devs, i'm running examples/pbd_liquid.py with `gs.init(backend=gs.metal, seed=0, precision="32", logging_level="debug")` and ran into the following issue:

```
[Genesis] [16:09:25] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.7, commit 0131dce9, osx, python 3.10.16
[Genesis] [16:09:25] [DEBUG] [Taichi] Starting on arch=metal
[Genesis] [16:09:25] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [16:09:25] [INFO] Scene <a7578ef> created.
[Genesis] [16:09:25] [INFO] Adding <gs.PBDParticleEntity>. idx: 0, uid: <244faed>, morph: <gs.morphs.Box>, material: <gs.PBD.Liquid>.
[Genesis] [16:09:25] [DEBUG] Sampled particles (`.ptc`) found in cache.
Traceback (most recent call last):
  File "/Users/barius/dev/3d/sim/code/Genesis/examples/pbd_liquid.py", line 82, in <module>
    main()
  File "/Users/barius/dev/3d/sim/code/Genesis/examples/pbd_liquid.py", line 36, in main
    liquid = scene.add_entity(
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/engine/scene.py", line 351, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/engine/simulator.py", line 143, in _add_entity
    entity = self.pbd_solver.add_entity(self.n_entities, material, morph, surface)
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/engine/solvers/pbd_solver.py", line 247, in add_entity
    entity = PBDParticleEntity(
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/pbd_entity.py", line 374, in __init__
    super().__init__(
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/particle_entity.py", line 69, in __init__
    self.sample()
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/particle_entity.py", line 259, in sample
    self._init_particles_offset = (gs.tensor(particles).contiguous() - gs.tensor(origin)).contiguous()
  File "<string>", line 3, in tensor
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/grad/creation_ops.py", line 64, in torch_op_wrapper
    gs_tensor = from_torch(torch_tensor, dtype, requires_grad, detach=True, scene=scene)
  File "/Users/barius/dev/mambaforge/envs/genesis/lib/python3.10/site-packages/genesis/grad/creation_ops.py", line 92, in from_torch
    gs_tensor = Tensor(torch_tensor.to(gs.device).to(dtype), scene=scene).clone()
TypeError: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64. Please use float32 instead.
```

Anywhere to modify simulation besides `precision="32"`? Tried to remove the `precision` arg in `gs.init()` and no go.

neilwth, 2024-12-19:
Same problem

neilwth, 2024-12-19:
> Dear devs, i'm running examples/pbd_liquid.py with `gs.init(backend=gs.metal, seed=0, precision="32", logging_level="debug")` and ran into the following issue:
>
> ```
> [Genesis] [16:09:25] [DEBUG] [...]

Just found that this only happens with backend=gs.metal. It works with backend=gs.cpu, but it is frustratingly slow on my MBP M2Max (~1.8 FPS).

barius, 2024-12-19:
> Just found that this only happens with backend=gs.metal. It works with backend=gs.cpu, but it is frustratingly slow on my MBP M2Max (~1.8 FPS).

using cpu on M3Max i get around 18 fps

barius, 2024-12-19:
while on windows i get 70 fps out of 9950x and an astounding 1000fps with 4090

zhenjia-xu, 2024-12-19:
With the help of [this PR](https://github.com/Genesis-Embodied-AI/Genesis/pull/8), the issue on macOS should be resolved.
Please pull the latest code, install Genesis using pip install -e ., and try it with Metal as the backend. You should notice a significant performance boost compared to using the CPU. :)
The PyPI version will be uploaded soon.

closed 2024-12-19 by zhenjia-xu

barius, 2024-12-19:
yup now i get ~3700 fps on metal :D, great work!

> With the help of [this PR](https://github.com/Genesis-Embodied-AI/Genesis/pull/8), the issue on macOS should be resolved. Please pull the latest code, install Genesis using pip install -e ., and try [...]