Genesis-Embodied-AI/genesis-world #36: Interactive viewer does not open on Mac M1

nalin, 2024-12-19:
I've installed genesis and am trying to run the Hello, Genesis example:

```
import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```

It runs without any error but the interactive viewer never opens.

The execution finishes with output like:
```
......
[Genesis] [00:31:43] [INFO] Running at 51.82 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.78 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.72 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.69 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.68 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.61 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.61 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.52 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.55 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.48 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.55 FPS.
[Genesis] [00:31:43] [INFO] Running at 51.52 FPS.
```

neilwth, 2024-12-19:
> I've installed genesis and am trying to run the Hello, Genesis example:
>
> ```
> import genesis as gs
> gs.init(backend=gs.cpu)
>
> scene = gs.Scene(show_viewer=True)
> plane = [...]

https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py

YutoYasunaga, 2024-12-19:
@RobRoyce Thanks, but still not working at all.
The interactive viewer never opens.

AlbertBrand, 2024-12-19:
I think you're missing a
```
scene.viewer.start()
```

RobRoyce, 2024-12-19:
> @RobRoyce Thanks, but still not working at all. The interactive viewer never opens.

You are likely seeing the following warning message:

```
[Genesis] [01:07:26] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`.
```

Try running `examples/render_on_macos.py` instead of the code you posted above. It uses `gs.tools.run_in_another_thread(fn=run_sim, args=(scene, args.vis))` to create the sim thread while the main thread shows the viewer.

YutoYasunaga, 2024-12-19:
@RobRoyce Yes, I tried `python examples/render_on_macos.py -v`

But I got same issue with #11 

```
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
```

zhuhu00, 2024-12-19:
I can get a rendered arm, but after a while, I get an error.
```
[Genesis] [17:14:29] [INFO] Running at 50.78 FPS.
52.999873638454346 FPS
Traceback (most recent call last):
  File "/Users/hu/code/Genesis/examples/render_on_macos.py", line 68, in <module>
    main()
  File "/Users/hu/code/Genesis/examples/render_on_macos.py", line 44, in main
    scene.viewer.start()
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/genesis/vis/viewer.py", line 91, in start
    self._pyrender_viewer.start()
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 386, in start
    self._init_and_start_app()
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/pyglet/clock.py", line 633, in tick
    return _default.tick(poll)
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/pyglet/clock.py", line 348, in tick
    self.call_scheduled_functions(delta_t)
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/pyglet/clock.py", line 292, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 939, in _time_event
    self.on_close()
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 575, in on_close
    pyglet.app.exit()
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/pyglet/app/__init__.py", line 91, in exit
    event_loop.exit()
  File "/Users/hu/miniconda3/lib/python3.10/site-packages/pyglet/app/cocoa.py", line 107, in exit
    self.platform_event_loop.notify()
AttributeError: 'CocoaAlternateEventLoop' object has no attribute 'platform_event_loop'
[Genesis] [17:14:29] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```
My platform is Macbook M3 Pro, 18GB

zhushunqing, 2024-12-19:
> @RobRoyce Yes, I tried `python examples/render_on_macos.py -v`
>
> But I got same issue with #11
>
> ```
>   File [...]

I got same issue. 
iMac 27 / 3.8 GHz 8-Core Intel Core i7 / AMD Radeon Pro 5500 XT 8 GB

Josh00-Lu, 2024-12-19:
Try this on macos:

```python
import numpy as np
import genesis as gs

########################## init ##########################
gs.init()

########################## create a scene ##########################
scene = gs.Scene(
    viewer_options= gs.options.ViewerOptions(
        camera_pos    = (0.0, -2, 1.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 40,
        max_FPS       = 200,
    ),
    rigid_options=gs.options.RigidOptions(
        enable_joint_limit = False,
    ),
)

########################## entities ##########################
plane = scene.add_entity(
    gs.morphs.Plane(),
)
robot = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

########################## build ##########################
n_envs = 16
scene.build(n_envs=n_envs, env_spacing=(1.0, 1.0))

target_quat = np.tile(np.array([0, 1, 0, 0]), [n_envs, 1]) # pointing downwards
center = np.tile(np.array([0.4, -0.2, 0.25]), [n_envs, 1])
angular_speed = np.random.uniform(-10, 10, n_envs)
r = 0.1

ee_link = robot.get_link('hand')
    
def run_sim():
    from time import time

    t_prev = time()
    i = 0
    while True:
        i += 1
        
        target_pos = np.zeros([n_envs, 3])
        target_pos[:, 0] = center[:, 0] + np.cos(i/360*np.pi*angular_speed) * r
        target_pos[:, 1] = center[:, 1] + np.sin(i/360*np.pi*angular_speed) * r
        target_pos[:, 2] = center[:, 2]
        target_q = np.hstack([target_pos, target_quat])

        q = robot.inverse_kinematics(
            link     = ee_link,
            pos      = target_pos,
            quat     = target_quat,
            rot_mask = [False, False, True], # for demo purpose: only restrict direction of z-axis
        )

        robot.set_qpos(q)

        scene.step()

        t_now = time()
        print(1 / (t_now - t_prev), "FPS")
        t_prev = t_now
            
gs.tools.run_in_another_thread(fn=run_sim, args=())
scene.viewer.start()
```

YutoYasunaga, 2024-12-19:
@Josh00-Lu Thanks, but not working, still same error.

```
  File "/Users/yuto/.anyenv/envs/pyenv/versions/3.11.0/lib/python3.11/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
```

ksDreamer, 2024-12-19:
@Josh00-Lu Thanks, it works for me. But a small issue that pressing `s` will result to quitting program.

wucaicoding, 2024-12-19:
@ksDreamer **Same!** It even caused my running pycharm program crashed.

wucaicoding, 2024-12-19:
@RobRoyce Thanks! You solved my problem.
Device: MacBook Air
Processor:M1
RAM:16G
Python:3.12
Running code:examples/render_on_macos.py

At first, I did everything but run without configurating the --vis flag, and the interactive viewer just don't pop up tried many times. I'm using PyCharm 2024.3.1 (Professional Edition) as IDE,so I did these steps:
1,find the "Run/Debug Configurations" dialog:
2,select "Edit..."
3,Select current  script from the list on the left.
4,In the "Script parameters" field, add --vis.
5,Click "OK" to save the configuration.

Then run it, the viewer finally popped up. But after a few seconds, it automatically closed. The problem is in the 60 and 61 line.
` if i > 200:
            break`
It set a time limit. So if we don't want a time limit, we can just simply
` #if i > 200:
            #break`
After this, the viewer runs normally.

But there are still some little bugs.  
1, nothing will happen when pressed 'h'
2, each time try to full screen the window will cause python quit unexpectedly
`Process finished with exit code 139 (interrupted by signal 11:SIGSEGV)`
3, each time press 's' also cause person quit expectedly
`Process finished with exit code 134 (interrupted by signal 6:SIGABRT)`
4, when press 'r', it starts recording, but press 'r' another time, it caused unexpected quit.
`*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplication macOSVersion]: unrecognized selector sent to instance 0x379f23ec0'
*** First throw call stack:
(
	0   CoreFoundation                      0x0000000190220300 __exceptionPreprocess + 176
	1   libobjc.A.dylib                     0x000000018fd06cd8 objc_exception_throw + 88
	2   CoreFoundation                      0x00000001902d5638 -[NSObject(NSObject) __retain_OA] + 0
	3   CoreFoundation                      0x000000019018ccc0 ___forwarding___ + 1568
	4   CoreFoundation                      0x000000019018c5e0 _CF_forwarding_prep_0 + 96
	5   libtk8.6.dylib                      0x000000017e955918 GetRGBA + 64
	6   libtk8.6.dylib                      0x000000017e955714 TkpGetColor + 428
	7   libtk8.6.dylib                      0x000000017e89f7bc Tk_GetColor + 168
	8   libtk8.6.dylib                      0x000000017e890b7c Tk_Get3DBorder + 152
	9   libtk8.6.dylib                      0x000000017e8909bc Tk_Alloc3DBorderFromObj + 144
	10  libtk8.6.dylib                      0x000000017e8a0e08 DoObjConfig + 1008
	11  libtk8.6.dylib                      0x000000017e8a0918 Tk_InitOptions + 372
	12  libtk8.6.dylib                      0x000000017e8a07fc Tk_InitOptions + 88
	13  libtk8.6.dylib                      0x000000017e8ce0f8 CreateFrame + 1472
	14  libtk8.6.dylib                      0x000000017e8ce424 TkListCreateFrame + 172
	15  libtk8.6.dylib                      0x000000017e8c6cbc Initialize + 1888
	16  _tkinter.cpython-312-darwin.so      0x000000016d19a1b8 Tcl_AppInit + 92
	17  _tkinter.cpython-312-darwin.so      0x000000016d199e08 Tkapp_New + 668
	18  _tkinter.cpython-312-darwin.so      0x000000016d199b68 _tkinter_create_impl + 268
	19  _tkinter.cpython-312-darwin.so      0x000000016d1997e8 _tkinter_create + 276
	20  python3.12                          0x00000001048e93a8 cfunction_vectorcall_FASTCALL + 96
	21  python3.12                          0x00000001049f2af4 _PyEval_EvalFrameDefault + 225236
	22  python3.12                          0x0000000104918920 slot_tp_init + 328
	23  python3.12                          0x000000010490cbc0 type_call + 148
	24  python3.12                          0x00000001049f2e94 _PyEval_EvalFrameDefault + 226164
	25  python3.12                          0x0000000104885d60 method_vectorcall + 180
	26  python3.12                          0x00000001049f7244 _PyEval_EvalFrameDefault + 243492
	27  python3.12                          0x00000001049b91b4 PyEval_EvalCode + 260
	28  python3.12                          0x0000000104a6a524 run_mod + 288
	29  python3.12                          0x0000000104a6a2a0 pyrun_file + 156
	30  python3.12                          0x0000000104a69c6c _PyRun_SimpleFileObject + 288
	31  python3.12                          0x0000000104a69550 _PyRun_AnyFileObject + 80
	32  python3.12                          0x0000000104a9596c pymain_run_file_obj + 180
	33  python3.12                          0x0000000104a95330 pymain_run_file + 72
	34  python3.12                          0x0000000104a94974 Py_RunMain + 1124
	35  python3.12                          0x0000000104a95cd8 pymain_main + 552
	36  python3.12                          0x000000010480faa8 main + 56
	37  dyld                                0x000000018fd44274 start + 2840
)
libc++abi: terminating due to uncaught exception of type NSException

Process finished with exit code 134 (interrupted by signal 6:SIGABRT)` 

But there is a way to record videos using codes according to the user guide. [https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html](url)
`# start camera recording. Once this is started, all the rgb images rendered will be recorded internally
cam.start_recording()

import numpy as np
for i in range(120):
    scene.step()

    # change camera position
    cam.set_pose(
        pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
        lookat = (0, 0, 0.5),
    )
    
    cam.render()

# stop recording and save video. If `filename` is not specified, a name will be auto-generated using the caller file name.
cam.stop_recording(save_to_filename='video.mp4', fps=60)`

ngoldbla, 2024-12-19:
Tried many ways many times. Only able to perform simulations without renders. MacOS 15.2, M2 Max.

buck-0x, 2024-12-19:
Same issue for me. Tried everything in this thread to no avail.

ZeroSaturn, 2024-12-20:
For me i used the code Josh00-lu provided which worked , but it didnt initialy , what i had to do was pip uninstall glfw, and brew uninstall glfw, also it will say view initilizaed but you wont actually see anything until 30 seconds later

ZeroSaturn, 2024-12-20:
I got this code to also work too, i made it using Gemeni, its just a blank scene but it should work for the Mac MX :
import genesis as gs

# Initialize Genesis with Metal backend and 32-bit precision
gs.init(backend=gs.cpu, precision="32", logging_level="debug")

# Create an empty scene with viewer options for macOS
scene = gs.Scene(
    viewer_options=gs.options.ViewerOptions(
        camera_pos=(0.0, -2, 1.5),  # Example camera position
        camera_lookat=(0.0, 0.0, 0.5),  # Example lookat point
        camera_fov=40,
        max_FPS=200,
    ),
)

# Build the scene (no entities to add yet)
scene.build()

# Function to run in a separate thread (currently does nothing)
def run_sim():
    pass

# Start the viewer (this should open a blank window)
gs.tools.run_in_another_thread(fn=run_sim, args=())
scene.viewer.start()

YutoYasunaga, 2024-12-20:
For those who want to test viewer and mp4 file in Mac, try this code. I tried and got viewer rendered & mp4 video.

### Requirement:

- Using Conda for environment management
- Python 3.12
- PyOpenGL 3.1.7
- ffmpeg already

### Test code:

```python
import genesis as gs
import numpy as np

def main():
    gs.init(backend=gs.cpu)

    scene = gs.Scene(
        show_viewer = True,
        viewer_options = gs.options.ViewerOptions(
            res           = (1280, 960),
            camera_pos    = (3.5, 0.0, 2.5),
            camera_lookat = (0.0, 0.0, 0.5),
            camera_fov    = 40,
            max_FPS       = 60,
        ),
        vis_options = gs.options.VisOptions(
            show_world_frame = True,
            world_frame_size = 1.0,
            show_link_frame  = False,
            show_cameras     = False,
            plane_reflection = True,
            ambient_light    = (0.1, 0.1, 0.1),
        ),
        renderer=gs.renderers.Rasterizer(),
    )

    plane = scene.add_entity(
        gs.morphs.Plane(),
    )
    franka = scene.add_entity(
        gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
    )

    cam = scene.add_camera(
        res    = (640, 480),
        pos    = (3.5, 0.0, 2.5),
        lookat = (0, 0, 0.5),
        fov    = 30,
        GUI    = False,
    )

    scene.build()
    gs.tools.run_in_another_thread(fn=run_sim, args=(scene, cam))
    cam.start_recording()
    scene.viewer.start()

def run_sim(scene, cam):
    for i in range(120):
        scene.step()
        cam.set_pose(
            pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
            lookat = (0, 0, 0.5),
        )
        cam.render()
    cam.stop_recording(save_to_filename='video.mp4', fps=60)
    scene.viewer.stop()

if __name__ == "__main__":
    main()
```

### Result:

**Viewer:**
[image]

**Video:**
https://github.com/user-attachments/assets/079c2d18-42fb-4171-9757-50523c7605fb

supernba123, 2024-12-20:
> For those who want to test viewer and mp4 file in Mac, try this code. I tried and got viewer rendered & mp4 video.
> 
> ### Requirement:
> * Using Conda for environment management
> * Python 3.12
> * PyOpenGL 3.1.7
> * ffmpeg already
> 
> ### Test code:
> ```python
> import genesis as gs
> import numpy as np
> 
> def main():
>     gs.init(backend=gs.cpu)
> 
>     scene = gs.Scene(
>         show_viewer = True,
>         viewer_options = gs.options.ViewerOptions(
>             res           = (1280, 960),
>             camera_pos    = (3.5, 0.0, 2.5),
>             camera_lookat = (0.0, 0.0, 0.5),
>             camera_fov    = 40,
>             max_FPS       = 60,
>         ),
>         vis_options = gs.options.VisOptions(
>             show_world_frame = True,
>             world_frame_size = 1.0,
>             show_link_frame  = False,
>             show_cameras     = False,
>             plane_reflection = True,
>             ambient_light    = (0.1, 0.1, 0.1),
>         ),
>         renderer=gs.renderers.Rasterizer(),
>     )
> 
>     plane = scene.add_entity(
>         gs.morphs.Plane(),
>     )
>     franka = scene.add_entity(
>         gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
>     )
> 
>     cam = scene.add_camera(
>         res    = (640, 480),
>         pos    = (3.5, 0.0, 2.5),
>         lookat = (0, 0, 0.5),
>         fov    = 30,
>         GUI    = False,
>     )
> 
>     scene.build()
>     gs.tools.run_in_another_thread(fn=run_sim, args=(scene, cam))
>     cam.start_recording()
>     scene.viewer.start()
> 
> 
> def run_sim(scene, cam):
>     for i in range(120):
>         scene.step()
>         cam.set_pose(
>             pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
>             lookat = (0, 0, 0.5),
>         )
>         cam.render()
>     cam.stop_recording(save_to_filename='video.mp4', fps=60)
>     scene.viewer.stop()
> 
> 
> if __name__ == "__main__":
>     main()
> ```
> 
> ### Result:
> **Viewer:** <img alt="Screenshot 2024-12-20 at 12 45 01" width="1277" src="https://private-user-images.githubusercontent.com/6220668/397608395-ef4deb1e-25f4-48c5-abb7-3843026b7a29.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NjY4NDksIm5iZiI6MTczNDY2NjU0OSwicGF0aCI6Ii82MjIwNjY4LzM5NzYwODM5NS1lZjRkZWIxZS0yNWY0LTQ4YzUtYWJiNy0zODQzMDI2YjdhMjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjBUMDM0OTA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzJiNDIxNGI4YTA3ZTQ0MDc2NmFhNWNjNTNhNTNjMjg1MDg3MzAxZjc2NzhiY2VjYTRkYWZkMWJmODAxNzkzNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Apo3ZsiShUThmTUDHF4SbaPvGsfj10dbZ70x73mpW3A">
> 
> **Video:** https://github.com/user-attachments/assets/079c2d18-42fb-4171-9757-50523c7605fb

works for me in M3 with conda

Jassy930, 2024-12-20:
> works for me in M3 with conda

could you provide package list and version like "pip freeze" output in conda? I have same issue, but currently i can't use conda.

sensahin, 2024-12-27:
For some reasons, I cannot and prefer not to install Conda on my system (I use pyenv and have had conflicts with Conda in the past). 

Is there a workaround to set up the project using a plain Python environment (e.g., python -m venv or similar) without relying on Conda?

snakajima, 2025-01-14:
To clarify the issue, here is the minimum code you need to run the simulator on MacOS.

```python
import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

def run_sim(scene):
    for i in range(100):
        scene.step()

# You need to run simulation in a separate thread on MacOS
gs.tools.run_in_another_thread(fn=run_sim, args=(scene,))
# This must be called from the main thread after starting the thread
scene.viewer.start() 
```

aresnasa, 2025-01-16:
# M3 MAX
# conda install python3.12
# pip install genesis
# gen101.py
# ------------------
import genesis as gs
gs.init(backend=gs.metal)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

def run_sim(scene,):
    for i in range(1000):
        scene.step()#585 

# add cam
gs.tools.run_in_another_thread(fn=run_sim, args=(scene,)) # start the simulation in another thread
scene.viewer.start() # start the viewer in the main thread (the render thread)

vlad-stability, 2025-02-04:
Hey all, I'm managing to run the basic vis on my M3, but can you please tell me how I can also render the different modes (e,g., depth, normal, rgb)?

This is the code I use, but it gives me a "Unknown C++ OpenCV2 error":

`import genesis as gs
import numpy as np

def recordVideo(scene, cam):

    #This currently does not seem to work
    #rgb, depth, segmentation, normal = cam.render(rgb=True, depth=True, segmentation=True, normal=True)

    cam.start_recording()

    for i in range(120):
        scene.step()
        cam.set_pose(
            pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
            lookat = (0, 0, 0.5),
        )
        cam.render()
    cam.stop_recording(save_to_filename='outputs/video.mp4', fps=60)

def main():

    gs.init(backend=gs.cpu)

    scene = gs.Scene(
        show_viewer = True,
        viewer_options = gs.options.ViewerOptions(
            res           = (1280, 960),
            camera_pos    = (3.5, 0.0, 2.5),
            camera_lookat = (0.0, 0.0, 0.5),
            camera_fov    = 40,
            max_FPS       = 60,
        ),
        vis_options = gs.options.VisOptions(
            show_world_frame = True,
            world_frame_size = 1.0,
            show_link_frame  = False,
            show_cameras     = False,
            plane_reflection = True,
            ambient_light    = (0.1, 0.1, 0.1),
        ),
        renderer=gs.renderers.Rasterizer(),
    )

    plane = scene.add_entity(
        gs.morphs.Plane(),
    )
    franka = scene.add_entity(
        gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
    )

    cam = scene.add_camera(
            res    = (640, 480),
            pos    = (3.5, 0.0, 2.5),
            lookat = (0, 0, 0.5),
            fov    = 30,
            GUI    = True,
        )

    scene.build()

    gs.tools.run_in_another_thread(fn=recordVideo, args=(scene, cam))

    scene.viewer.start()

if __name__ == "__main__":
    main()`

closed 2025-02-22 by YilingQiao