Genesis-Embodied-AI/genesis-world #37: An err occured running code in doc Hello Genesis

DaMiTian, 2024-12-19:
Hello,I got an error while running the code under:

import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='/home/nvidia/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()

the path is correct,but I got the err:
[image]
after this error occured,I fixed up an error for my openGL:https://stackoverflow.com/questions/72110384/libgl-error-mesa-loader-failed-to-open-iris
I run the code :   conda install -c conda-forge libstdcxx-ng   
in my virtual conda environment,and replay the python demo,then got the err:
[Genesis] [16:37:16] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮                                                                                                                                                                
[Genesis] [16:37:16] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│                                                                                                                                                                
[Genesis] [16:37:16] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯                                                                                                                                                                
[Genesis] [16:37:16] [INFO] Running on [13th Gen Intel(R) Core(TM) i7-13700H] with backend gs.cpu. Device memory: 62.46 GB.                                                                                                                                                        
[Genesis] [16:37:16] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.                                                                                                                                       
[Genesis] [16:37:16] [INFO] Scene <1313ad1> created.
[Genesis] [16:37:16] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <146f4e4>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [16:37:16] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <ccc73a1>, morph: <gs.morphs.MJCF(file='/home/nvidia/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [16:37:16] [INFO] Building scene <1313ad1>...
[Genesis] [16:37:19] [INFO] Compiling simulation kernels...
[Genesis] [16:37:24] [INFO] Building visualizer...
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/nvidia/Genesis/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/nvidia/Genesis/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/nvidia/Genesis/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/nvidia/Genesis/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/home/nvidia/Genesis/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/nvidia/Genesis/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/nvidia/Genesis/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 469, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context

zhenjia-xu, 2024-12-19:
As suggested in /issues/10
Please give it a try:
```
PYOPENGL_PLATFORM=osmesa python try.py 
```

DaMiTian, 2024-12-19:
> As suggested in [/issues/10](https://github.com/Genesis-Embodied-AI/Genesis/issues/10) Please give it a try:
> 
> ```
> PYOPENGL_PLATFORM=osmesa python try.py 
> ```
thanks a lot!
by using it,I got another err:
Traceback (most recent call last):
  File "/home/nvidia/Genesis/genesis/assets/try.py", line 7, in <module>
    scene = gs.Scene(show_viewer=False)
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
NameError: name 'trimesh' is not defined

DaMiTian, 2024-12-19:
> > As suggested in [/issues/10](https://github.com/Genesis-Embodied-AI/Genesis/issues/10) Please give it a try:
> > ```
> > PYOPENGL_PLATFORM=osmesa python try.py 
> > ```
> 
> thanks a lot! by using it,I got another err: Traceback (most recent call last): File "/home/nvidia/Genesis/genesis/assets/try.py", line 7, in scene = gs.Scene(show_viewer=False) File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/utils/misc.py", line 27, in new_init original_init(self, *args, **kwargs) File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/engine/scene.py", line 148, in **init** self._visualizer = Visualizer( File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 26, in **init** self._context = RasterizerContext(vis_options) File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 37, in **init** self.init_meshes() File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0) NameError: name 'trimesh' is not defined
and I can find trimesh in my envs path:/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/ext/trimesh

DaMiTian, 2024-12-19:
will i trying to remove try ...(import for trimesh) else pass,I got the error:
Traceback (most recent call last):
  File "/home/nvidia/Genesis/genesis/assets/try.py", line 1, in <module>
    import genesis as gs
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/__init__.py", line 274, in <module>
    from .engine.scene import Scene
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/engine/scene.py", line 24, in <module>
    from genesis.vis import Visualizer
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/__init__.py", line 1, in <module>
    from .visualizer import Visualizer
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 8, in <module>
    from .rasterizer_context import RasterizerContext
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 9, in <module>
    from genesis.ext import pyrender, trimesh
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/ext/pyrender/__init__.py", line 2, in <module>
    from .light import Light, PointLight, DirectionalLight, SpotLight
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/ext/pyrender/light.py", line 11, in <module>
    from OpenGL.GL import *
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/GL/__init__.py", line 4, in <module>
    from OpenGL.GL.VERSION.GL_1_1 import *
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/GL/VERSION/GL_1_1.py", line 14, in <module>
    from OpenGL.raw.GL.VERSION.GL_1_1 import *
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/raw/GL/VERSION/GL_1_1.py", line 7, in <module>
    from OpenGL.raw.GL import _errors
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/raw/GL/_errors.py", line 4, in <module>
    _error_checker = _ErrorChecker( _p, _p.GL.glGetError )
AttributeError: 'NoneType' object has no attribute 'glGetError'

david-wb, 2024-12-19:
Same issue for me. Can't get the quick start to launch. I'm on ubuntu 22.04, with python 3.10, and I installed genesis in a poetry environment.

Hussainflr, 2024-12-19:
I'm also getting same error

> thanks a lot! by using it,I got another err: Traceback (most recent call last): File "/home/nvidia/Genesis/genesis/assets/try.py", line 7, in scene = gs.Scene(show_viewer=False) File [...]

vhartman, 2024-12-19:
I had a similar error to the one in the initial post, and got it to work with 

```
 import os
 os.environ['PYOPENGL_PLATFORM'] = 'glx'  # or try 'glx' if this doesn't work
```

before 
```
import genesis
```

Hussainflr, 2024-12-19:
I was also getting this issue with python3.11 this fix  [worked](https://github.com/Genesis-Embodied-AI/Genesis/issues/11#issuecomment-2553024034) for me 

> will i trying to remove try ...(import for trimesh) else pass,I got the error: Traceback (most recent call last): File "/home/nvidia/Genesis/genesis/assets/try.py", line 1, in import genesis as gs File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/**init**.py", line 274, in from .engine.scene import Scene File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/engine/scene.py", line 24, in from genesis.vis import Visualizer File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/**init**.py", line 1, in from .visualizer import Visualizer File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 8, in from .rasterizer_context import RasterizerContext File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 9, in from genesis.ext import pyrender, trimesh File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/ext/pyrender/**init**.py", line 2, in from .light import Light, PointLight, DirectionalLight, SpotLight File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/genesis/ext/pyrender/light.py", line 11, in from OpenGL.GL import * File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/GL/**init**.py", line 4, in from OpenGL.GL.VERSION.GL_1_1 import * File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/GL/VERSION/GL_1_1.py", line 14, in from OpenGL.raw.GL.VERSION.GL_1_1 import * File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/raw/GL/VERSION/GL_1_1.py", line 7, in from OpenGL.raw.GL import _errors File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/OpenGL/raw/GL/_errors.py", line 4, in _error_checker = _ErrorChecker( _p, _p.GL.glGetError ) AttributeError: 'NoneType' object has no attribute 'glGetError'

evrenesat, 2024-12-19:
same error on macos 15.1, py3.11 with following packages.
[reqs.txt](https://github.com/user-attachments/files/18196388/reqs.txt)

DaMiTian, 2024-12-19:
hi there,finally I solved all bug on my machine:
My platform:2000ada,i7,550nvdriver,cuda12.4,torch2.4,py3.10
firstly,you should check i PyopeGL on your machine,if yes upgrade it to the newest version(mine is 3.17.0 though the project req for 3.10)
then check for some mesa req:
sudo apt-get install libosmesa6-dev
sudo apt-get install libgl1-mesa-glx libglu1-mesa
sudo apt-get install mesa-utils
maybe you should install ffpeg?though it doesnt work for me.
may of the bug in it caused by openGL,check the openGL version is helpful,in my device,while using3.10,I got an err:
TypeError: unhashable type
and 3.17 will not.
wish everyone chould solve the bug in it,may we have some err,but the fixing solution wont be the same.

chenglingq, 2024-12-19:
[Error]: 'NoneType' object has no attribute 'glGetError'

Failed to import pyrender. Rendering will not work.
[Genesis] [19:40:45] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [19:40:45] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [19:40:45] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [19:40:45] [INFO] Running on [NVIDIA H100 80GB HBM3] with backend gs.cuda. Device memory: 79.11 GB.

both 
PYOPENGL_PLATFORM=osmesa 
and
os.environ['PYOPENGL_PLATFORM'] = 'glx' 

is not working for a linux environment, still seek help there, thanks

skyswordx, 2024-12-20:
> [Error]: 'NoneType' object has no attribute 'glGetError'
>
> Failed to import pyrender. Rendering will not work. [Genesis] [19:40:45] [INFO] [...]

By trying above methods
my trace back is `TypeError: 'NoneType' object is not callable`
```shell
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 916, in _update_context
    texture._add_to_context()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/texture.py", line 219, in _add_to_context
    max_aniso = glGetFloat(texture_filter_anisotropic.GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/wrapper.py", line 678, in wrapperCall
    raise err
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/wrapper.py", line 671, in wrapperCall
    result = wrappedOperation( *cArguments )
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1280,
        description = b'invalid enumerant',
        baseOperation = glGetFloatv,
        pyArgs = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                <object object at 0x7fcaeedf01c0>,
        ),
        cArgs = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                array([0.], dtype=float32),
        ),
        cArguments = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                array([0.], dtype=float32),
        )
)

```

jash101, 2024-12-20:
If you have an older OpenGL version (like 3.1), updating OpenGL might solve the error:
```
sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt upgrade
```

AgentOfLyra, 2024-12-21:
#43 Solved
Maybe helpful to you

jash101, 2024-12-21:
> #43 Solved Maybe helpful to you

from your reply (https://github.com/Genesis-Embodied-AI/Genesis/issues/43#issuecomment-2557624624), it looks like you have the same low fps issue (https://github.com/Genesis-Embodied-AI/Genesis/issues/138#issue-2752075095).
have you been able to solve it?

AgentOfLyra, 2024-12-21:
> > #43 Solved Maybe helpful to you
> 
> @AgentOfLyra from your reply ([#43 (comment)](https://github.com/Genesis-Embodied-AI/Genesis/issues/43#issuecomment-2557624624)), it looks like you have the same low fps issue ([#138 (comment)](https://github.com/Genesis-Embodied-AI/Genesis/issues/138#issue-2752075095)). have you been able to solve it?

My fps is actually not low and can reach 60 (although its response is very laggy), so maybe we are not encountering the same problem?
However, I also solved this problem according to the method in #43 
try these code
```shell
export GALLIUM_DRIVER=d3d12
export MESA_D3D12_DEFAULT_ADAPTER_NAME=NVIDIA  # if wanna nvidia gpu
```

BruskB, 2024-12-23:
Did everything here, the program works but around 0.99 to 0.33 FPS on a RTX4070Ti Super, is this normal?

closed 2025-02-22 by YilingQiao