Genesis-Embodied-AI/genesis-world #38: Is there a demo video of gymnastics generated by Genesis available?

tumluliu, 2024-12-19:
in order to understand its different understanding of physics other than the existing AI video generators

e.g.

https://imgur.com/a/heres-sora-generated-video-of-gymnastics-LQyKTf0 

Thanks

antragoudaras, 2024-12-19:
Nice question, or how do we even generate a video after installation? The demos in Twitter (X) it is demonstrated using gs.generate, but when looking at their documentation I did not find anything similar.

zhouxian, 2024-12-19:
Hi we are only opensourcing the low-level physics engine so far as written in the twitter. The generative work requires quite a bit of more work before ready for public use, but we are working on it:)

tumluliu, 2024-12-19:
thanks! Looking forward to seeing a generated gymnastics video with normal physics, either as a demo or (the best, if possible) from my local env with your open codebase

closed 2024-12-22 by ziyanx02