Genesis-Embodied-AI/genesis-world #42: Build Ray Tracing Renderer error

zhuhu00, 2024-12-19:
When I build [Ray Tracing Renderer](https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html#optional-ray-tracing-renderer), I got these error. 
```
LuisaRender git:(030a375f) cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.10 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF -D ZLIB_INCLUDE_DIR=$CONDA_INCLUDE_PATH -D CMAKE_OSX_ARCHITECTURES=arm64
-- Could not find sccache
-- Shared libraries enabled
-- Looking for ZLIB...
-- Checking for module 'zzip-zlib-config'
--   No package 'zzip-zlib-config' found
-- Found ZLIB: optimized;/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/lib/libz.tbd;debug;/Library/Developer/CommandLineTools/SDKs/MacOSX15.2.sdk/usr/lib/libz.tbd
-- Checking for module 'minizip'
--   No package 'minizip' found
-- Build an import-only version of Assimp.
INFO Hunter enabled or RT-extension not found. glTF import/export will be built without Open3DGC-compression.
-- Enabled importer formats: AMF 3DS AC ASE ASSBIN B3D BVH COLLADA DXF CSM HMP IRRMESH IQM IRR LWO LWS M3D MD2 MD3 MD5 MDC MDL NFF NDO OFF OBJ OGRE OPENGEX PLY MS3D COB BLEND IFC XGL FBX Q3D Q3BSP RAW SIB SMD STL TERRAGEN 3D X X3D GLTF 3MF MMD
-- Disabled importer formats:
-- Treating all warnings as errors (for assimp library only)
-- Could not find sccache
-- Build spdlog: 1.12.0
-- Build type: Release
CMake Deprecation Warning at src/compute/src/ext/EASTL/packages/EABase/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Rust Toolchain: stable-aarch64-apple-darwin
-- Rust Target: aarch64-apple-darwin
-- Using Corrosion as a subdirectory
warning: profiles for the non root package will be ignored, specify profiles at the workspace root:
package:   /Users/hu/code/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/luisa_compute_ir/Cargo.toml
workspace: /Users/hu/code/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/Cargo.toml
warning: profiles for the non root package will be ignored, specify profiles at the workspace root:
package:   /Users/hu/code/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/luisa_compute_backend_impl/Cargo.toml
workspace: /Users/hu/code/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/Cargo.toml
warning: virtual workspace defaulting to `resolver = "1"` despite one or more workspace members being on edition 2021 which implies `resolver = "2"`
note: to keep the current resolver, specify `workspace.resolver = "1"` in the workspace root's manifest
note: to use the edition 2021 resolver, specify `workspace.resolver = "2"` in the workspace root's manifest
note: for more details see https://doc.rust-lang.org/cargo/reference/resolver.html#resolver-versions
-- Could NOT find Vulkan (missing: Vulkan_LIBRARY Vulkan_INCLUDE_DIR) (found version "")
CMake Warning at src/compute/src/backends/common/CMakeLists.txt:48 (message):
  Vulkan not found, vulkan swapchain will not be enabled.

-- Build with Metal backend
-- No CUDA Built! OFF
-- Build with CPU backend
luisa-render-pipe-render created!
CMake Warning (dev) at /Users/hu/miniconda3/share/cmake/pybind11/FindPythonLibsNew.cmake:101 (message):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning, or
  preferably upgrade to using FindPython, either by calling it explicitly
  before pybind11, or by setting PYBIND11_FINDPYTHON ON before pybind11.
Call Stack (most recent call first):
  /Users/hu/miniconda3/share/cmake/pybind11/pybind11Tools.cmake:50 (find_package)
  /Users/hu/miniconda3/share/cmake/pybind11/pybind11Common.cmake:228 (include)
  /Users/hu/miniconda3/share/cmake/pybind11/pybind11Config.cmake:250 (include)
  src/apps/CMakeLists.txt:10 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found pybind11: /Users/hu/miniconda3/include (found version "2.13.6")
-- Configuring done (2.6s)
-- Generating done (1.3s)
-- Build files have been written to: /Users/hu/code/Genesis/genesis/ext/LuisaRender/build
```
after `cmake --build build -j $(nproc)`
```
[ 91%] Built target luisa-render-environment-null
[ 92%] Built target luisa-render-transform-stack
Undefined symbols for architecture arm64:
  "luisa::compute::Window::~Window()", referenced from:
      luisa::render::AuxiliaryBufferPathTracingInstance::~AuxiliaryBufferPathTracingInstance() in aov.cpp.o
      luisa::render::AuxiliaryBufferPathTracingInstance::~AuxiliaryBufferPathTracingInstance() in aov.cpp.o
  "luisa::compute::Framerate::clear()", referenced from:
      luisa::render::AuxiliaryBufferPathTracingInstance::render(luisa::compute::Stream&) in aov.cpp.o
  "luisa::compute::Framerate::Framerate(unsigned long)", referenced from:
      luisa::render::AuxiliaryBufferPathTracing::build(luisa::render::Pipeline&, luisa::render::CommandBuffer&) const in aov.cpp.o
ld: symbol(s) not found for architecture arm64
c++: error: linker command failed with exit code 1 (use -v to see invocation)
[ 93%] Built target luisa-render-surface-glass
make[2]: *** [bin/libluisa-render-integrator-aov.so] Error 1
make[1]: *** [src/integrators/CMakeFiles/luisa-render-integrator-aov.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 93%] Built target luisa-render-integrator-pssmlt
[ 93%] Built target luisa-render-shape-mesh
[ 93%] Built target luisa-render-integrator-direct
[ 93%] Built target luisa-render-integrator-normal
[ 93%] Built target luisa-render-integrator-megavpt
[ 93%] Built target luisa-render-texture-uvmapping
[ 93%] Built target luisa-render-filter-lanczossinc
[ 93%] Built target luisa-render-texture-constant
[ 93%] Built target luisa-render-camera-thinlens
[ 95%] Built target luisa-render-integrator-megapath
[ 95%] Built target luisa-render-sampler-paddedsobol
[ 95%] Built target luisa-render-surface-null
[ 95%] Built target luisa-render-texture-checkerboard
[ 95%] Built target luisa-render-lightsampler-uniform
[ 95%] Built target luisa-render-surface-layered
[ 95%] Built target luisa-render-integrator-megapm
[ 95%] Linking CXX shared module ../../bin/libluisa-render-texture-nishitasky.so
[ 95%] Linking CXX executable ../../bin/test_sky
[ 96%] Built target luisa-render-spectrum-hero
ld: warning: ignoring duplicate libraries: '-lSystem'
ld: warning: ignoring duplicate libraries: '-lSystem'
[ 96%] Built target luisa-render-texture-nishitasky
[ 96%] Built target test_sky
make: *** [all] Error 2

```

Evgeny105, 2024-12-19:
### How to Resolve Compilation Issues in LuisaRender

If you encounter compilation errors while building **LuisaRender**, follow these steps to resolve the issues:

---

#### 1. **Install Required Dependencies**
First, ensure that the necessary development libraries are installed on your system. Run the following commands:

```bash
sudo apt-get install libimgui-dev
sudo apt-get install libzzip-dev
sudo apt-get install libminizip-dev
```

---

#### 2. **Locate the Installation Path of Dear ImGui**
After installing `libimgui-dev`, find the installation path of the `imgui.h` header file. Run the following command:

```bash
dpkg -L libimgui-dev | grep imgui.h
```

This will output something like:

```
/usr/include/imgui/imgui.h
```

The path `/usr/include/imgui` is the directory where the `imgui.h` header file is located.

---

#### 3. **Update `CMakeLists.txt`**
Open the `CMakeLists.txt` file in your project and add the following line **after** the `project(LuisaRender LANGUAGES C CXX VERSION 0.1)` line:

```cmake
include_directories(/usr/include/imgui)
```

This tells CMake where to find the `imgui.h` header file.

---

#### 4. **Fix Compilation Errors in `display.cpp`**
In the file `src/films/display.cpp`, you may encounter a compilation error related to the `_framebuffer` object. To fix this, add the following lines:

1. Locate the line where `_framebuffer` is created, typically around line 306.
2. Add the following lines:

```cpp
_framebuffer = device.create_image<float>(_window->swapchain().backend_storage(), size);
uint64_t _background; // Added line
_background = _window->register_texture(_framebuffer, TextureSampler::linear_point_zero());
```

This ensures that the `_background` variable is properly initialized and used.

---

#### 5. **Fix Compilation Errors in `nfor.cpp`**
In the file `src/integrators/nfor.cpp`, you may encounter an error related to the `Framerate` type. To fix this, add the following `#include` directive at the top of the file:

```cpp
#include <luisa/gui/framerate.h>
```

This ensures that the `Framerate` type is properly defined and recognized by the compiler.

---

#### 6. **Rebuild the Project**
After making the above changes, rebuild the project using the following commands:

```bash
cmake --build build --clean-first
```

zswang666, 2024-12-20:
Does @Evgeny105 comment or the [doc and fqa](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#photo-realistic-ray-tracing-rendering) fix your issue?

zhuhu00, 2024-12-22:
> Does @Evgeny105 comment or the [doc and fqa](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#photo-realistic-ray-tracing-rendering) fix your issue?

No.. I use Mac, Not Ubuntu..

zhangiser, 2025-01-06:
@Evgeny105 
`CMakeLists.txt` doesn't contains `project(LuisaRender LANGUAGES C CXX VERSION 0.1) line`, So can i add `include_directories(/usr/include/imgui)` at last?
```bash
cmake_minimum_required(VERSION 3.18...3.21)
cmake_policy(VERSION 3.18)

cmake_policy(SET CMP0069 NEW)
cmake_policy(SET CMP0072 NEW)
cmake_policy(SET CMP0091 NEW)

set(CMAKE_POLICY_DEFAULT_CMP0069 NEW)
set(CMAKE_POLICY_DEFAULT_CMP0072 NEW)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_C_STANDARD_REQUIRED ON)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_C_EXTENSIONS OFF)
set(CMAKE_CXX_EXTENSIONS OFF)
set(BUILD_SHARED_LIBS ON)

project(LuisaRender LANGUAGES C CXX VERSION 0.1)

if (NOT CMAKE_SIZEOF_VOID_P EQUAL 8)
    message(FATAL_ERROR "LuisaRender only supports 64-bit platforms")
endif ()

if (NOT DEFINED LUISA_RENDER_MASTER_PROJECT)
    if (CMAKE_CURRENT_SOURCE_DIR STREQUAL CMAKE_SOURCE_DIR)
        set(LUISA_RENDER_MASTER_PROJECT ON)
    else ()
        set(LUISA_RENDER_MASTER_PROJECT OFF)
    endif ()
endif ()

option(LUISA_RENDER_BUILD_TESTS "Build tests for LuisaRender" ${LUISA_RENDER_MASTER_PROJECT})
option(LUISA_RENDER_ENABLE_UNITY_BUILD "Enable unity build to speed up compilation" OFF)

include(src/compute/scripts/setup_output_dirs.cmake)

# rpath
set(CMAKE_MACOSX_RPATH ON)
set(CMAKE_SKIP_BUILD_RPATH OFF)
set(CMAKE_BUILD_RPATH_USE_ORIGIN ON)
set(CMAKE_BUILD_WITH_INSTALL_RPATH ON)
set(CMAKE_INSTALL_RPATH_USE_LINK_PATH ON)

if (APPLE)
    set(CMAKE_INSTALL_RPATH "@loader_path;@loader_path/../bin;@loader_path/../lib")
elseif (UNIX)
    set(CMAKE_INSTALL_RPATH "$ORIGIN;$ORIGIN/../bin;$ORIGIN/../lib")
endif ()

if (NOT CMAKE_INSTALL_PREFIX)
    set(CMAKE_INSTALL_PREFIX "dist")
endif ()

# set up install directories
include(GNUInstallDirs)
if (NOT WIN32)
    # DLLs are installed in the same directory as executables
    set(CMAKE_INSTALL_LIBDIR ${CMAKE_INSTALL_BINDIR})
endif ()

set(CMAKE_FIND_PACKAGE_SORT_ORDER NATURAL)
set(CMAKE_FIND_PACKAGE_SORT_DIRECTION DEC)

add_subdirectory(src)
# add in last?
include_directories(/usr/include/imgui)

```

closed 2025-02-22 by YilingQiao