Genesis-Embodied-AI/genesis-world #43: Win10-WSL2-pyglet.gl.ContextException: Could not create GL context

skyswordx, 2024-12-19:
## Windows-10-WSL2 Environment information

cuda driver in windows version = 12.6
[image]

cuda toolkit version in WSL   
```shell
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Thu_Sep_12_02:18:05_PDT_2024
Cuda compilation tools, release 12.6, V12.6.77
Build cuda_12.6.r12.6/compiler.34841621_0
```
openGL version in WSL
```shell
glxinfo | grep "OpenGL version"
OpenGL version string: 3.1 Mesa 21.2.6

glxinfo | grep "OpenGL"
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce RTX 4060 Ti)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 21.2.6
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 21.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
```
## Problem
When I finish setting up conda virtual environment(pytorch & genesis)  and start to run hello_genesis.py, I encounter the following error message: 

```shell
(genesis) root@L6o0:~/Genesis# /home/circlemoon/miniconda3/envs/genesis/bin/python /home/circlemoon/Genesis/examples/tutorials/hello_genesis.py
[Genesis] [17:20:39] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [17:20:39] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [17:20:39] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [17:20:39] [INFO] Running on [AMD Ryzen 5 7500F 6-Core Processor] with backend gs.cpu. Device memory: 15.44 GB.
[Genesis] [17:20:39] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [17:20:40] [INFO] Scene <5708222> created.
[Genesis] [17:20:40] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <1535f0f>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [17:20:40] [INFO] Preprocessing geom idx 0.
[Genesis] [17:20:40] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <0c70d30>, morph: <gs.morphs.MJCF(file='/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [17:20:40] [INFO] Preprocessing geom idx 1.
[Genesis] [17:20:40] [INFO] Preprocessing geom idx 2.
[Genesis] [17:20:41] [INFO] Preprocessing geom idx 3.
[Genesis] [17:20:41] [INFO] Preprocessing geom idx 4.
[Genesis] [17:20:42] [INFO] Preprocessing geom idx 5.
[Genesis] [17:20:42] [INFO] Preprocessing geom idx 6.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 7.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 8.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 9.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 10.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 11.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 12.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 13.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 14.
[Genesis] [17:20:43] [INFO] Preprocessing geom idx 16.
[Genesis] [17:20:43] [INFO] Building scene <5708222>...
[Genesis] [17:20:51] [INFO] Compiling simulation kernels...
[Genesis] [17:21:14] [INFO] Building visualizer...
Exception in thread Thread-3 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1126, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
    super().__init__(*args, **kwargs)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/window/__init__.py", line 527, in __init__
    context = config.create_context(gl.current_context)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/gl/xlib.py", line 114, in create_context
    return XlibContext(self, share)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/gl/xlib.py", line 149, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context
```

skyswordx, 2024-12-19:
I found this [issue: pyglet.gl.ContextException: Could not create GL context](https://github.com/pyglet/pyglet/issues/768) related to this error message AND tried some test in it

Here is the output when I run `python -m pyglet.info`
It seems that output is fine? 
```shell
python -m pyglet.info
Platform
------------------------------------------------------------------------------
platform:  Linux-5.15.137.3-rt71-yuanyue-edition-WSL2-rt-usbcam-x86_64-with-glibc2.31
release:   5.15.137.3-rt71-yuanyue-edition-WSL2-rt-usbcam
machine:   x86_64

Python
------------------------------------------------------------------------------
implementation: CPython
sys.version: 3.12.8 | packaged by Anaconda, Inc. | (main, Dec 11 2024, 16:31:09) [GCC 11.2.0]
sys.maxint: 9223372036854775807
os.getcwd(): /home/circlemoon/Genesis

pyglet
------------------------------------------------------------------------------
pyglet.version: 2.0.20
pyglet.compat_platform: linux
pyglet.__file__: /home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/__init__.py
pyglet.options['audio'] = ('xaudio2', 'directsound', 'openal', 'pulse', 'silent')
pyglet.options['debug_font'] = False
pyglet.options['debug_gl'] = True
pyglet.options['debug_gl_trace'] = False
pyglet.options['debug_gl_trace_args'] = False
pyglet.options['debug_gl_shaders'] = False
pyglet.options['debug_graphics_batch'] = False
pyglet.options['debug_lib'] = False
pyglet.options['debug_media'] = False
pyglet.options['debug_texture'] = False
pyglet.options['debug_trace'] = False
pyglet.options['debug_trace_args'] = False
pyglet.options['debug_trace_depth'] = 1
pyglet.options['debug_trace_flush'] = True
pyglet.options['debug_win32'] = False
pyglet.options['debug_input'] = False
pyglet.options['debug_x11'] = False
pyglet.options['shadow_window'] = True
pyglet.options['vsync'] = None
pyglet.options['xsync'] = True
pyglet.options['xlib_fullscreen_override_redirect'] = False
pyglet.options['search_local_libs'] = True
pyglet.options['win32_gdi_font'] = False
pyglet.options['headless'] = False
pyglet.options['headless_device'] = 0
pyglet.options['win32_disable_shaping'] = False
pyglet.options['dw_legacy_naming'] = False
pyglet.options['win32_disable_xinput'] = False
pyglet.options['com_mta'] = False
pyglet.options['osx_alt_loop'] = False
pyglet.options['shader_bind_management'] = True

pyglet.window
------------------------------------------------------------------------------
display: <pyglet.canvas.xlib.XlibDisplay object at 0x7f2ae4631e20>
screens[0]: XlibScreen(display=<pyglet.canvas.xlib.XlibDisplay object at 0x7f2ae4631e20>, x=0, y=0, width=1920, height=1080, xinerama=False)
config['double_buffer'] = 1
config['stereo'] = 0
config['buffer_size'] = 32
config['aux_buffers'] = 0
config['sample_buffers'] = 0
config['samples'] = 0
config['red_size'] = 8
config['green_size'] = 8
config['blue_size'] = 8
config['alpha_size'] = 8
[... 122 lines cut ...]
FFmpeg version: 4.2.7-0ubuntu0.1

pyglet.media.drivers.openal
------------------------------------------------------------------------------
Library: <CDLL 'libopenal.so.1', handle 2bcc4b0 at 0x7f2ae3eaa690>
Version: 1.1
Extensions:
   ALC_ENUMERATE_ALL_EXT
   ALC_ENUMERATION_EXT
   ALC_EXT_CAPTURE
   ALC_EXT_DEDICATED
   ALC_EXT_disconnect
   ALC_EXT_EFX
   ALC_EXT_thread_local_context
   ALC_SOFT_device_clock
   ALC_SOFT_HRTF
   ALC_SOFT_loopback
   ALC_SOFT_output_limiter
   ALC_SOFT_pause_device

pyglet.input.wintab
------------------------------------------------------------------------------
WinTab not available.
AL lib: (EE) alc_cleanup: 1 device not closed
```

heyuanYao-pku, 2024-12-19:
I met save issue and solved it with https://github.com/mmatl/pyrender/issues/13#issuecomment-1402054024, which runs conda install -c conda-forge libstdcxx-ng

zhouxian, 2024-12-19:
pyrender has a issue with windows in our setup
see https://github.com/Genesis-Embodied-AI/Genesis/issues/64

skyswordx, 2024-12-19:
## I found the reason of this error in WSL: 
Due to **the mapping NVida drivers to WSL2** **limits** the version of **OpenGL** is **3.1 (If using GPU hardware to render)**
AND pyglet requires OpenGL 3.3+ since v2.0
[image]
 
SO the solution is to **use software render OpenGL** (With **PREFIX**: `LIBGL_ALWAYS_SOFTWARE=1 python xxx.py`

AND In order to enable software rendering, you need to use the following commands
```shell
sudo add-apt-repository ppa:kisak/kisak-mesa (or ppa:oibaf/graphics-drivers)
sudo apt update && sudo apt upgrade -y
```
references: https://github.com/pyglet/pyglet/issues/1047#issuecomment-1947999503

## BUT I encounter another problem

Although I have rendering an OpenGL window, but the content is empty
[image]

and here is the output:
```shell
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 469, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
              ^^^^^^^^^^^^^^^^^^^^^
  File "/home/circlemoon/miniconda3/envs/genesis/lib/python3.12/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context
```

QixuanAI, 2024-12-19:
the same problem

SWivid, 2024-12-19:
> 2. OpenGL.error.Error: Attempt to retrieve context when no valid context

personally solved with https://github.com/Genesis-Embodied-AI/Genesis/issues/37#issuecomment-2553368222

i.e. add 
```
import os
os.environ['PYOPENGL_PLATFORM'] = 'glx'
```

before

```
import genesis
```

AgentOfLyra, 2024-12-20:
I solved the problem, just upgrade opengl, the latest version can be 4.5.
'OpenGL.error.Error: Attempt to retrieve context when no valid context'
```shell
sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt upgrade
``` 
and use direct rendering:
```shell
export LIBGL_ALWAYS_INDIRECT=0
```
now use the command to check your version of opengl:
```shell
glxinfo -B
```
[image]

[image]
Now I can see the scene, but it doesn't appear on the screen rapidly, I don't know why.

skyswordx, 2024-12-21:
> I solved the problem personally(windows 11, wsl2), just by upgrading opengl, the latest version can be 4.5. 'OpenGL.error.Error: Attempt to retrieve context when no valid context'
> 
> ```shell
> sudo add-apt-repository ppa:kisak/kisak-mesa
> sudo apt update
> sudo apt upgrade
> ```
> 
> and use direct rendering:
> 
> ```shell
> export LIBGL_ALWAYS_INDIRECT=0
> ```
> 
> now use the command to check your version of opengl:
> 
> ```shell
> glxinfo -B
> ```

@AgentOfLyra  **Thanks a lot!!!**  your sugestion commands also help me update my OpenGL version to 4.5 in **windows 10, WSL2**

By using `LIBGL_ALWAYS_INDIRECT=0`, I aslo find my scene  doesn't appear on the screen rapidly.
MAYBE it is related whether the rendering using GPU in host?
I also tried @SWivid 's recommendation, ALSO  **Thanks a lot!!! for your sugestions!!!**
> > 1. pyglet.gl.ContextException: Could not create GL context
> 
> rather than software rendering, could just unset or `LIBGL_ALWAYS_SOFTWARE=0`, and alter to use
> 
> ```
> export GALLIUM_DRIVER=d3d12
> export MESA_D3D12_DEFAULT_ADAPTER_NAME=NVIDIA  # if wanna nvidia gpu
> ```

after setting ` GALLIUM_DRIVER=d3d12` and `MESA_D3D12_DEFAULT_ADAPTER_NAME=NVIDIA`
I found my render string is my host GPU AND the scene appears faster (Though there is a short time of Black scene)
```shell
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce RTX 4060 Ti)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.3.2 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
```

closed 2024-12-21 by skyswordx

chenlinzhe, 2024-12-24:
对于4090，也出现这样的问题（1），这样做就解决了:
1.强制指定 OpenGL 的 GLX 实现使用 NVIDIA 的驱动程序
export __GLX_VENDOR_LIBRARY_NAME=nvidia

2.强制渲染交给GPU
export __NV_PRIME_RENDER_OFFLOAD=1

3.强制 Vulkan 应用程序只使用 NVIDIA GPU
export __VK_LAYER_NV_optimus=NVIDIA_only

zhangiser, 2024-12-27:
I'm used win11 wsl ubuntu,run Hello Genesis got Error: CUDA_ERROR_NO_DEVICE:
```txt
(torch) root@zhangiser-pc:/usr/Genesis-main# /root/miniconda3/envs/torch/bin/python /usr/Genesis-main/examples/tutorials/hello_genesis.py
[Genesis] [09:35:05] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [09:35:05] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [09:35:05] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [09:35:06] [INFO] Running on [NVIDIA GeForce RTX 4080 Laptop GPU] with backend gs.cuda. Device memory: 11.99 GB.
[E 12/27/24 09:35:07.263 1371] [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected while calling init (cuInit)

Traceback (most recent call last):
  File "/usr/Genesis-main/examples/tutorials/hello_genesis.py", line 3, in <module>
    gs.init(backend=gs.cuda)
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/__init__.py", line 97, in init
    ti.init(arch=TI_ARCH[platform][backend], debug=debug, force_scalarize_matrix=True)
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/taichi/lang/misc.py", line 458, in init
    impl.get_runtime().create_program()
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/taichi/lang/impl.py", line 388, in create_program
    self.prog = _ti_core.Program()
RuntimeError: [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected while calling init (cuInit)
[Genesis] [09:35:07] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

wsl info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# wsl -l -v
  NAME                   STATE           VERSION
* docker-desktop-data    Stopped         2
  docker-desktop         Stopped         2
  Ubuntu-22.04           Running         2
```
CUDA info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Thu_Mar_28_02:18:24_PDT_2024
Cuda compilation tools, release 12.4, V12.4.131
Build cuda_12.4.r12.4/compiler.34097967_0
```
nvidia-smi info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# nvidia-smi
Fri Dec 27 17:07:40 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 565.77.01              Driver Version: 566.36         CUDA Version: 12.7     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4080 ...    On  |   00000000:01:00.0 Off |                  N/A |
| N/A   48C    P0            593W /  155W |       0MiB /  12282MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```
Pytorch info:
```txt
(base) root@zhangiser-pc:/usr/Genesis-main# conda activate torch
(torch) root@zhangiser-pc:/usr/Genesis-main# python
Python 3.9.21 (main, Dec 11 2024, 16:24:11) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> torch.cuda.is_available()
True
```
Everying seems OK! But why got CUDA_ERROR_NO_DEVICE ? Is there anyone know this problem?

chenlinzhe, 2024-12-27:
[image]
WSL2不支持CUDA-OPENGL,要用软件渲染，或别的LINUX系统。

huang-xiansheng, 2024-12-30:
For the second problem , OpenGL.error.Error: Attempt to retrieve context when no valid context , you may try to run this command , "export PYOPENGL_PLATFORM='glx' " .

Since the latest version of Ubuntu on WSL2 has chosen by default the Wayland that pyglet does not support yet , utilizing pyglet separately is available but is invalid with Wayland , so the graphics protocol should be set explicitly.

Max-promax, 2025-02-14:
[Genesis] [13:54:54] [WARNING] Windows system detected. Viewer may have some issues.
set_primitive: 60
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "E:\Anaconda\envs\Genesis\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "E:\Anaconda\envs\Genesis\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "E:\Anaconda\envs\Genesis\lib\site-packages\genesis\ext\pyrender\viewer.py", line 1168, in _init_and_start_app
    pyglet.clock.tick()
  File "E:\Anaconda\envs\Genesis\lib\site-packages\pyglet\clock.py", line 528, in tick
    return _default.tick(poll)
  File "E:\Anaconda\envs\Genesis\lib\site-packages\pyglet\clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "E:\Anaconda\envs\Genesis\lib\site-packages\pyglet\clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "E:\Anaconda\envs\Genesis\lib\site-packages\genesis\ext\pyrender\viewer.py", line 955, in _time_event
    self.on_draw()
  File "E:\Anaconda\envs\Genesis\lib\site-packages\genesis\ext\pyrender\viewer.py", line 654, in on_draw
    self._render()
  File "E:\Anaconda\envs\Genesis\lib\site-packages\genesis\ext\pyrender\viewer.py", line 1105, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "E:\Anaconda\envs\Genesis\lib\site-packages\genesis\ext\pyrender\renderer.py", line 173, in render
    self._shadow_mapping_pass(scene, ln, flags, env_idx=env_idx)
  File "E:\Anaconda\envs\Genesis\lib\site-packages\genesis\ext\pyrender\renderer.py", line 655, in _shadow_mapping_pass
    self.jit.shadow_mapping_pass(self, V, P, flags, ProgramFlags.NONE, env_idx=env_idx)
TypeError: SimpleNonJITRenderer.shadow_mapping_pass() got an unexpected keyword argument 'env_idx'

直接再windows  下的 conda里面跑这个  为啥会出现这个呀

hyf-5979, 2025-02-19:
> [Genesis] [13:54:54] [WARNING] Windows system detected. Viewer may have some issues. set_primitive: 60 Exception in thread Thread-2 (_init_and_start_app): Traceback (most recent call last): File [...]

请问你解决了吗？我也出现了同样的问题

NKgithubET, 2025-05-08:
circLΣMoon
skyswordx · wwv

your suggestion helped me fix the problem in fsl 
i was using xlaunch for fmri brain scan images and the glxgears works when i used export DISPLAY=$(hostname).local:0 for the display but it was literaaly stuck it was not rotating at all 
and the fselyes command tries to open but it says cant create open gl context 
but after your suggestion i was able to fix it thank you very much for your help

duburcqa, 2025-05-10:
> but after your suggestion i was able to fix it thank you very much for your help

Can you share a bit more details so that it could be added to the documentation ?

NKgithubET, 2025-05-10:
[duburcqa](https://github.com/duburcqa)

I installed wsl 2 using powershell then installed ubuntu 22.04 and it installed cleanly then i installed FSL(a comprehensive library of analysis tools for FMRI, MRI and diffusion brain imaging data) I followed this link https://fsl.fmrib.ox.ac.uk/fsl/docs/#/install/index then i tried some commands they were not working so i edited the path from the trouble shooting section from the link given above.The commnds started to work.

Following the verify the installation section i tried the "glxgears " command the gears appeared but there were rotating extemely , it can be said they were not rotating at all. Then i tried the "fsl &" command , this command also worked and open a minin GUI.

I installed and launched the Xlaunch application(x-server) by following from the link given above

Finally i tried the "fsleyes -std &" command it tries to open the larger GUI but is gives out a pop up notification which says "failed to creatre opengl context".After trouble shooting i found out soft pipe rendering was being used instead of my machines GPU, so i tries to mannualy set the $LIBGL_ALWAYS_SOFTWARE to zero but that doest work at all .Finally after installing the suggested Packages from @AgentOfLyra 

sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt upgrade

then export LIBGL_ALWAYS_INDIRECT=0  then checking glxinfo -B i was able to see my machine does not use soft pipe any more it detected my machines GPU and when i tried "fsleyes -std &" command it worked perfectly 

@AgentOfLyra  thank you very much , the suggestion helped a lot

duburcqa, 2025-05-11:
@NKgithubET Thank you for your report. This was helpful!