Genesis-Embodied-AI/genesis-world #44: AttributeError: module 'LuisaRenderPy' has no attribute 'create_scene'

saprrow, 2024-12-19:
[image]

When running **gs.renderers.RayTracer()**, I get an error saying that module 'LuisaRenderPy' has no attribute 'create_scene'. However, I cannot find the 'create_scene' API in LuisaRenderPy's pybind API code.

zhouxian, 2024-12-19:
@YilingQiao could this be due to a commit mismatch?

YilingQiao, 2024-12-19:
I changed the commit number. Can you try again?

Xu-Xuanhui, 2024-12-20:
@YilingQiao 
I built Alif-01'LuisaRender fork. However,  I cannot find the 'create_scene' API in LuisaRenderPy's pybind API code too. 

 print(dir(LuisaRenderPy))
['Camera', 'Integrator', 'LogLevel', 'Shape', 'Spectrum', 'Surface', 'Texture', 'Transform', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', 'add_emission', 'add_environment', 'add_surface', 'destroy', 'init', 'render_frame', 'update_camera', 'update_shape']
It seems that LuisaRenderPy don't have 'create_scene' ?

Xu-Xuanhui, 2024-12-20:
I have install the LuisaRender
Ubuntu 22.04 Nvidia driver 535.183.01,cuda 12.2! I update the cuda! python 3.10
git clone https://github.com/Genesis-Embodied-AI/Genesis.git
cd Genesis
git submodule update --init --recursive

You should follow the user guider to install the Dependencies.

cd genesis/ext/LuisaRender
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.10 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF 
cmake --build build -j $(nproc)

Finall I copy the etx folder to my conda envs.
It works!

zswang666, 2024-12-20:
Yet again thanks @Xu-Xuanhui  :D 

doc [here](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#photo-realistic-ray-tracing-rendering)

closed 2024-12-20 by zswang666