Genesis-Embodied-AI/genesis-world #49: windows run go2 train example gives opengl error

rocketman123456, 2024-12-19:
running `python examples/locomotion/go2_train.py` on win11 with conda+python3.10 gives error:

```
Traceback (most recent call last):
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_train.py", line 168, in <module>
    main()
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_train.py", line 155, in main
    env = Go2Env(num_envs=args.num_envs, env_cfg=env_cfg, obs_cfg=obs_cfg, reward_cfg=reward_cfg, command_cfg=command_cfg)
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_env.py", line 68, in __init__
    self.scene.build(n_envs=num_envs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
    main()
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_train.py", line 155, in main
    env = Go2Env(num_envs=args.num_envs, env_cfg=env_cfg, obs_cfg=obs_cfg, reward_cfg=reward_cfg, command_cfg=command_cfg)
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_env.py", line 68, in __init__
    self.scene.build(n_envs=num_envs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_train.py", line 155, in main
    env = Go2Env(num_envs=args.num_envs, env_cfg=env_cfg, obs_cfg=obs_cfg, reward_cfg=reward_cfg, command_cfg=command_cfg)
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_env.py", line 68, in __init__
    self.scene.build(n_envs=num_envs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
    env = Go2Env(num_envs=args.num_envs, env_cfg=env_cfg, obs_cfg=obs_cfg, reward_cfg=reward_cfg, command_cfg=command_cfg)
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_env.py", line 68, in __init__
    self.scene.build(n_envs=num_envs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
  File "E:\LibraryCode\Simulation\Genesis\examples\locomotion\go2_env.py", line 68, in __init__
    self.scene.build(n_envs=num_envs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
    return method(self, *args, **kwargs)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
    self._visualizer.build()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self._context.build(self._scene)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.jit = JITRenderer(self._scene, [], [])
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
    self.gen_func_ptr()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
    self.gl = GLWrapper()
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
    func_ptr = GL.platform.ctypesloader.buildFunction(
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
  File "C:\Users\ASUS\.conda\envs\genesis\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
    return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found
```

ByronLaw1025, 2024-12-19:
Same issue. Waiting for solution. Thx.

Nebuluss, 2024-12-19:
I’ve been facing issues with glutInit() failing to initialize and encountering errors like AttributeError: function 'glGetUniformLocation' not found. Despite numerous attempts to resolve the problem, I haven’t been able to make progress. Here’s a detailed summary of what I’ve tried so far:

Initial Error:
The first issue I encountered was:

vbnet
Copier le code
AttributeError: function 'glGetUniformLocation' not found
This error seemed related to OpenGL failing to initialize properly. Later, attempts to resolve this led to another error indicating that opengl32.dll could not be loaded:

arduino
Copier le code
FileNotFoundError: Could not find module 'C:\Users\Neb\anaconda3\envs\Genesis\Library\bin\opengl32.dll' (or one of its dependencies).
Steps I’ve Taken:
Python Environment Setup:

Confirmed my Python installation is 64-bit (platform.architecture() outputs 64bit, WindowsPE).
Using Anaconda with a virtual environment named Genesis.
Mesa3D Deployment:

Downloaded the latest Mesa3D release (v24.3.1).
Deployed opengl32.dll from the x64 folder into:
C:\Users\Neb\anaconda3\envs\Genesis\Library\bin
Also tested with system-wide deployment by copying to C:\Windows\System32.
Environment PATH:

Added C:\Users\Neb\anaconda3\envs\Genesis\Library\bin to the PATH variable to ensure Python prioritizes the correct opengl32.dll.
FreeGLUT:

Verified freeglut.dll exists in C:\Users\Neb\anaconda3\envs\Genesis\Library\bin.
Reinstalled FreeGLUT using conda install -c conda-forge freeglut.
Dependency Resolution:

Used Dependency Walker to check opengl32.dll and freeglut.dll:
All dependencies for opengl32.dll seem resolved.
Universal C Runtime (UCRT) is installed (conda install -c conda-forge ucrt).
Debugging Attempts:

Ran PYOPENGL_DEBUG=1 to capture additional error messages.
Errors indicated issues with glGetUniformLocation and later:
vbnet
Copier le code
ImportError: Unable to load OpenGL library (or one of its dependencies).
Current Errors:
Despite these efforts, I am still seeing the following errors depending on the context:

AttributeError: function 'glGetUniformLocation' not found during OpenGL function calls.
FileNotFoundError: Could not find module 'C:\Users\Neb\anaconda3\envs\Genesis\Library\bin\opengl32.dll' when initializing OpenGL.
Environment Details:
OS: Windows 11
Python: 3.10 (64-bit)
OpenGL: Mesa3D v24.3.1
GPU: NVIDIA RTX 3090 (driver 566.03)
What I Need Help With:
Why glGetUniformLocation is failing to load.
Why Python cannot find opengl32.dll even when it’s present in C:\Users\Neb\anaconda3\envs\Genesis\Library\bin.
Any guidance on resolving these issues would be greatly appreciated. Please let me know if additional logs or details are needed.

zhouxian, 2024-12-19:
see https://github.com/Genesis-Embodied-AI/Genesis/issues/64

As suggested here: https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html
rendering on windows is not working yet due to gl error :(

Would appreciate if someone could help

ziyanx02, 2024-12-27:
Close as a duplicate.

closed 2024-12-27 by ziyanx02