Genesis-Embodied-AI/genesis-world #50: Need docker

yytang0204, 2024-12-19:
Really need a docker to run this on linux. Taichi is hard to run naturally

zhouxian, 2024-12-19:
@Kashu7100

Kashu7100, 2024-12-19:
I will create PR

yytang0204, 2024-12-20:
> is there any problem with pip install taichi?
> 
> @ Kashu7100 could you help with this?

Basically my offline enviroment is messy since many people work on it. So I want to use it on cloud. A docker file will be best.

antoinebou12, 2024-12-21:
https://github.com/ETSTribology/Genesis

Still working on it

Riften, 2024-12-21:
I make it work with following Dockerfile. It does not build genesis from source. But install genesis through pip and install all dependencies.

```Dockerfile
FROM pytorch/pytorch:2.5.1-cuda12.4-cudnn9-devel
ENV DEBIAN_FRONTEND=noninteractive
# install basic tools
RUN apt-get update && apt-get install -y git wget curl \
        vim tar xz-utils

# install python package genesis
RUN pip install genesis-world
# install pillow for image processing
RUN pip install pillow

# install OpenGL
RUN apt-get update && apt-get install -y \
       libglvnd0 libgl1 libglx0 libegl1 libgles2 

# fix missing libraries
RUN apt-get update && apt-get install -y libxrender1 libglib2.0-0

# fix pyrender
# install ffmpeg
# https://github.com/Genesis-Embodied-AI/Genesis/pull/27
RUN apt-get update && apt-get install -y ffmpeg
# install pyrender
RUN pip install pyrender
RUN pip install PyOpenGL==3.1.7
# note that there would be a conflict between PyOpenGL and pyrender.
# but we still need PyOpenGL 3.1.7 refer to https://github.com/Genesis-Embodied-AI/Genesis/issues/74
# PyOpenGL 3.1.7 raise error when using pyrender even if i build pyrender from source
# fix glGetError refer to: https://github.com/MPI-IS/mesh/issues/23
RUN apt-get install -y python3-opengl
```

Kashu7100, 2024-12-25:
Sorry for taking some time. Finally created a PR with Luisa support. 
https://github.com/Genesis-Embodied-AI/Genesis/pull/303

closed 2025-02-22 by YilingQiao