Genesis-Embodied-AI/genesis-world #51: Input Error occured when import genesis

xiaopengyou0000, 2024-12-19:
Exception has occurred: ValueError
numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
  File "/home/xxx/project/gennesis-world/test_genesis_world.py", line 16, in <module>
    import genesis as gs
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject

lupinjia, 2024-12-19:
In my case, this error is caused by `tetgen` package, which imports _tetgen in its `__init__.py`.

nchuly, 2024-12-19:
same problem

xiaopengyou0000, 2024-12-19:
Problem solved:
 The numpy version in this project is 1.26.4

xiaopengyou0000, 2024-12-19:
You just need to reinstall numpy,  pip install numpy=1.26.4, also same problem for tetgen, genesis-world 0.2.0 requires tetgen==0.6.4.

zhouxian, 2024-12-19:
yeah sorry about that. numpy 2.0 introduced a bunch of compatibilty issues that would make some of our dependencies break...

ChengYaofeng, 2024-12-20:
> Problem solved: The numpy version in this project is 1.26.4

I tried numpy==1.25.0, this version is ok, just try `pip install numpy==1.25.0`

closed 2025-02-22 by YilingQiao