Genesis-Embodied-AI/genesis-world #52: can't install due to pymeshlab not available of M1

PawitKoch, 2024-12-19:
Chip: Apple M1 Pro
Metal 3

`pip install genesis-world` fails due to:

```
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on pymeshlab
    genesis-world 0.1.1 depends on pymeshlab
```

whereby trying to install `pymeshlab` via `pip install pymeshlab` gets me:

```
ERROR: Could not find a version that satisfies the requirement pymeshlab (from versions: none)
ERROR: No matching distribution found for pymeshlab
```

danhussey, 2024-12-19:
Same. Pymeshlab seems to actually just not support Apple silicon.

danhussey, 2024-12-19:
I had the same issue, but downgraded python to v3.11 and `pip install` worked after that

illiafox, 2024-12-19:
@danhussey to which exact 3.11.x version did you downgrade? 

```bash
$ python3 --version
Python 3.11.11

$ pip install genesis-world
....
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on pymeshlab
    genesis-world 0.1.1 depends on pymeshlab

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflict
```

zhouxian, 2024-12-19:
@zhenjia-xu

danhussey, 2024-12-20:
Just thought I'd add this is my environment I've got a few examples working in:

[environment.yml.txt](https://github.com/user-attachments/files/18205696/environment.yml.txt)

Note python is v3.11.11

fbaptista, 2024-12-23:
same issue here, unable to install on python 3.9.6 macARM

ziyanx02, 2024-12-24:
The issue is caused by `pymeshlab` not being automatically installable in Python version 3.9.
If you encounter the same problem with another Python version, please try using a different version.

closed 2024-12-24 by ziyanx02

fbaptista, 2024-12-26:
Ok, given that, it will be nice to update genesis documentation about minimum Python version supported.

zacamaso, 2025-01-21:
Worked with python 3.10 on Mac M1. 3.9 and 3.11 ran into these kinds of dependency errors