Genesis-Embodied-AI/genesis-world #6: Timeline on releasing the generative framework

chanhee-luke, 2024-12-19:
Hi team,

I am excited to see Genesis released! I briefly played with it and it is so easy to use and fast.
I am interested in the generative framework. Do you have a timeline for when it will be available?

Thank you!

Best,
Luke Song

twinsant, 2024-12-19:
Demo 式开源吗？

ziyanx02, 2024-12-22:
Providing a detailed timeline is challenging due to limited contributors, but stay tuned. We will keep you informed by updating this issue as soon as we make any progress.

wyahnke, 2024-12-23:
How can we sign up to the wait list for the generative features?

PyroFilmsFX, 2025-01-02:
Yeah this was quite the misrepresentation to not once mention that that portion is unreleased. But also would like to be added to a waitlist.

gauravdhiman, 2025-01-08:
Thanks for creating this issue. Is there a way to add ourself to waitlist or get notified when it gets released ?

PyroFilmsFX, 2025-01-17:
@ziyanx02 hello?

PyroFilmsFX, 2025-02-16:
@ziyanx02 hello?

phazei, 2025-03-19:
Monthly bump ;)  Can't wait...

NashvilleBrandon, 2025-04-02:
👀