Genesis-Embodied-AI/genesis-world #61: Conflict between numpy and tetgen

BryceWang0527, 2024-12-19:
When run the hello Genesis, get the following error:
```
tetgen._tetgen
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
```

Following the instructions, the env default numpy and tetgen version is:

> tetgen == 0.6.4     
> numpy == 1.26.4

It seems work when downgrade NumPy to version 1.24.0

zswang666, 2024-12-19:
Hmm...weird. We tested on the exact same versioning (`tetgen==0.6.4` and `numpy==1.26.4`) and there is no error.  Also, a fresh install of the two packages and no issue (even using different machines ubuntu 22.04 and 20.04). Another random test with a combination of `tetgen==0.6.5` and `numpy==2.1.1`, and it worked (side note: not recommending numpy 2 though as I remember there are some issues -- not 100 positive). 

Not sure if this matters but what python version are you using and how did you set up your environment?

aemarkov, 2024-12-19:
Have the same issue.
Distro: Manjaro, kernel 5.10.226
Virtual env manager: mamba 1.5.5, conda 23.11.0
Pip: 24.3.1
Python: 3.12
numpy: 1.26.4, channel pypi
tetgen: 0.6.4, channel pypi

Tried install numpy==1.24.0, but genesis-world requires 1.26.0

hlchen1043, 2024-12-19:
Have the same issue.
Distro: Ubuntu 22.04.05
Virtual env manager: conda 24.5.0
Pip: 24.3.1
Python: 3.10
numpy: 1.26.4, channel pypi
tetgen: 0.6.4, channel pypi

Does not work with numpy 1.24.0, error: 
`Traceback (most recent call last):
  File "/mnt/ssd1/ProvingGrounds/genesis/helloworld.py", line 1, in <module>
    import genesis as gs
  File "/home/user/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/__init__.py", line 262, in <module>
    from .utils.geom import *
  File "/home/user/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/utils/geom.py", line 4, in <module>
    from scipy.spatial.transform import Rotation
  File "/home/user/anaconda3/envs/genesis/lib/python3.10/site-packages/scipy/spatial/__init__.py", line 110, in <module>
    from ._kdtree import *
  File "/home/user/anaconda3/envs/genesis/lib/python3.10/site-packages/scipy/spatial/_kdtree.py", line 4, in <module>
    from ._ckdtree import cKDTree, cKDTreeNode
  File "_ckdtree.pyx", line 11, in init scipy.spatial._ckdtree
  File "/home/user/anaconda3/envs/genesis/lib/python3.10/site-packages/scipy/sparse/__init__.py", line 294, in <module>
    from ._csr import *
  File "/home/user/anaconda3/envs/genesis/lib/python3.10/site-packages/scipy/sparse/_csr.py", line 11, in <module>
    from ._sparsetools import (csr_tocsc, csr_tobsr, csr_count_blocks,
ImportError: numpy._core.multiarray failed to import`

Jianghanxiao, 2024-12-19:
install torch first with pip and pip uninstall numpy, then install gensis. This works for me

yu-mei, 2024-12-20:
> install torch first with pip and pip uninstall numpy, then install gensis. This works for me

This works for me as well. My env version is also tetgen==0.6.4 and numpy==1.26.4.

BryceWang0527, 2024-12-20:
> install torch first with pip and pip uninstall numpy, then install gensis. This works for me

Thanks for the reply, maybe the installation order matters, cause my env install genesis first

achaosss, 2024-12-23:
> install torch first with pip and pip uninstall numpy, then install gensis. This works for me

Thanks，it works for me.

sdwyc, 2024-12-25:
> install torch first with pip and pip uninstall numpy, then install gensis. This works for me

Yes! Thanks for your answer. This is also work for me.

BoruiHe, 2025-01-20:
I am not sure if the tool for installation matters since I installed Pytorch with conda and did NOT uninstall Numpy before installing genesis-world. This will throw an importError caused by tetgen.

> install torch first with pip and pip uninstall numpy, then install gensis. This works for me

Confirm this works for Windows 11 + Python 3.9.21 and hope headless rendering will work for Windows soon.

closed 2025-02-22 by YilingQiao