Genesis-Embodied-AI/genesis-world #62: Maybe it would be good to support simple mechanical structure development?

win10ogod, 2024-12-19:
Maybe it would be good to support simple mechanical structure development?

zhouxian, 2024-12-19:
Could you be more specific or share some reference?

win10ogod, 2024-12-19:
> Could you be more specific or share some reference?

Use case:
Train artificial intelligence to generate mechanical structures! (Complex examples: training generative models, generating electric generators, or designing robotic arms)
Simple example:
training generative models, generating new rocket nozzles, or exoskeleton structures (maybe complex?)

zswang666, 2024-12-19:
Just to make sure I understand correctly. You meant doing mechanical or robot design right? If so, there should be two major parts of it: (i) the optimization algorithm like topology optimization, evolution strategy, differentiable physics, etc and (ii) the simulation infrastructure to allow efficient evaluation (like parallelism), include interface for design, or provide gradients. The first point is sorta out of the scope of Genesis but the second one is on the plate. Stay tuned.

Some of our previous works did computational design for soft robots like [this](https://arxiv.org/pdf/2303.09555) and [this](https://arxiv.org/pdf/2311.17053). Genesis should be sufficient to replicate some simpler algorithms (at least in the first work), e.g., using differentiable physics to optimize the geometry of a MPM body via a mask that represents the "existence" of a particle. For rigid body, there is an internal project working on it and may integrate the infra part once it's done.

JonathanKuelz, 2025-01-16:
Hey @zswang666 , thank you for the details provided.

I don't know if what you're referring to corresponds to the initial authors request, but I would be interested in genesis' capabilities as a tool for computational design of rigid body mechanisms, similar to the soft robot works you mentioned. The genesis roadmap lists differentiable rigid body simulation as "in progress". If I understand you correctly, there's some non-public version of this already. Are there any updates on a possible release date and are you currently looking for contributors in this area?

YilingQiao, 2025-02-22:
Yes, we definitely welcome contributors for differentiable rigid body simulation. We have some unfinished code for a differentiable rigid body simulator, but it might take significant effort to complete and make it robust. We welcome PRs if you have any ideas or prototype demos for it.

closed 2025-02-22 by YilingQiao