Genesis-Embodied-AI/genesis-world #63: TypeError: 'NoneType' object is not callable

Han230104, 2024-12-19:
```
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 916, in _update_context
    texture._add_to_context()
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/ext/pyrender/texture.py", line 219, in _add_to_context
    max_aniso = glGetFloat(texture_filter_anisotropic.GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT)
File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/OpenGL/wrapper.py", line 678, in wrapperCall
    raise err
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/OpenGL/wrapper.py", line 671, in wrapperCall
    result = wrappedOperation( *cArguments )
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
  File "/home/ubuntu15/anaconda3/envs/genesis/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1280,
        description = b'invalid enumerant',
        baseOperation = glGetFloatv,
        pyArgs = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                <object object at 0x7f0fd69868b0>,
        ),
        cArgs = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                array([0.], dtype=float32),
        ),
        cArguments = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                array([0.], dtype=float32),
        )
)

```

Has anyone encountered this error? 
Does anyone know how to solve it?
Thanks a lot !!!

LChen-99, 2024-12-19:
I get same error

TomatoPicker278, 2024-12-19:
yes , i face the same problems

huayong, 2024-12-19:
Same problem, solved by reinstall nvidia driver

wwdok, 2024-12-19:
> Same problem, solved by reinstall nvidia driver

Do you upgrade the nvidia driver version ? and what is your nvidia driver version ?

futuyao, 2024-12-20:
Same issue here. Driver version 550.90.07, CUDA 11.8.

Ye-2077, 2024-12-21:
@huayong Hi, I faced the same issue and solved by setting the nvidia gpu to 'Perfomance Mode' in NVIDIA Settings.
```
$ nvidia-settings
```
[image]

The main cause of my issue is that OpenGL uses integrated graphics by default when NVIDIA graphics is set to 'On-Demand' mode.

You can check if OpenGL is using the correct graphics by:
```
$ glxinfo | grep "OpenGL"
```
For the correct case, you will see something like
```
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 4090/PCIe/SSE2
...
```
Here is my environment
- Ubuntu 22.04
- python 3.9
- RTX4090
- CUDA 11.8
- Driver version 550.90.07

closed 2025-02-22 by YilingQiao