Genesis-Embodied-AI/genesis-world #64: AttributeError: function 'glGetUniformLocation' not found

BlackTea-c, 2024-12-19:
[Genesis] [21:18:17] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [21:18:17] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [21:18:17] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [21:18:17] [INFO] Running on [AMD64 Family 25 Model 68 Stepping 1, AuthenticAMD] with backend gs.cpu. Device memory: 15.26 GB.
[Genesis] [21:18:18] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [21:18:19] [INFO] Scene <1002ad2> created.
[Genesis] [21:18:19] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <f270438>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [21:18:19] [INFO] Preprocessing geom idx 0.
[Genesis] [21:18:19] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <14f978d>, morph: <gs.morphs.MJCF(file='D:\PaperReview\venv\lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [21:18:19] [INFO] Preprocessing geom idx 1.
[Genesis] [21:18:20] [INFO] Preprocessing geom idx 2.
[Genesis] [21:18:21] [INFO] Preprocessing geom idx 3.
[Genesis] [21:18:21] [INFO] Preprocessing geom idx 4.
[Genesis] [21:18:22] [INFO] Preprocessing geom idx 5.
[Genesis] [21:18:23] [INFO] Preprocessing geom idx 6.
[Genesis] [21:18:23] [INFO] Preprocessing geom idx 7.
[Genesis] [21:18:23] [INFO] Preprocessing geom idx 8.
[Genesis] [21:18:23] [INFO] Preprocessing geom idx 9.
[Genesis] [21:18:23] [INFO] Preprocessing geom idx 10.
[Genesis] [21:18:24] [INFO] Preprocessing geom idx 11.
[Genesis] [21:18:24] [INFO] Preprocessing geom idx 12.
[Genesis] [21:18:24] [INFO] Preprocessing geom idx 13.
[Genesis] [21:18:24] [INFO] Preprocessing geom idx 14.
[Genesis] [21:18:24] [INFO] Preprocessing geom idx 16.
[Genesis] [21:18:24] [INFO] Building scene <1002ad2>...
[Genesis] [21:19:01] [INFO] Compiling simulation kernels...
[Genesis] [21:21:22] [INFO] Building visualizer...
Traceback (most recent call last):
  File "D:\PaperReview\Simulate.py", line 10, in <module>
    scene.build()
  File "D:\PaperReview\venv\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "D:\PaperReview\venv\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "D:\PaperReview\venv\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "D:\PaperReview\venv\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "D:\PaperReview\venv\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "D:\PaperReview\venv\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
  File "D:\PaperReview\venv\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "D:\PaperReview\venv\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
  File "D:\PaperReview\venv\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found
[Genesis] [21:21:23] [INFO] 💤 Exiting Genesis and caching compiled kernels...

Process finished with exit code 1

JustLuoxi, 2024-12-19:
Same error using Windows

zhouxian, 2024-12-19:
As suggested here: https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html
rendering on windows is not working yet due to gl error :(

Would appreciate if someone could help

barius, 2024-12-20:
same issue here: https://github.com/Genesis-Embodied-AI/Genesis/issues/12

darrendsy, 2024-12-20:
Same issue here with win11

dLeav, 2024-12-20:
Same issue here with windows 11. Any help would be appreciated.

FlopCoat, 2024-12-20:
same issue, win11

zxsedas, 2024-12-21:
same issue, win11

vstavros02, 2024-12-27:
Is there still no solution?

ziyanx02, 2024-12-27:
Close as a duplicate of #12.

closed 2024-12-27 by ziyanx02

interpidtjuniversity, 2025-01-01:
not supported yet
https://blog.csdn.net/xiangshangdemayi/article/details/144703710

AliBehdar, 2025-04-19:
Same error using Windows