Genesis-Embodied-AI/genesis-world #65: Is there an interface for computing Forward Kinematics in Genesis?

yilin404, 2024-12-19:
I hope you are doing well.
I am currently working with the Genesis framework for robotic simulations, and I am trying to compute the forward kinematics (FK) of a robotic arm. I have reviewed the documentation but could not find any direct mention of an interface or function for performing forward kinematics calculations.
Could you kindly confirm whether Genesis provides an API or method to compute forward kinematics? If such functionality exists, could you please share any relevant documentation, code examples, or guidelines to help integrate this into my project?
Thank you very much for your time and assistance. I look forward to your response.

zhouxian, 2024-12-19:
Set the joint position via `entity.set_dofs_position()` and get the corresponding link pose. Is this what you are looking for or you are asking for something else?

yilin404, 2024-12-19:
Thank you for your response and assistance!

I have a follow-up question regarding the entity.set_dofs_position() function. Specifically, I would like to know whether calling this function will change the joint positions of the object in the simulation environment.

My goal is to compute forward kinematics without altering the joint positions in the simulation. Could you kindly clarify if using entity.set_dofs_position() will affect the joint configurations in the environment, or if there is a way to calculate forward kinematics without modifying the actual joint positions?

I appreciate your help and look forward to your response.

yilin404, 2024-12-19:
Thank you for your response and assistance!
I have a follow-up question regarding the `entity.set_dofs_position()` function. Specifically, I would like to know whether calling this function will change the joint positions of the object in the simulation environment.
My goal is to compute forward kinematics without altering the joint positions in the simulation. Could you kindly clarify if using `entity.set_dofs_position()` will affect the joint configurations in the environment, or if there is a way to calculate forward kinematics without modifying the actual joint positions?
I appreciate your help and look forward to your response.

ziyanx02, 2024-12-22:
Forward kinematics runs in rigid solver, typically when the solver steps.
I'm not sure if I fully understand your point, but I hope this helps:
https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L1967

yilin404, 2024-12-24:
Thank you for the clarification and the link! I appreciate your input.

I have a follow-up question: Does Genesis have an interface similar to Sapien's Pinocchio integration? Specifically, I am wondering if there’s a way to compute forward kinematics directly, given the model of an object, without relying on the simulation process.

Looking forward to your insights!

YilingQiao, 2025-02-22:
The current forward kinematics does not rely on force-related computations, if that’s what you mean by the "simulation process."

closed 2025-02-22 by YilingQiao