Genesis-Embodied-AI/genesis-world #66: I encountered an error related to the Taichi version during installation. The highest version of Taichi I can install is 1.7.1, and the Python version I am using is 3.11.11.

Yutong2022, 2024-12-19:
Collecting genesis-world
  Using cached genesis_world-0.2.0-py3-none-any.whl.metadata (11 kB)
Collecting psutil (from genesis-world)
  Using cached psutil-6.1.0-cp36-abi3-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (22 kB)
Collecting scikit-image (from genesis-world)
  Downloading scikit_image-0.25.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
Collecting genesis-world
  Using cached genesis_world-0.1.1-py3-none-any.whl.metadata (11 kB)
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on taichi==1.7.2
    genesis-world 0.1.1 depends on taichi==1.7.2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

miamia0, 2024-12-19:
I encountered the same issue and successfully installed taichi after switching to python 3.9.6.

zhouxian, 2024-12-19:
We didn't encounter this when testing. @zhenjia-xu Could you take a look?

rootglitch, 2024-12-19:
I had the same issue with python 3.13.1. Switched to 3.9.6 and it installed successfully.

nediol, 2024-12-19:
Switched to 3.10 and taichi installed successfully

edmacdonald, 2024-12-19:
I had the same error with `3.13.1` but not with `3.12.8`

samuell, 2024-12-19:
> I had the same error with `3.13.1` but not with `3.12.8`

Thanks, solved it for me too. Using pixi.sh and:

```bash
pixi init
pixi add python=3.12.8
pixi add pip
pip install genesis-world
```

... is what worked for me.

(I'm on Linux Mint 21.1)

weijing-tao, 2024-12-20:
I have the same issue, tried with python 3.11.11, 3.12.8, 3.9.6, 3.10.16, 3.11.10, unfortunately all do not work :(

zhijian11, 2024-12-20:
I think it might be due to the version of glibc. I resolved it by downloading the .whl and renaming it for installation.

dmattox10, 2024-12-21:
> > I had the same error with `3.13.1` but not with `3.12.8`
> 
> Thanks, solved it for me too. Using [pixi](https://pixi.sh/) and:
> 
> ```shell
> pixi init
> pixi add python=3.12.8
> pixi add pip
> pixi shell
> pip install genesis-world
> ```
> 
> ... is what worked for me.
> 
> (I'm on Linux Mint 21.1)

Can confirm these steps worked on an M3 Mac, starting with:
$ brew install pixi

yriuss, 2024-12-24:
Facing the same issue for python 3.9 ubuntu 20.04. I didn't find the minimum specification (besides python version) in their doc as well, it would be a nice thing to have right in the front page.

closed 2025-02-22 by YilingQiao