Genesis-Embodied-AI/genesis-world #68: Window Error: NameError: name 'trimesh' is not defined

Zachary-amateur, 2024-12-19:
[Genesis] [21:35:00] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [21:35:00] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [21:35:00] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [21:35:00] [INFO] Running on [AMD64 Family 25 Model 97 Stepping 2, AuthenticAMD] with backend gs.cpu. Device memory: 31.72 GB.
[Genesis] [21:35:00] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "D:\PersonalFiles\PycharmProjects\Learning\PycharmProjects\pythonProject\Genesis\Demo.py", line 10, in <module>
    scene = gs.Scene(show_viewer=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Administrator.DESKTOP-S3U83U6\AppData\Roaming\Python\Python312\site-packages\genesis\utils\misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "C:\Users\Administrator.DESKTOP-S3U83U6\AppData\Roaming\Python\Python312\site-packages\genesis\engine\scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "C:\Users\Administrator.DESKTOP-S3U83U6\AppData\Roaming\Python\Python312\site-packages\genesis\vis\visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Administrator.DESKTOP-S3U83U6\AppData\Roaming\Python\Python312\site-packages\genesis\vis\rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "C:\Users\Administrator.DESKTOP-S3U83U6\AppData\Roaming\Python\Python312\site-packages\genesis\vis\rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
                           ^^^^^^^
NameError: name 'trimesh' is not defined
[Genesis] [21:35:00] [INFO] 💤 Exiting Genesis and caching compiled kernels...

-----------
I installed trimesh and imported it, but the error still exists.

zhouxian, 2024-12-19:
is `ext.trimesh` imported in the file?

Zachary-amateur, 2024-12-19:
> is `ext.trimesh` imported in the file?

yeath, I imort it. here are codes:

import genesis as gs
import os
import trimesh

print(trimesh.__version__)
os.environ["KMP_DUPLICATE_LIB_OK"] = "TRUE"

gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()

JosephPai, 2024-12-19:
Same problem on Linux. I switched the package from pypi into github and have manually installed ffmpeg. Still have the problem:
```
Traceback (most recent call last):
  File "/home/ubuntu/genesisi_toy/hello_world.py", line 6, in <module>
    scene = gs.Scene(show_viewer=True)
  File "/opt/conda/envs/showo/lib/python3.10/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/opt/conda/envs/showo/lib/python3.10/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
  File "/opt/conda/envs/showo/lib/python3.10/site-packages/genesis/vis/visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
  File "/opt/conda/envs/showo/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "/opt/conda/envs/showo/lib/python3.10/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
NameError: name 'trimesh' is not defined
[Genesis] [13:56:30] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

Zachary-amateur, 2024-12-19:
> Same problem on Linux. I switched the package from pypi into github and have manually installed ffmpeg. Still have the problem:
>
> ```
> Traceback (most recent call last):
>   File [...]

you can try pip install OpenGL OpenGL-accelerate, i did it and found the error disappear, but another error (function 'glGetUniformLocation' not found) appeared, which is founded by others in windows. Considering your system is Linux, it may help

mathematicalmichael, 2024-12-19:
it's
```bash
pip install PyOpenGL PyOpenGL_accelerate
```

PS may need `sudo apt-get install mesa-utils libgl1-mesa-glx libgl1-mesa-dri`

however, I'm still having issues getting it to work due to basic inability to import opengl.

(in a headless env)

tkornuta, 2024-12-19:
For MacOS please see #32 - (partial) resolution can be found there

kaka45inablink, 2024-12-20:
Hi guys，I meet same problem on Ubuntu，any suggestion？

BlueDyee, 2024-12-20:
I think, the error occur due to the error of GUI support
It is obvious when you modify the line 3-10 of
`python3.9/site-packages/genesis/vis/rasterizer_context.py`

```
try:
    from genesis.ext import pyrender, trimesh
    from genesis.ext.pyrender.jit_render import JITRenderer
except:
    pass
```
to

```
from genesis.ext import pyrender, trimesh
from genesis.ext.pyrender.jit_render import JITRenderer
```
And I get GUI error from Linux backend,
because I use remote SSH from Windows, instead of Linux OS

mathematicalmichael, 2024-12-21:
> Hi guys，I meet same problem on Ubuntu，any suggestion？

try my suggestion above

kaka45inablink, 2024-12-23:
> Hi guys，I meet same problem on Ubuntu，any suggestion？

solved by https://github.com/Genesis-Embodied-AI/Genesis/issues/159#issuecomment-2558005819

kaka45inablink, 2024-12-23:
> > Hi guys，I meet same problem on Ubuntu，any suggestion？
> 
> try my suggestion above

Thanks,but didn't work

IaMaAVerYRANdOmPerSoN, 2025-01-05:
> it's
> 
> ```shell
> pip install PyOpenGL PyOpenGL_accelerate
> ```
> 
> PS _may_ need `sudo apt-get install mesa-utils libgl1-mesa-glx libgl1-mesa-dri mesa-utils-extra` depending on your setup https://pyrender.readthedocs.io/en/latest/examples/offscreen.html#offscreen-guide
> 
> (I am in a headless env)
> 
> I needed `export PYOPENGL_PLATFORM=egl` to run `python -c 'from genesis.ext import trimesh'` successfully.

Collecting PyOpenGL_accelerate
  Using cached PyOpenGL-accelerate-3.1.7.tar.gz (562 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: PyOpenGL_accelerate
  Building wheel for PyOpenGL_accelerate (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for PyOpenGL_accelerate (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [150 lines of output]
      /tmp/pip-build-env-og1mft0w/overlay/lib/python3.12/site-packages/setuptools/config/expand.py:125: SetuptoolsWarning: File '/tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2/readme.txt' cannot be found
        return '\n'.join(
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-312/OpenGL_accelerate
      copying OpenGL_accelerate/__init__.py -> build/lib.linux-x86_64-cpython-312/OpenGL_accelerate
      running build_ext
      /tmp/pip-build-env-og1mft0w/overlay/lib/python3.12/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2/OpenGL_accelerate/wrapper.pxd
        tree = Parsing.p_module(s, pxd, full_module_name)
      Compiling src/wrapper.pyx because it changed.
      [1/1] Cythonizing src/wrapper.pyx
      building 'OpenGL_accelerate.wrapper' extension
      creating build/temp.linux-x86_64-cpython-312/src
      x86_64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2/src -I/tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2 -I/home/mxie/genesis/include -I/usr/include/python3.12 -c src/wrapper.c -o build/temp.linux-x86_64-cpython-312/src/wrapper.o
      x86_64-linux-gnu-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-x86_64-cpython-312/src/wrapper.o -L/usr/lib/x86_64-linux-gnu -o build/lib.linux-x86_64-cpython-312/OpenGL_accelerate/wrapper.cpython-312-x86_64-linux-gnu.so
      /tmp/pip-build-env-og1mft0w/overlay/lib/python3.12/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2/OpenGL_accelerate/formathandler.pxd
        tree = Parsing.p_module(s, pxd, full_module_name)
      Compiling src/formathandler.pyx because it changed.
      [1/1] Cythonizing src/formathandler.pyx
      building 'OpenGL_accelerate.formathandler' extension
      x86_64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2/src -I/tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2 -I/home/mxie/genesis/include -I/usr/include/python3.12 -c src/formathandler.c -o build/temp.linux-x86_64-cpython-312/src/formathandler.o
      x86_64-linux-gnu-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-x86_64-cpython-312/src/formathandler.o -L/usr/lib/x86_64-linux-gnu -o build/lib.linux-x86_64-cpython-312/OpenGL_accelerate/formathandler.cpython-312-x86_64-linux-gnu.so
      /tmp/pip-build-env-og1mft0w/overlay/lib/python3.12/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2/src/arraydatatype.pyx
        tree = Parsing.p_module(s, pxd, full_module_name)
      Compiling src/arraydatatype.pyx because it changed.
      [1/1] Cythonizing src/arraydatatype.pyx
      building 'OpenGL_accelerate.arraydatatype' extension
      x86_64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2/src -I/tmp/pip-install-8nicslau/pyopengl-accelerate_65d0a8a7c5d9440d943502ef9df11dc2 -I/home/mxie/genesis/include -I/usr/include/python3.12 -c src/arraydatatype.c -o build/temp.linux-x86_64-cpython-312/src/arraydatatype.o
      x86_64-linux-gnu-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 build/temp.linux-x86_64-cpython-312/src/arraydatatype.o -L/usr/lib/x86_64-linux-gnu -o build/lib.linux-x86_64-cpython-312/OpenGL_accelerate/arraydatatype.cpython-312-x86_64-linux-gnu.so
      /tmp/pip-build-env-og1mft0w/overlay/lib/python3.12/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: / [... cut ...]
          new_ext = cythonize(
                    ^^^^^^^^^^
        File "/tmp/pip-build-env-og1mft0w/overlay/lib/python3.12/site-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
          cythonize_one(*args)
        File "/tmp/pip-build-env-og1mft0w/overlay/lib/python3.12/site-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
          raise CompileError(None, pyx_file)
      Cython.Compiler.Errors.CompileError: src/numpy_formathandler.pyx
      [end of output]

closed 2025-02-22 by YilingQiao