Genesis-Embodied-AI/genesis-world #69: Nothing happens/ no GUI launched after running demo on M-series Mac

ksDreamer, 2024-12-19:
While running the [Hello, Genesis](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/hello_genesis.html) demo, it recommended
```
[Genesis] [21:38:30] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`.
```
so I tried`python examples/render_on_macos.py` in terminal, it went without errors. But nothing happened. 
Following are some infos in terminal 
```
[Genesis] [21:41:54] [INFO] Running at 4360.85 FPS.
3435.1384111384114 FPS
[Genesis] [21:41:54] [INFO] Running at 4342.33 FPS.
3775.251125112511 FPS
[Genesis] [21:41:54] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```
Also another issues is failure while installing the `genesis-world` python package, as mentioned in https://github.com/Genesis-Embodied-AI/Genesis/issues/52. To replicate  and locate this issue, my previous case is python=3.9.
The error messages are shown below.
```INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
Collecting genesis-world
  Using cached genesis_world-0.1.1-py3-none-any.whl.metadata (11 kB)
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on pymeshlab
    genesis-world 0.1.1 depends on pymeshlab

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
``` 
Now I use python=3.11 and it goes well. But I think it's a problem to be solved.
My device is M1 8GB Mac, with Sonoma MacOS version. Is it due to out of memory?

Jinbo-He, 2024-12-19:
I meet the same question on m2 air, it's running and no gui shows.
[image]

TyTheLoser, 2024-12-19:
Have you solved this problem? I am also an m1 macbook, but I can only run to 50 frames
```[Genesis] [00:45:55] [INFO] Running at 58.49 FPS.
[Genesis] [00:45:55] [INFO] Running at 58.55 FPS.
[Genesis] [00:45:55] [INFO] Running at 58.67 FPS.
[Genesis] [00:45:55] [INFO] Running at 58.62 FPS.
[Genesis] [00:45:55] [INFO] Running at 58.69 FPS.
[Genesis] [00:45:55] [INFO] 💤 Exiting Genesis and caching compiled kernels...

au79, 2024-12-19:
I am able to run `python examples/render_on_macos.py` successfully, but no GUI seems to appear.  I see a new item in my Dock for a moment, and then it disappears.

Is it possible that it opens and closes too quickly?  Could it pause on open, wait for a keypress, and then continue?

RobRoyce, 2024-12-19:
> I am able to run `python examples/render_on_macos.py` successfully, but no GUI seems to appear. I see a new item in my Dock for a moment, and then it disappears.
> 
> Is it possible that it opens and closes too quickly? Could it pause on open, wait for a keypress, and then continue?
> 
> M2 Max, 64GB memory, Sonoma 14.5

Are you running it with the `--vis` argument? If not, it won't display the GUI.

au79, 2024-12-19:
Thanks, @RobRoyce, that solved my problem.  Sorry I missed it.

analogjedi, 2024-12-20:
@au79 , I am not able to run this example on my M3 Max within my Python 3.11 venv.
I filed this issue: #132 

Could you please share your environment details and pip list?

PieterBecking, 2024-12-21:
Running into the same issue. Tried --v and --vis. Also tried different python versions but can't seem to get the visuals up. Only the icon in the dock while it is generating.

jwjohns, 2024-12-21:
Same for me on m3 max.

Jinbo-He, 2024-12-21:
> Running into the same issue. Tried --v and --vis. Also tried different python versions but can't seem to get the visuals up. Only the icon in the dock while it is generating.

set default=true, OpenGL==3.1.5

dafiiit, 2024-12-21:
It works for me. When using the render_on_macos script, setting the --vis flag and using python3.11 
I ran into some isseus with higher Python versions.

closed 2025-02-22 by YilingQiao