Genesis-Embodied-AI/genesis-world #70: Documents/examples about setting entity physcial properties(for domain randomization in RL training)

annan-tang, 2024-12-19:
Thank you for creating this excellent simulator! I am currently migrating my reinforcement learning code from IsaacGym to Genesis. While working on this transition, I've noticed that I need some clarification regarding the APIs for setting physical properties of entities.
Specifically, I'm looking for information on how to set:

Link properties:

Mass
Inertial properties
Center of mass

Joint properties:

Friction
Motor rotor parameters (armature/inertia)

RigidShape material properties:

Friction coefficients
Restitution

I've found some relevant APIs in the documentation:

[set_mass(link)](https://genesis-world.readthedocs.io/en/latest/api_reference/entity/rigidentity/rigidlink.html#genesis.engine.entities.rigid_entity.rigid_link.RigidLink.set_mass)
[set_friction(link)](https://genesis-world.readthedocs.io/en/latest/api_reference/entity/rigidentity/rigidlink.html#genesis.engine.entities.rigid_entity.rigid_link.RigidLink.set_friction)
[set_friction(geometry)](https://genesis-world.readthedocs.io/en/latest/api_reference/entity/rigidentity/rigidgeom.html#genesis.engine.entities.rigid_entity.rigid_geom.RigidGeom.set_friction)
[set_friction_ratio(entity)](https://genesis-world.readthedocs.io/en/latest/api_reference/entity/rigidentity/rigidentity.html#genesis.engine.entities.rigid_entity.rigid_entity.RigidEntity.set_friction_ratio)

However, I'm wondering if there are additional APIs for setting other physical properties that I might have missed. Would you be able to:

Point me to any documentation covering these features?
Share examples demonstrating how to set these various properties?
Provide a comprehensive list of all available property-setting APIs?

Thank you for your time and assistance.

zswang666, 2024-12-19:
Thanks for bringing this up. Theoretically, we should be able to set most of the properties, but we need to write an interface to extract those tensor via the kernel, e.g., [from dofs](https://github.com/Genesis-Embodied-AI/Genesis/blob/2985aa9c63b4162cea10a52c98161e1270477746/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L261), [from link](https://github.com/Genesis-Embodied-AI/Genesis/blob/2985aa9c63b4162cea10a52c98161e1270477746/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L382), [from geom](https://github.com/Genesis-Embodied-AI/Genesis/blob/2985aa9c63b4162cea10a52c98161e1270477746/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L678), etc. We only implemented those we were using. Would you mind make a list of physical properties that you would like to set so that we can incorporate the interface later?

Another note is that modifying some physical properties may make simulation unstable.

ziyanx02, 2024-12-19:
In RigidSolver, you can find
set_links_mass_shift
set_links_COM_shift
for domain randomization of mass and com.

Hellod035, 2024-12-19:
It seems that there is no interface for setting link speed and force

annan-tang, 2024-12-20:
> Thanks for bringing this up. Theoretically, we should be able to set most of the properties, but we need to write an interface to extract those tensor via the kernel, e.g., [from [...]

Thank you very much for the quick respose. From my experience, the most common used physical properties in DR are:
- from dofs: armature; friction.
- from link: mass, com, inertia
- from geom: friction, restitution
- others: add external force on links/base.
I've found that most of these properties can already be set using the existing APIs, which is great. For the few remaining properties that aren't yet implemented, thank you very much.

anahrendra, 2024-12-20:
> Thanks for bringing this up. Theoretically, we should be able to set most of the properties, but we need to write an interface to extract those tensor via the kernel, e.g., [from [...]

Bumping in the discussion for this topic.

AFAIK, the common practice for the entitity-related randomized parameters such as frictions, restitution, etc. are randomized per entity at the time the environment is created. Thus, there is no need to reset the values at every episode reset.

Then would it still affect the simulation stability/accuracy?

shirgur, 2024-12-20:
> Bumping in the discussion for this topic.
>
> AFAIK, the common practice for the entitity-related randomized parameters such as frictions, restitution, etc. are randomized per entity at the time the [...]

This doesn’t seem like a common practice but rather a limitation, so I wouldn’t recommend keeping it this way.

ziyanx02, 2024-12-20:
> It seems that there is no interface for setting link speed and force

Hi! I've added a function to apply external force and torque on links. Please check examples/rigid/apply_external_force_torque.py fir the detail

zswang666, 2024-12-21:
> Thank you very much for the quick respose. From my experience, the most common used physical properties in DR are:
>
> * from dofs: armature; friction.
> * from link: mass, com, inertia
> * from geom: [...]

Currently some of the attributes are not stored in batch and thus we cannot set different params across environments, including [`dofs_info`](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L303), [`links_info`](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L445), [`geoms_info`](https://github.com/Genesis-Embodied-AI/Genesis/blob/6c3c9389e1713e4bc4301089fa22aaf218b3c021/genesis/engine/solvers/rigid/rigid_solver_decomp.py#L714). @YilingQiao do we want to set back to include the batch dimension for these info stuffs? Or perhaps include an argument like `include_batch_dim_in_info` in rigid solver option to toggle this? Or doing the shift thingy in the `xxx_state` field?

closed 2025-02-22 by YilingQiao