Genesis-Embodied-AI/genesis-world #71: Unrecognized attribute: res when running examples/smoke on M3 Max

apandy02, 2024-12-19:
python smoke.py 
[Taichi] version 1.7.2, llvm 15.0.7, commit 0131dce9, osx, python 3.12.6
objc[17667]: Class GLFWHelper is implemented in both /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x303af8570) and /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x3605497c0). One of the two will be used. Which one is undefined.
objc[17667]: Class GLFWApplicationDelegate is implemented in both /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x303af85c0) and /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x360549810). One of the two will be used. Which one is undefined.
objc[17667]: Class GLFWWindowDelegate is implemented in both /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x303af85e8) and /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x360549838). One of the two will be used. Which one is undefined.
objc[17667]: Class GLFWContentView is implemented in both /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x303af8638) and /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x360549888). One of the two will be used. Which one is undefined.
objc[17667]: Class GLFWWindow is implemented in both /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x303af86b0) and /Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x360549900). One of the two will be used. Which one is undefined.
Failed to import pyrender. Rendering will not work.
[Genesis] [09:19:38] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [09:19:38] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [09:19:38] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [09:19:38] [INFO] Running on [Apple M3 Max] with backend gs.metal. Device memory: 64.00 GB.
[Genesis] [09:19:38] [DEBUG] [Taichi] Starting on arch=metal
[Genesis] [09:19:38] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/Users/apandya/genesis_experiments/smoke.py", line 140, in <module>
    main()
  File "/Users/apandya/genesis_experiments/smoke.py", line 87, in main
    sf_options=gs.options.SFOptions(
               ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/apandya/genesis_experiments/.venv/lib/python3.12/site-packages/genesis/options/options.py", line 27, in __init__
    gs.raise_exception(f"Unrecognized attribute: {key}")

[Genesis] [09:19:38] [ERROR] Unrecognized attribute: res
[Genesis] [09:19:38] [INFO] 💤 Exiting Genesis and caching compiled kernels...

closed 2024-12-19 by apandy02

godzukisdad, 2024-12-20:
How did you resolve this error? Having the same problem.

apandy02, 2024-12-20:
@godzukisdad brew instal ffmpeg (this was missing)

Then follow the instructions here: https://github.com/Genesis-Embodied-AI/Genesis/issues/11#issuecomment-2557519124

Yzhang367, 2024-12-21:
Same problem on Ubuntu but cannot be solved by installing `ffmpeg`
```Terminal
(genesis) (base) yzhang@yzhang-ASUS-TUF-Gaming-F16-FX607JV-FX607JV:~/桌面/Genesis$ python -u "/home/yzhang/桌面/Genesis/examples/smoke.py"
[Taichi] version 1.7.2, llvm 15.0.4, commit 0131dce9, linux, python 3.12.8
[Genesis] [11:50:25] [INFO] ╭───────────────────────────────────────────────────────────────────────╮
[Genesis] [11:50:25] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [11:50:25] [INFO] ╰───────────────────────────────────────────────────────────────────────╯
[Genesis] [11:50:26] [INFO] Running on [NVIDIA GeForce RTX 4060 Laptop GPU] with backend gs.cuda. Device memory: 7.65 GB.
[W 12/21/24 11:50:26.071 39710] [cuda_driver.cpp:load_lib@36] libcuda.so lib not found.
[W 12/21/24 11:50:26.071 39710] [misc.py:adaptive_arch_select@758] Arch=[<Arch.cuda: 3>] is not supported, falling back to CPU
[Genesis] [11:50:26] [DEBUG] [Taichi] Starting on arch=x64
[Genesis] [11:50:26] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/home/yzhang/桌面/Genesis/examples/smoke.py", line 140, in <module>
    main()
  File "/home/yzhang/桌面/Genesis/examples/smoke.py", line 87, in main
    sf_options=gs.options.SFOptions(
               ^^^^^^^^^^^^^^^^^^^^^
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/options/options.py", line 27, in __init__
    gs.raise_exception(f"Unrecognized attribute: {key}")

[Genesis] [11:50:26] [ERROR] Unrecognized attribute: res
[Genesis] [11:50:26] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```