Genesis-Embodied-AI/genesis-world #75: Can we use this for molecular dynamics (MD)?

subercui, 2024-12-19:
Thank you so much for the great library! Just wonder is there a way we can use this for molecular dynamics simulation? That can be so powerful.

murphyericjames, 2024-12-19:
Seems like an extension in a nonsensical direction. MD is all about proper force field usage and sampling. So it probably makes sense to extend an MD code to SPH, but does it make strategic sense to do the reverse? How do you couple MD to any of the other packages here? There are already many open source packages that are easy to run, and do from python. You can even ask chatgpt to generate the deck for you. Generate post processing visualization etc.

YilingQiao, 2025-02-22:
Yeah, agreed. Their scales and applications are quite different.

closed 2025-02-22 by YilingQiao