Genesis-Embodied-AI/genesis-world #86: RuntimeError: [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_UNKNOWN: unknown error while calling device_get_default_mem_pool (cuDeviceGetDefaultMemPool)

patweatharva, 2024-12-19:
I am trying to run the given locomotion example with CUDA GPU but I get "RuntimeError: [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_UNKNOWN: unknown error while calling device_get_default_mem_pool (cuDeviceGetDefaultMemPool)"

Specific error- 
'''
[E 12/19/24 18:19:19.038 35800] [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_UNKNOWN: unknown error while calling device_get_default_mem_pool (cuDeviceGetDefaultMemPool)

Traceback (most recent call last):
  File "/home/blahblah/RL/Genesis_projects/test_project/go2_train.py", line 170, in <module>
    main()
  File "/home/blahblah/RL/Genesis_projects/test_project/go2_train.py", line 145, in main
    gs.init(logging_level="warning")
  File "/home/blahblah/RL/Genesis_projects/genesis_venv/lib/python3.12/site-packages/genesis/__init__.py", line 97, in init
    ti.init(arch=TI_ARCH[platform][backend], debug=debug, force_scalarize_matrix=True)
  File "/home/blahblah/RL/Genesis_projects/genesis_venv/lib/python3.12/site-packages/taichi/lang/misc.py", line 458, in init
    impl.get_runtime().create_program()
  File "/home/blahblah/RL/Genesis_projects/genesis_venv/lib/python3.12/site-packages/taichi/lang/impl.py", line 388, in create_program
    self.prog = _ti_core.Program()
                ^^^^^^^^^^^^^^^^^^
RuntimeError: [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_UNKNOWN: unknown error while calling device_get_default_mem_pool (cuDeviceGetDefaultMemPool)
'''

nvidia-smi output-
'''
Thu Dec 19 18:30:34 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 1660 Ti     Off | 00000000:01:00.0 Off |                  N/A |
| N/A   48C    P0              24W /  80W |    425MiB /  6144MiB |     34%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1328      G   /usr/lib/xorg/Xorg                          163MiB |
|    0   N/A  N/A      1805      G   /usr/bin/gnome-shell                         56MiB |
|    0   N/A  N/A      5034      G   /usr/lib/firefox/firefox                    152MiB |
|    0   N/A  N/A     33073      G   ...erProcess --variations-seed-version       46MiB |

'''

nvcc --version
'''
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Sun_Jul_28_19:07:16_PDT_2019
Cuda compilation tools, release 10.1, V10.1.243
'''

ChipsICU, 2024-12-19:
Your CUDA Toolkit is 10.1, it's too old to use, authors recommend CUDA Toolkit 11.7+, check this:
https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html

patweatharva, 2024-12-19:
Yesss, Got it! Thanks

closed 2024-12-19 by patweatharva

TTPlanetPig, 2024-12-20:
I have the same issue with my cuda 12.1
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Mon_Apr__3_17:16:06_PDT_2023
Cuda compilation tools, release 12.1, V12.1.105
Build cuda_12.1.r12.1/compiler.32688072_0
I got the similar error like:
[E 12/20/24 08:45:40.689 2984] [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected while calling init (cuInit)