Genesis-Embodied-AI/genesis-world #90: "NameError: name 'trimesh' is not defined" on M3 Mac

zixinz990, 2024-12-19:
Tried several examples, but kept getting this error. For examples, when I ran [Beyond Rigid Bodies](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/beyond_rigid_bodies.html), I got:

```
objc[3175]: Class GLFWHelper is implemented in both /opt/homebrew/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x147e08570) and /opt/homebrew/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x17ec6d7c0). One of the two will be used. Which one is undefined.
objc[3175]: Class GLFWApplicationDelegate is implemented in both /opt/homebrew/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x147e085c0) and /opt/homebrew/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x17ec6d810). One of the two will be used. Which one is undefined.
objc[3175]: Class GLFWWindowDelegate is implemented in both /opt/homebrew/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x147e085e8) and /opt/homebrew/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x17ec6d838). One of the two will be used. Which one is undefined.
objc[3175]: Class GLFWContentView is implemented in both /opt/homebrew/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x147e08638) and /opt/homebrew/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x17ec6d888). One of the two will be used. Which one is undefined.
objc[3175]: Class GLFWWindow is implemented in both /opt/homebrew/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x147e086b0) and /opt/homebrew/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x17ec6d900). One of the two will be used. Which one is undefined.
Failed to import pyrender. Rendering will not work.
[Genesis] [11:13:27] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [11:13:27] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [11:13:27] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [11:13:28] [INFO] Running on [Apple M3] with backend gs.metal. Device memory: 16.00 GB.
[Genesis] [11:13:28] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/Users/zixinzhang/Documents/genesis_test.py", line 8, in <module>
    scene = gs.Scene(
            ^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/opt/homebrew/lib/python3.12/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/genesis/vis/visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "/opt/homebrew/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
                           ^^^^^^^
NameError: name 'trimesh' is not defined
```

I am sure trimesh has been installed in my Python environment.

pluti, 2024-12-19:
I have the same issue. Seems to be "Pyrender" that can't handle OpenGL on Arm mac. Not sure how to solve it either ...
 https://github.com/mmatl/pyrender/tree/master?tab=readme-ov-file

miguelalonsojr, 2024-12-19:
Same issue on my end. :(

miguelalonsojr, 2024-12-19:
on my M3, it seems to be related to not having ffmpeg installed. try

```bash
brew update
brew install ffmpeg
```

The hello, world example runs, but I can't see the render window. But at least there's no exception! :)

miguelalonsojr, 2024-12-19:
> M4 Mac, same issue...

Try installing ffmpeg. Worked for me.

miguelalonsojr, 2024-12-19:
Figured out why the viewer didn't show up. Got this message:

```bash
[Genesis] [15:06:37] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`
```

For macs, the viewer needs to be run on a separate thread.

iandanforth, 2024-12-19:
> Figured out why the viewer didn't show up. Got this message:
> 
> ```shell
> [Genesis] [15:06:37] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`
> ```
> 
> For macs, the viewer needs to be run on a separate thread.

Running this example script on Mac M4 results in the same error so this is not a general solution

miguelalonsojr, 2024-12-19:
Hmmm. It's working for me with `examples/render_on_macos.py`. Did you install ffmpeg?

RobRoyce, 2024-12-19:
Check #26 

Duplicate of #32

miguelalonsojr, 2024-12-19:
> Check #26
> 
> Duplicate of #32

Lol, wish I would have seen this earlier. Took me a bit to figure it out. Thanks!

miguelalonsojr, 2024-12-19:
@iandanforth I'm on an M3 mac though... and I'm using the gs.metal backend for the example. Not sure if that matters. Also, see https://github.com/Genesis-Embodied-AI/Genesis/pull/27. For more details.

YutoYasunaga, 2024-12-20:
You can try this. It works for me, I'm using M1 Mac.

https://github.com/Genesis-Embodied-AI/Genesis/issues/36#issuecomment-2556229369

futuyao, 2024-12-20:
Same issue. Fixed by installing pyrender.

closed 2024-12-29 by zixinz990