Genesis-Embodied-AI/genesis-world #93: hello world example doesnt produce video

jtoy, 2024-12-19:
import genesis as gs

gs.init(backend=gs.gpu)

scene = gs.Scene(
    show_viewer = True,
    viewer_options = gs.options.ViewerOptions(
        res           = (1280, 960),
        camera_pos    = (3.5, 0.0, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 40,
        max_FPS       = 60,
    ),
    vis_options = gs.options.VisOptions(
        show_world_frame = True,
        world_frame_size = 1.0,
        show_link_frame  = False,
        show_cameras     = False,
        plane_reflection = True,
        ambient_light    = (0.1, 0.1, 0.1),
    ),
    renderer=gs.renderers.Rasterizer(),
)

plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

cam = scene.add_camera(
    res    = (640, 480),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = False,
)

scene.build()

# render rgb, depth, segmentation, and normal
# rgb, depth, segmentation, normal = cam.render(rgb=True, depth=True, segmentation=True, normal=True)

cam.start_recording()
import numpy as np

for i in range(120):
    scene.step()
    cam.set_pose(
        pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
        lookat = (0, 0, 0.5),
    )
    cam.render()
cam.stop_recording(save_to_filename='video.mp4', fps=30)

it just seems to sit there:

[Genesis] [11:36:48] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [11:36:48] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [11:36:48] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [11:36:49] [INFO] Running on [Apple M1 Max] with backend gs.metal. Device memory: 32.00 GB.
[Genesis] [11:36:49] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [11:36:49] [INFO] Scene <69300f1> created.
[Genesis] [11:36:49] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <7cfebb9>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [11:36:49] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <3a76479>, morph: <gs.morphs.MJCF(file='/Users/jtoy/.asdf/installs/python/3.10.4/lib/python3.10/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [11:36:50] [INFO] Building scene <69300f1>...
[Genesis] [11:36:54] [INFO] Compiling simulation kernels...
[Genesis] [11:36:59] [INFO] Building visualizer...
[Genesis] [11:37:01] [WARNING] Non-linux system detected. In order to use the interactive viewer, you need to manually run simulation in a separate thread and then start viewer. See `examples/render_on_macos.py`.
[Genesis] [11:37:01] [INFO] Viewer created. Resolution: 1280×960, max_FPS: 60.
[Genesis] [11:37:01] [INFO] Running at 2824.03 FPS.

apandy02, 2024-12-19:
Same problem while running `examples/render_on_macos.py` on M3 Max

RobRoyce, 2024-12-19:
@apandy02 are you running it with the `--vis` flag?

@jtoy as a baseline, does the [hello_genesis.py](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/tutorials/hello_genesis.py) tutorial work for you? 

I had a similar issue trying to add a new camera to the scene, so it's hard to tell if that's your issue or something more fundamental.

jtoy, 2024-12-19:
I got the elastic_dragon example working,I modified it to add a camera and get video, so I think the hello world example is not fully working properly.

 import argparse

import numpy as np

import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    parser.add_argument("-c", "--cpu", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(backend=gs.cpu if args.cpu else gs.gpu, logging_level="debug")

    ########################## create a scene ##########################

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            substeps=10,
            gravity=(0, 0, -9.8),
        ),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(2, 2, 1.5),
            camera_lookat=(0, 0, 0.5),
            camera_up=(0, 0, 1),
        ),
        show_viewer=args.vis,
    )
    cam = scene.add_camera(
        res    = (1280, 720),
        pos    = (3.5, 0.0, 2.5),
        lookat = (0, 0, 0.5),
        fov    = 30,
        GUI    = False,
    )

    ########################## materials ##########################
    mat_elastic = gs.materials.PBD.Elastic()

    ########################## entities ##########################

    bunny = scene.add_entity(
        material=mat_elastic,
        morph=gs.morphs.Mesh(
            file="meshes/dragon/dragon.obj",
            scale=0.003,
            pos=(0, 0, 0.8),
        ),
        surface=gs.surfaces.Default(
            # vis_mode='recon',
        ),
    )
    ########################## build ##########################
    scene.build()

    horizon = 1000
    # forward pass
    cam.start_recording()
    for i in range(horizon):
        scene.step()
        cam.set_pose(
            pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
            lookat = (0, 0, 0.5),
        )
        cam.render()

    cam.stop_recording(save_to_filename='video.mp4', fps=30)

if __name__ == "__main__":
    main()

analogjedi, 2024-12-20:
@vinny-888 , I am also having OpenGL error issues and filed a new issue for this: #165 
I am not exactly sure if this is the issue or not related but the end result is the same error as yours.

apandy02, 2024-12-20:
@RobRoyce My issue was resolved by switching from uv to conda for package management. Would be interested in understanding why, but for now I'm all set!

duburcqa, 2025-02-27:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/777 (and probably https://github.com/Genesis-Embodied-AI/Genesis/pull/782)