Genesis-Embodied-AI/genesis-world #99: AttributeError: function 'glGetUniformLocation' not found

IvanGaideek, 2024-12-19:
When calling scene.build() such an error (using Windows 11):

"""[Genesis] [22:23:38] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [22:23:38] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [22:23:38] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [22:23:39] [INFO] Running on [NVIDIA GeForce RTX 3060 Laptop GPU] with backend gs.cuda. Device memory: 6.00 GB.
[Genesis] [22:23:39] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [22:23:39] [INFO] Scene <57eb10a> created.
[Genesis] [22:23:39] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <a548b23>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [22:23:39] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <c2fbe27>, morph: <gs.morphs.MJCF(file='C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [22:23:40] [WARNING] Overriding base link's pos with user provided value in morph.
[Genesis] [22:23:40] [WARNING] Overriding base link's quat with user provided value in morph.
[Genesis] [22:23:40] [INFO] Building scene <57eb10a>...
[Genesis] [22:24:10] [INFO] Compiling simulation kernels...
[Genesis] [22:25:49] [INFO] Building visualizer...
Traceback (most recent call last):
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\main.py", line 36, in <module>
    scene.build()
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 58, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
  File "C:\Users\gajna\PycharmProjects\DataScience\physics_gen\venv\lib\site-packages\OpenGL\platform\ctypesloader.py", line 99, in buildFunction
    return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found
[Genesis] [22:25:50] [INFO] 💤 Exiting Genesis and caching compiled kernels..."""

closed 2024-12-19 by IvanGaideek

cloudhu, 2024-12-20:
how did you fix it？

rlagarto, 2024-12-20:
gettin same error, I tried to updating the drivers but seems that is not the issue