Genesis-Embodied-AI/genesis-world #112: MESA: error: ZINK: failed to choose pdev glx: failed to create drisw screen Exception in thread Thread-2 (_init_and_start_app): Traceback (most recent call last):

RRGGZZ, 2024-12-20:
device :[WSL2]
PyOpenGL==3.1.7

the error is 
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):

RRGGZZ, 2024-12-20:
I running the example script :

import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()

RRGGZZ, 2024-12-20:
and my WSL2 is Ubuntu 24.04.01

cloudhu, 2024-12-20:
same error

cloudhu, 2024-12-20:
fixed in Ubuntu
`sudo add-apt-repository ppa:kisak/kisak-mesa  
sudo apt update  
sudo apt upgrade`

closed 2024-12-21 by RRGGZZ