Genesis-Embodied-AI/genesis-world #114: CUDA Error CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected while calling init (cuInit)

TTPlanetPig, 2024-12-20:
I have the same issue with my cuda 12.1 run under wsl2 with both python 310-312 tested

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Mon_Apr__3_17:16:06_PDT_2023
Cuda compilation tools, release 12.1, V12.1.105
Build cuda_12.1.r12.1/compiler.32688072_0

I got the error like:
[E 12/20/24 08:45:40.689 2984] [cuda_driver.h:operator()@92] CUDA Error CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected while calling init (cuInit)

hougd7, 2024-12-23:
the same problem with wsl2，python310,  cuda11.6

LongHZ140516, 2024-12-24:
Hi @TTPlanetPig .
I encountered the same problem a few days ago, and I have solved it. Maybe you can refer to my issue #187  and try to solve it. Hope it can help you.

hongbo-miao, 2025-01-17:
@LongHZ140516 `export LD_LIBRARY_PATH=/usr/lib/wsl/lib` is a workaround solution. The root issue is incorrect driver got installed. I posted the detail at https://stackoverflow.com/a/79357542/2000548

closed 2025-02-23 by YilingQiao