Genesis-Embodied-AI/genesis-world #124: water simulation on MacOS

neilwth, 2024-12-20:
After pulling the latest src with [this PR](https://github.com/Genesis-Embodied-AI/Genesis/pull/8), the float32/64 issue resolved, but I still got the following issues with backend gs.metal:

[Genesis] [11:36:48] [WARNING] `pbs` sampler failed. Falling back to `random` sampler.

and

[Genesis] [11:36:58] [INFO] Compiling simulation kernels...
RHI Error: cannot create compute pipeline state: Compiler encountered an internal error (code=3)
RHI Error: cannot create compute pipeline state: Compiler encountered an internal error (code=3)
Assertion failed: (p != nullptr), function bind_pipeline, file metal_device.mm, line 382.

Still, the script works fine but slowly with cpu, and simulations without liquid are OK for both cpu and metal.

jrryzh, 2024-12-20:
Met same problem.

arielhgl, 2024-12-22:
same issue here

ai-bits, 2025-01-03:
> Still, the script works fine but slowly with cpu, and simulations without liquid are OK for both cpu and metal.

Are you saying you SAW sth / the Viewer on the Mac? Guess no.

hckang-dev, 2025-01-15:
I commented out this line and it worked.

Genesis/genesis/engine/solvers/rigid/constraint_solver_decomp.py
line 840: self.efc_force[i_c, i_b] = -self.efc_D[i_c, i_b] * self.Jaref[i_c, i_b] * self.active[i_c, i_b]

RoryMB, 2025-01-16:
+1, same issue here.

duburcqa, 2025-03-22:
This issue has been fixed on main branch.

closed 2025-03-22 by duburcqa