Genesis-Embodied-AI/genesis-world #125: set the center gravity for the robot

DaMiTian, 2024-12-20:
hello,finally I successfully run the genesis on my device,and take my dual arm robot in it by URDF format.
however, while playing demo like "Control your robot",my robot fell down.
if i can set my robot gravity for robot unit?
[image]

ziyanx02, 2024-12-21:
I recommend visualizing the robot's center of mass to ensure its projection falls within the range covered by wheels.

DaMiTian, 2024-12-21:
thanks for you reply ! I have overcome this question ,and now i am putting effort on setting my camera on my robot,if there any sample showing how to binding the camera to the end of joint?

DaMiTian, 2024-12-21:
And some of the functions like language generate scene showed in the demo, when will it release? And now the project is running at a relatively slow speed with poor physical simulation. Will it be improved after the release of the beta version in the future？
Looking forward to your reply!

zhouxian, 2024-12-21:
Could you elaborate on slow speed and poor simulation? -- if there's any issue we will fix asap, but we treat it seriously

DaMiTian, 2024-12-21:
Thank for your reply.
My device is 2000 ada Laptop,and when I running my robot without any camera, it will have 60+FPS,while adding a camera only with RGB,the FPS of it will decrease less than 15 FPS, is it working properly?I will show my code this morning.

ziyanx02, 2024-12-21:
4x slower seems reasonable to me.

I will check if there is an API for mounting cameras, but in the meantime, setting the camera's position could serve as a temporary solution to your problem.

zhouxian, 2024-12-21:
No it shouldn't be so slow. Could you provide your script?
Also - if you are not looking for gpu-aceelerated parallel env, you should us `gs.cpu` backend. it will be much faster

DaMiTian, 2024-12-22:
sorry for so late,here is my code runing on my device:
`import genesis as gs
import numpy as np
import cv2
gs.init(backend=gs.gpu)

# -----------场景初始化-----------
scene = gs.Scene(show_viewer=True)
# -----------背景-----------
plane = scene.add_entity(gs.morphs.Plane())
# -----------机械臂-----------
cam = scene.add_camera(
    res    = (1280, 960),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = True
)

rm_dual = scene.add_entity(
    gs.morphs.URDF(file='/home/nvidia/Genesis/rm_models/dual_arm_robot/urdf/dual_arm_robot.urdf'),
)
# -----------生成初始化-----------
scene.build()
# 相机操作
# rgb, depth, segmentation, normal = cam.render(depth=True, segmentation=True, normal=True)

# 绑定对应左手控制关节
left_jnt_names = [f"Joint_left{i}" for i in range(1,7)]+[f"Joint_l_finger{i}" for i in range(1,3)]

l_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in left_jnt_names]

# 绑定对应右手控制关节
right_jnt_names = [f"Joint_right{i}" for i in range(1,7)]+[f"Joint_r_finger{i}" for i in range(1,3)]

r_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in right_jnt_names]

# 绑定车轮
left_wheel_names = [f'joint_wheel_lf{i}' for i in range(1,3)]+[f'joint_wheel_lb{i}' for i in range(1,3)]
right_wheel_names = [f'joint_wheel_rf{i}' for i in range(1,3)]+[f'joint_wheel_rb{i}' for i in range(1,3)]

l_wheel_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in left_wheel_names]
r_wheel_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in right_wheel_names]

wheel_go_names =['joint_wheel_right','joint_wheel_left']
wheel_go_index = [rm_dual.get_joint(name).dof_idx_local for name in wheel_go_names]

# 绑定相机
cam_names =['joint_camera_ro']
cam_index = [rm_dual.get_joint(name).dof_idx_local for name in cam_names]
# 设置一些运动参数
# -----------左臂参数-----------
# 设置位置增益参数
rm_dual.set_dofs_kp(
    kp             = np.array([4500, 4500, 3500, 3500, 2000, 2000, 100, 100]),
    dofs_idx_local = l_dofs_idx,
)
# 设置速度增益参数：
rm_dual.set_dofs_kv(
    kv             = np.array([1000, 1000, 1000, 1000, 1000, 1000, 10, 10]),
    dofs_idx_local = l_dofs_idx,
)
# 设置力控安全区间
rm_dual.set_dofs_force_range(
    lower          = np.array([-87, -87, -87, -87, -12, -12, -100, -100]),
    upper          = np.array([ 87,  87,  87,  87,  12,  12,  100,  100]),
    dofs_idx_local = l_dofs_idx,
)
# -----------右臂参数-----------
# 设置位置增益参数
[... 124 lines cut ...]
            r_dofs_idx
        )
        rm_dual.control_dofs_velocity(
            np.array([5,5]), 
            wheel_go_index
        )
    if i==900:
        rm_dual.control_dofs_position(
            np.array([0,0,0,0,0,0,0,0]), 
            l_dofs_idx
        )
        rm_dual.control_dofs_position(
            np.array([0,0,0,0,0,0,0.1,1]), 
            r_dofs_idx
        )
        rm_dual.control_dofs_velocity(
            np.array([5,-5]), 
            wheel_go_index
        )
    cam_pos = rm_dual.get_dofs_position(cam_index[:1])
    print(rm_dual.joints[cam_index[0]])
    
    print(rm_dual.joints[cam_index[0]].get_pos())
    scene.step()
    # cam.render()`

DaMiTian, 2024-12-22:
here is my code:
```
import genesis as gs
import numpy as np
import cv2
gs.init(backend=gs.gpu)

# -----------场景初始化-----------
scene = gs.Scene(show_viewer=True)
# -----------背景-----------
plane = scene.add_entity(gs.morphs.Plane())
# -----------机械臂-----------
cam = scene.add_camera(
    res    = (1280, 960),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = True
)

rm_dual = scene.add_entity(
    gs.morphs.URDF(file='/home/nvidia/Genesis/rm_models/dual_arm_robot/urdf/dual_arm_robot.urdf'),
)
# -----------生成初始化-----------
scene.build()
# 相机操作
# rgb, depth, segmentation, normal = cam.render(depth=True, segmentation=True, normal=True)

# 绑定对应左手控制关节
left_jnt_names = [f"Joint_left{i}" for i in range(1,7)]+[f"Joint_l_finger{i}" for i in range(1,3)]

l_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in left_jnt_names]

# 绑定对应右手控制关节
right_jnt_names = [f"Joint_right{i}" for i in range(1,7)]+[f"Joint_r_finger{i}" for i in range(1,3)]

r_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in right_jnt_names]

# 绑定车轮
left_wheel_names = [f'joint_wheel_lf{i}' for i in range(1,3)]+[f'joint_wheel_lb{i}' for i in range(1,3)]
right_wheel_names = [f'joint_wheel_rf{i}' for i in range(1,3)]+[f'joint_wheel_rb{i}' for i in range(1,3)]

l_wheel_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in left_wheel_names]
r_wheel_dofs_idx = [rm_dual.get_joint(name).dof_idx_local for name in right_wheel_names]

wheel_go_names =['joint_wheel_right','joint_wheel_left']
wheel_go_index = [rm_dual.get_joint(name).dof_idx_local for name in wheel_go_names]

# 绑定相机
cam_names =['joint_camera_ro']
cam_index = [rm_dual.get_joint(name).dof_idx_local for name in cam_names]
# 设置一些运动参数
# -----------左臂参数-----------
# 设置位置增益参数
rm_dual.set_dofs_kp(
    kp             = np.array([4500, 4500, 3500, 3500, 2000, 2000, 100, 100]),
    dofs_idx_local = l_dofs_idx,
)
# 设置速度增益参数：
rm_dual.set_dofs_kv(
    kv             = np.array([1000, 1000, 1000, 1000, 1000, 1000, 10, 10]),
    dofs_idx_local = l_dofs_idx,
)
# 设置力控安全区间
rm_dual.set_dofs_force_range(
    lower          = np.array([-87, -87, -87, -87, -12, -12, -100, -100]),
    upper          = np.array([ 87,  87,  87,  87,  12,  12,  100,  100]),
    dofs_idx_local = l_dofs_idx,
)
# -----------右臂参数-----------
[... 126 lines cut ...]
        )
        rm_dual.control_dofs_velocity(
            np.array([5,5]), 
            wheel_go_index
        )
    if i==900:
        rm_dual.control_dofs_position(
            np.array([0,0,0,0,0,0,0,0]), 
            l_dofs_idx
        )
        rm_dual.control_dofs_position(
            np.array([0,0,0,0,0,0,0.1,1]), 
            r_dofs_idx
        )
        rm_dual.control_dofs_velocity(
            np.array([5,-5]), 
            wheel_go_index
        )
    cam_pos = rm_dual.get_dofs_position(cam_index[:1])
    print(rm_dual.joints[cam_index[0]])
    
    print(rm_dual.joints[cam_index[0]].get_pos())
    scene.step()
    # cam.render()
```

DaMiTian, 2024-12-22:
and I try to get the camera pos but failed,get the err under:
```
Traceback (most recent call last):
  File "/home/nvidia/Genesis/try.py", line 218, in <module>
    print(rm_dual.joints[cam_index[0]].get_pos())
  File "/home/nvidia/Genesis/genesis/utils/misc.py", line 48, in wrapper
    return method(self, *args, **kwargs)
  File "/home/nvidia/Genesis/genesis/engine/entities/rigid_entity/rigid_joint.py", line 84, in get_pos
    self._kernel_get_pos(tensor)
  File "/home/nvidia/.conda/envs/Genesis/lib/python3.10/site-packages/taichi/lang/kernel_impl.py", line 1178, in __call__
    raise type(e)("\n" + str(e)) from None
taichi.lang.exception.TaichiIndexError: 
File "/home/nvidia/Genesis/genesis/engine/entities/rigid_entity/rigid_joint.py", line 94, in _kernel_get_pos:
            l_info = self._solver.links_info[self._idx, i_b]
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Field with dim 1 accessed with indices of dim 2
```

Kashu7100, 2024-12-26:
for this error, please check https://github.com/Genesis-Embodied-AI/Genesis/pull/323

closed 2025-02-22 by YilingQiao