Genesis-Embodied-AI/genesis-world #128: where is gs.generate()?

hamutama, 2024-12-20:
AttributeError: module 'genesis' has no attribute 'generate'
Do you really have the thing in your propaganda video?

zhouxian, 2024-12-20:
Hi, thank you very much for your interest!

And apologize for letting you down because the generative module is not released yet :)

As stated in our twitter announcement, we are only open sourcing the low level physics engine as of now. The upper level generative framework requires quite a bit of more work and consists of multiple generative modules, and the current status is for demo purpose and not ready for public release yet, but we are actively working on it.

We will gradually release the modules in the coming months, starting with robot policy generation, then to open-world articulated object generation, then to the rest :)

closed 2024-12-22 by ziyanx02

gregory6316, 2024-12-24:
HI! Very impressed with content on your web site, when generative part is planned to be released?