Genesis-Embodied-AI/genesis-world #129: Mimic Support for URDF & Equality Support for MJCF

COMoER, 2024-12-20:
Thanks for releasing such an awesome simulator for robotics researchers. For simulation of the gripper or dexterous hand, the complex linkage structures are often simplified as underactuated joints, which are defined as the mimic attribute or equality attribute in URDF or MJCF respectively. Is there any support for mimic joints in Genesis? I would appreciate it if you could reply.

Richal13Yu, 2025-01-03:
Same question here

RaneWouters, 2025-01-08:
Same question here.

zhouxian, 2025-01-08:
Yes this could be a helpful feature and should be relatively simple to implement

YilingQiao, 2025-02-22:
We now support MJCF's equality constraint and will enhance its compatibility with ball joints in a future PR.
https://github.com/Genesis-Embodied-AI/Genesis/pull/636

PR #928 (2025-03-25): [FEATURE] Support mimic joint in URDF

## Description

Support mimic joint in URDF

## Related Issue

Resolves #678 
Resolves #129 

## Motivation and Context
closed 2025-02-22 by YilingQiao